platformio/platform-espressif32 #1153: Include errors caused by 'esp32_mock.h' in idf up to v5.0

sporniket, 2023-07-24:
**Reference**

_see https://community.platformio.org/t/esp32-mock-h-undefined-esp-websocket-client/34450/3?u=sporniket_

**Description**

It seems that up to v5.0 included, an include file named "esp32_mock.h", located at `components/lwip/test_afl_host/esp32_mock.h`, obviously from a test suite in `lwip` component, is included in regular include files from the framework, causing include errors in VSCode.

The error hampers normal working under VSCode (squiggles disabled for any translation unit ending ending up including this, navigation, code completion is missing for translation units including the affected components), but building seems still OK.

"esp32_mock.h" is included by :
* `components/lwip/port/esp32/include/lwipopts.h`
* `components/lwip/port/esp32/include/arch/sys_arch.h`

In both files, it is included with some guards : 

```cpp
#ifdef __linux__
#include "esp32_mock.h"
#else
```

**Request**

It seems that v5.1 does not include this folder anymore. So I guess that this problem will be fixed by supporting v5.1 of idf.

valeros, 2023-07-24:
Hi @sporniket ! PlatformIO relies on the underlying IDF build system for providing build information, so I'm not sure if we can do here anything, but please provide a minimal example to reproduce the issue.

sporniket, 2023-07-25:
Hello,

> Hi @sporniket ! PlatformIO relies on the underlying IDF build system for providing build information, so I'm not sure if we can do here anything, but please provide a minimal example to reproduce the issue.

Hello @valeros, I believe that my issue looks like a roundabout way of asking for supporting IDF 5.1 that has removed the problem. I understand that it seems not trivial though, as I can see in #1129 

Anyway, I have extracted the minimal exemple from my current project, deleting almost all the file : the `main.cpp` file include the `NetworkTimeKeeperEsp32.cpp` file that include `esp_sntp.h` : 
https://github.com/sporniket/the-clock--firmware-esp32-idf/tree/minimal-code-issue-1153-at-github--platformio--platform-espressif32

Then, when altering the type declaration of `networkTimeKeeper` on line 34, e.g. by deleting the `T` in `NetworkTimeKeeperEsp32`, the not existing typename is not reported.

> The error hampers normal working under VSCode (squiggles disabled for any translation unit ending ending up including this, navigation, code completion is missing for translation units including the affected components), but building seems still OK.

It seems that navigation and code completion are ok now, compared to my last session of work -maybe a rebuild of Intellisense took place ?-. So the only concern remaining is the "squiggles disabled for any translation unit ending ending up including this", in effect, preventing to see errors following the include.

**Other reference**

I found this issue on the IDF project : https://github.com/espressif/esp-idf/issues/10802

Jason2866, 2023-07-27:
@sporniket You can use my [fork](https://github.com/Jason2866/platform-espressif32/tree/Arduino/IDF5) which does work with IDF5.1
Use with
```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
```

sporniket, 2023-07-29:
Seems to be working. 

before : 
[image]

after : 
[image]

I tried to build my full code with the fork too, I didn't have any problem.

sporniket, 2023-12-15:
since then, the problem has been fixed.

closed 2023-12-15 by sporniket