platformio/platform-espressif32 #1154: SNTP very slow

marchingband, 2023-07-24:
i am using 
```
platform_packages =
    framework-espidf @ ~3.40404.0
    toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```
and the SNTP server often takes > 30 seconds to obtain a sync signal. It should be under 4 seconds.

I found this issue in esp-idf https://github.com/espressif/esp-idf/issues/4718 ... 

here is my code:
```
    ESP_LOGI(TAG, "Initializing SNTP");
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, "ntp.aliyun.com");
    sntp_setservername(1, "ntp1.aliyun.com");
    sntp_setservername(2, "1.uk.pool.ntp.org");
    sntp_setservername(3, "2.uk.pool.ntp.org");
    sntp_setservername(4, "pool.ntp.org");
    sntp_setservername(5, "time1.google.com");
    sntp_set_time_sync_notification_cb(time_sync_notification_cb);
    sntp_init();
    while (sntp_get_sync_status() == SNTP_SYNC_STATUS_RESET && ++retry < retry_count)
    {
        ESP_LOGI(TAG, "Waiting for system time to be set... (%d/%d)", retry, retry_count);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
```

the log
```
I (43123) m5stack:wake-up: Initializing SNTP
[0;32mI (43123) m5stack:wake-up: Waiting for system time to be set... (1/80)
[0;32mI (44133) m5stack:wake-up: Waiting for system time to be set... (2/80)
[0;32mI (45133) m5stack:wake-up: Waiting for system time to be set... (3/80)
[0;32mI (46133) m5stack:wake-up: Waiting for system time to be set... (4/80)
[0;32mI (47133) m5stack:wake-up: Waiting for system time to be set... (5/80)
[0;32mI (48133) m5stack:wake-up: Waiting for system time to be set... (6/80)
[0;32mI (49133) m5stack:wake-up: Waiting for system time to be set... (7/80)
[0;32mI (50133) m5stack:wake-up: Waiting for system time to be set... (8/80)
[0;32mI (51133) m5stack:wake-up: Waiting for system time to be set... (9/80)
[0;32mI (52133) m5stack:wake-up: Waiting for system time to be set... (10/80)
[0;32mI (53133) m5stack:wake-up: Waiting for system time to be set... (11/80)
[0;32mI (54133) m5stack:wake-up: Waiting for system time to be set... (12/80)
[0;32mI (55133) m5stack:wake-up: Waiting for system time to be set... (13/80)
[0;32mI (56133) m5stack:wake-up: Waiting for system time to be set... (14/80)
[0;32mI (57133) m5stack:wake-up: Waiting for system time to be set... (15/80)
[0;32mI (58133) m5stack:wake-up: Waiting for system time to be set... (16/80)
[0;32mI (59133) m5stack:wake-up: Waiting for system time to be set... (17/80)
[0;32mI (60133) m5stack:wake-up: Waiting for system time to be set... (18/80)
[0;32mI (61133) m5stack:wake-up: Waiting for system time to be set... (19/80)
[0;32mI (62133) m5stack:wake-up: Waiting for system time to be set... (20/80)
[0;32mI (63133) m5stack:wake-up: Waiting for system time to be set... (21/80)
[0;32mI (64133) m5stack:wake-up: Waiting for system time to be set... (22/80)
[0;32mI (65133) m5stack:wake-up: Waiting for system time to be set... (23/80)
[0;32mI (66133) m5stack:wake-up: Waiting for system time to be set... (24/80)
[0;32mI (67133) m5stack:wake-up: Waiting for system time to be set... (25/80)
[0;32mI (68133) m5stack:wake-up: Waiting for system time to be set... (26/80)
[0;32mI (69133) m5stack:wake-up: Waiting for system time to be set... (27/80)
[0;32mI (70133) m5stack:wake-up: Waiting for system time to be set... (28/80)
[0;32mI (71133) m5stack:wake-up: Waiting for system time to be set... (29/80)
[0;32mI (72133) m5stack:wake-up: Waiting for system time to be set... (30/80)
[0;32mI (73133) m5stack:wake-up: Waiting for system time to be set... (31/80)
[0;32mI (74133) m5stack:wake-up: Waiting for system time to be set... (32/80)
[0;32mI (74793) m5stack:wake-up: Notification of a time synchronization event
```

valeros, 2023-07-25:
Hi @marchingband ! I'm not sure if this issue has something to do with PlatformIO. Have you tried compiling the same code but using the native IDF build system? Does it work faster? Also, any chance that IDF v5 will show better results?

marchingband, 2023-07-26:
The reason I posted here is that the issue I linked seems to point to a patch. My project is far too complicated to try on another version, or even with IDF toolchain, without significant pain (multiple components with their own sub components  linked in, all pinned to this same version). Not sure what it would prove anyhow. Are you able to determine if that patch has been applied to the pio idf, or if it could be? It does seem like I have the same issue that was patched in the issue I linked.

Jason2866, 2023-07-26:
The issue you linked is for IDF 4.1. You are using IDF 4.4.4.  The used IDF platformio version is exactly the same as the release IDF 4.4.4 version.
This github is not the place for discussing IDF issues at all.

valeros, 2023-07-26:
> Are you able to determine if that patch has been applied to the pio idf, or if it could be? It does seem like I have the same issue that was patched in the issue I linked.

We don't patch anything, we simply reuse the same package published by the Espressif team from the official IDF repository.
The issue seems project-specific, so I'm not sure how to help your here.

closed 2023-07-26 by valeros