platformio/platform-espressif32 #1155: src_filter using espidf

fariouche, 2023-07-25:
Hello!

I’m wondering, why is the espidf ignoring the src_filter… I can see that platformio can generate a CMakeLists.txt in src if none is provided, so why doesn’t it use the src_filter when generating it?

Maybe because platformio cannot know if the CMakeLists.txt file was auto generated or modified by the user?
In that case why not generate the CMakeLists.txt file inside the build dir? (using absolute paths in that case when referencing the source folder)… and to stay backward compatible, platformio can do it only if no src/CMakeLists.txt exists.

It’s kind of annoying to have to maintain this file for esp32 only 

Thanks!

valeros, 2023-07-25:
Hi @fariouche ! The main reason is backward compatibility with the native IDF build system so developers can easily switch between environments. The `CMakeLists.txt` file is part of the project and putting it in a volatile path like build directory is a bit hacky workaround, besides different behavior when there is/isn't a `src/CMakeLists.txt` file will be surprising for developers.

closed 2023-07-25 by valeros

fariouche, 2023-07-25:
ok, thanks. I will not argue, but I think that what is hacky is the fact that src_filter does not work the same way depending on the platform and the fact that a file is generated inside "our" src folder :wink:
Anyway, would you accept a patch from me to add this feature? (keeping current behavior unchanged of course)
Maybe by passing the src_filter to the cmakelists.txt and let cmake handle the filter (that way changing the src_filter in platformio.ini can be done and the cmakelists.txt can adapt automatically )?

valeros, 2023-07-25:
> Anyway, would you accept a patch from me to add this feature?

Sure thing, just keep your changes compatible with both PlatformIO and IDF build system.