platformio/platform-espressif32 #1156: 404 error

CharlesXSong, 2023-07-25:
> Tool Manager: Looking for another mirror...
Tool Manager: Warning! Package Mirror: Got the unrecognized status code '404' when downloaded https://dl.registry.ns3.platformio.org/tools/cb/36/aefbf6f69f5e27302916f0a1d13c03da57946e5b251be009f7cd93179d41/framework-arduinoespressif32-3.20004.0.tar.gz

when I was building a project, how to solve it?

ivankravets, 2023-07-25:
Thanks for the report. The issue has been fixed.

ivankravets, 2023-07-25:
What is your speed for this Chinese mirror https://dl.registry.ns3.platformio.org/tools/cb/36/aefbf6f69f5e27302916f0a1d13c03da57946e5b251be009f7cd93179d41/framework-arduinoespressif32-3.20004.0.tar.gz ?

CharlesXSong, 2023-07-25:
No, I didn't use a proxy. I'm using the China Telecom Wuhan Branch(1000Mbps).  
But the speed increased to 500KB/s when I switched the wired network to my phone's USB shared network (China Mobile 5G)
Anyway, I‘m currently downloading this package. Thank you very much!