platformio/platform-espressif32 #1157: Running `idf.py reconfigure` from PlatformIO

kevinhikaruevans, 2023-07-26:
Is there a way of running `idf.py reconfigure` from PlatformIO's interface? Or perhaps an easy way of accessing `idf.py`?

I have some components that I would like to bring down from the ESP-IDF Component Manager/Registry https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html and it requires me to run `idf.py reconfigure` from my project's directory. I don't see an option to do this in the PIO UI. Is there something that I'm missing?

valeros, 2023-07-27:
Hi @kevinhikaruevans ! There is no support for the `idf.py` utility as PlatformIO uses its own build tools. However, it should be possible to use IDF components. First, add that dependency to the `src/idf_component.yml` file (if it doesn't exist, create a new one) with the following content:

```yml
dependencies:
  espressif/mdns: ">=1.2.0"
```

Then clean and compile your project from scratch, the internal IDF build system will download that dependency and PlatformIO should pick up that new component automatically.

kevinhikaruevans, 2023-07-27:
I've done that. Should there be a `dependencies.lock` file generated?

kevinhikaruevans, 2023-07-27:
Thanks! I understand now.

closed 2023-07-27 by kevinhikaruevans

gojimmypi, 2024-07-15:
Hi @valeros 

I have a similar question but with _Managed Components_ available from https://components.espressif.com

Specifically, I've been working with @Jason2866 to improve wolfSSL support in PlatformIO, and in general with the ESP-IDF.

I've successfully published a PlatformIO version of wolfSSL here: https://registry.platformio.org/libraries/wolfssl/wolfSSL 

and it works by including this text in the `idf_component.yml` as described above: 

```
dependencies:
  wolfssl/wolfssl: "^5.7.1-preview2e"
```

For reference, see also the ESP Registry version here: https://components.espressif.com/components/wolfssl/wolfssl

However... what I need to do, is to inform the ESP-IDF that I have a _local_ wolfssl component _that can be used in the ESP-IDF_ (specifically by the `esp-tls` component of ESP-IDF).

This is described in https://github.com/espressif/esp-idf/issues/13966#issuecomment-2165529503 and specifically implemented in my case like [this (WIP)](https://github.com/gojimmypi/esp-idf/blob/7de6d455fe775f188f71fd5b913c04d382003389/components/esp-tls/CMakeLists.txt#L64):

```
idf_component_get_property(wolfssl [wolfssl | wolfssl__wolfssl | esp-wolfssl | etc] COMPONENT_LIB)
```

As you may know: when PlatformIO installs "components" they are _not_ in the local project components directory. Instead, the source is copying into `[project]\.pio\libdeps\[env:[name, e.g. esp32dev]]\wolfssl` for example `\.pio\libdeps\esp32dev\wolfssl`.

I need to somehow identify that as a "component" to  pass a parameter to `idf_component_get_property` so that the ESP-IDF "knows" about wolfSSL in the local project, and not in the ESP-IDF `components` directory.

Clear as mud? I realize this seems like a convoluted solution. It does however, solve the license and distribution issues noted in the home page of [Espressif/esp-wolfssl](https://github.com/espressif/esp-wolfssl). It seems to work well for regular "components" as recognized by Espressif: either simply in the local project "components" directory or installed as a Managed Component (e.g. named `wolfssl__wolfssl`). But this does not seem to work for PlatformIO components.

My only alternative seems to be installing the Managed Component directly... but it needs `idf.py` and as you noted above:

> There is no support for the idf.py utility as PlatformIO uses its own build tools

Do you have any suggestions on how I should proceed? Thank you.

valeros, 2024-07-18:
Hi @gojimmypi, libraries installed via `lib_deps` won't be recognized as IDF components, but the native IDF approach with regular components should work as expected.

gojimmypi, 2024-07-18:
Hi @valeros - thank you for your reply & confirmation. (I'll stop trying to figure that one out!)

Given your comment above, just to clarify:

> no support for the idf.py

... and the fact that managed components _require_ the `idf.py` for installation: can I assume that enabling the `idf.py` in a given directory for a PlatformIO  project anyhow... will allow the Managed Component to be _installed_ and used as normal, but then later used without problem in the `pio` in the local project? 

I ask as Managed Components are a little different than the IDF components. Just want to make sure there are no known issues. 

This is of course assuming the same library is _not_ installed as a `lib_deps` for pio.

valeros, 2024-07-18:
>  the fact that managed components require the idf.py for installation

Do they? I believe there is no need to integrate the entire `idf.py` package to use the managed components. There is an underlying package called [idf-component-manager](https://github.com/espressif/idf-component-manager) responsible for handling third-party components. We declare this package as a [dependency](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/espidf.py#L1211) so it should be available for the native IDF build system when you compile your project using PlatformIO.

gojimmypi, 2024-07-18:
> Do they? I believe there is no need to integrate the entire idf.py package

Ah yes, one could in theory manually craft an `idf_component.yml` file. For [wolfssl](https://components.espressif.com/components/wolfssl/wolfssl), the instructions are vastly simpler:

"_To add this component to your project, run:_"
```
idf.py add-dependency "wolfssl/wolfssl^5.7.0"
```

I tried your `idf-component-manager` suggestion. Indeed you are correct: it appears PlatformIO honors the `idf_component.yml` file. Very cool. 

I copied a `main/idf_component.yml` into the [template example](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/template) and removed the existing `wolfssl` [directory](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl).

```
dependencies:
  wolfssl/wolfssl: "^5.7.1-preview2e"
  protocol_examples_common:
    path: ${IDF_PATH}/examples/common_components/protocol_examples_common
  esp_stubs:
    path: ${IDF_PATH}/examples/protocols/linux_stubs/esp_stubs
    rules:
    - if: "target in [linux]"

```

Has that always worked? I thought I had tried that and the component was not actually downloaded with pio. Today it seems to have worked! (PlatformIO Core, version 6.1.15) yay! Thanks very much for the suggestion.

For completeness for the future reader: one could also use the [install script](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/setup.sh) as well, but there's something really nice about the fixed release, no-clone, easy install of the Managed Components.

Perhaps at some point, this functionality could be moved the `platformio.ini` file?