platformio/platform-espressif32 #1158: AddPreAction hook executed after the build (buildfs)

bmedici, 2023-07-26:
The execution order is:
1) Building FS image from 'data' directory to .pio/build/ser/spiffs.bin
2) before_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])
3) after_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])

Where I would have expected before_buildfs first, then "Building FS image", and finally after_buildfs.

Some context:

platformio.ini
```
[env:esp32]
	board = esp32dev
	framework = arduino
	board_build.partitions = shared/partitions.csv
	extra_scripts =
    shared/extra_script.py
```

shared/extra_script.py
```
import os
Import("env")

print(">>> Current CLI targets", COMMAND_LINE_TARGETS)
print(">>> Current Build targets", BUILD_TARGETS)
# print(env.Dump())
# host = env.GetProjectOption("custom_ping_host")

def before_buildfs(source, target, env):
    print("\n>>>  before_buildfs: build-www")
    env.Execute("ruby shared/build-www.rb")

    print("\n>>> before_buildfs: copy assets")
    env.Execute("rsync --delete -av shared/assets/ data/assets/")

    print("\n>>> before_buildfs: gzip assets")
    env.Execute("find data/ -type f \( -name \*.css -o -name \*.js -o -name \*.html \) -exec gzip -f \"{}\" \;")

    print("\n>>> before_buildfs: total size")
    env.Execute("du -sh data")
    env.Execute("du -sh data/*")

env.AddPreAction("buildfs", before_buildfs)

def after_buildfs(source, target, env):
    print("\n>>> after_buildfs: unzip assets")
    # env.Execute("find data/ -type f -name *.gz -exec gunzip -f \"{}\" \;")

env.AddPostAction("buildfs", after_buildfs)
```

execution output for `platformio run --target buildfs --environment ser`
```
Processing ser (board: esp32dev; framework: arduino; platform: espressif32  @6.3.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing jfturcot/SimpleTimer @ *
Library Manager: Warning! Could not install `{'owner': 'jfturcot', 'name': 'SimpleTimer', 'version': '*'}` dependency for the`vanbox-lib` package
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 57 compatible libraries
Scanning dependencies...
Dependency Graph
|-- vanbox-lib @ 1.0.0
|-- SimpleTimer @ 0.0.0-alpha+sha.b30890b8f7
|-- RunningAverage @ 0.4.3
|-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.19.2
|-- ESP32Time @ 2.0.4
|-- SPI @ 2.0.0
|-- Button2 @ 2.2.4
|-- Coordinates @ 1.0.0
|-- PID @ 1.2.1
|-- Adafruit Unified Sensor @ 1.1.11
|-- Adafruit MPU6050 @ 2.2.4
|-- Adafruit BMP085 Library @ 1.2.2
|-- Adafruit BME280 Library @ 2.2.2
|-- Adafruit SGP30 Sensor @ 2.0.0
|-- EspSoftwareSerial @ 6.16.1
|-- TinyGPSPlus-ESP32 @ 0.0.2
|-- LovyanGFX @ 1.1.7
|-- Wire @ 2.0.0
Building in release mode
>>> Current CLI targets ['buildfs']
>>> Current Build targets ['buildfs']
Building FS image from 'data' directory to .pio/build/ser/spiffs.bin
/setup.html.gz
/img/topview160.png
/img/cog16.png
/location.html.gz
/index.html.gz
/assets/css/jquery-ui.min.css.gz
SPIFFS_write error(-10010): unknown
/assets/img/topview160.png
error adding file!
Error for adding content from css!
/assets/img/cog16.png
before_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])

>>>  before_buildfs: build-www
ruby shared/build-www.rb
> build_page  [index]
  layout  [2023-07-26 19:39:41 +0200]
  page    [2023-07-24 21:34:53 +0200]
  output  [0]
> build_page  [location]
  layout  [2023-07-26 19:39:41 +0200]
  page    [2023-07-22 01:19:54 +0200]
  output  [0]
> build_page  [setup]
  layout  [2023-07-26 19:39:41 +0200]
  page    [2023-07-21 21:35:06 +0200]
  output  [0]

>>> before_buildfs: copy assets
rsync --delete -av shared/assets/ data/assets/
building file list ... done
deleting css/style.css.gz
deleting css/jquery-ui.min.css.gz
deleting css/helpers.css.gz
deleting css/bootstrap.min.css.gz
css/
css/bootstrap.min.css
css/helpers.css
css/jquery-ui.min.css
css/style.css

sent 174360 bytes  received 114 bytes  348948.00 bytes/sec
total size is 189322  speedup is 1.09

>>> before_buildfs: gzip assets
find data/ -type f \( -name \*.css -o -name \*.js -o -name \*.html \) -exec gzip -f "{}" \;

>>> before_buildfs: total size
du -sh data
 88K	data
du -sh data/*
 56K	data/assets
 20K	data/img
4.0K	data/index.html.gz
  0B	data/js
4.0K	data/location.html.gz
4.0K	data/setup.html.gz
after_buildfs(["buildfs"], [".pio/build/ser/spiffs.bin"])

>>> after_buildfs: unzip assets
=================================================================================================================== [SUCCESS] Took 3.50 seconds ===================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
ser            SUCCESS   00:00:03.495
```

ArnoCtr, 2023-12-23:
Same issue.

bmedici, 2023-12-26:
Any update on this (looks-like-a-) bug ?

valeros, 2024-03-08:
I'd recommend using the path to the file system binary instead of the `buildfs` target (see an [example](https://docs.platformio.org/en/latest/scripting/actions.html)):

```python
env.AddPreAction("$BUILD_DIR/${ESP32_FS_IMAGE_NAME}.bin", before_buildfs)
env.AddPostAction("$BUILD_DIR/${ESP32_FS_IMAGE_NAME}.bin", after_buildfs)
```