platformio/platform-espressif32 #1159: Wrong build size when using custom factory and ota partitions

kostuch, 2023-07-27:
I try to build for setup when I have custom small app factory partition and two ota.
CASE1:
When partition table has only factory partition, the compilation size reflect the partition size
CASE2:
When I add one or two ota partitions, the build (and probably upload) reflect the ota partition size.
The code is just empty setup() and loop().

platformio.ini

```
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino, espidf
board_build.partitions = part_table.csv
```

part_table.csv

```
nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        512K,
#ota_0,    app,  ota_0,   ,        1M,   #at this moment commented out and everything is fine
#ota_1,    app,  ota_1,   ,        1M,
```

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, 
olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - framework-espidf @ 3.40404.0 (4.4.4)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating partitions .pio\build\esp32-c3-devkitm-1\partitions.bin
Linking .pio\build\esp32-c3-devkitm-1\firmware.elf
Retrieving maximum program size .pio\build\esp32-c3-devkitm-1\firmware.elf
Checking size .pio\build\esp32-c3-devkitm-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.8% (used 9044 bytes from 327680 bytes)
Flash: [===       ]  29.1% (used 152336 bytes from 524288 bytes)   <--- 512k as defined for factory partition
```

After uncomment ota definition in part_tables.csv and build again

```
RAM:   [          ]   2.8% (used 9044 bytes from 327680 bytes)
Flash: [=         ]  14.5% (used 152336 bytes from 1048576 bytes)  <---1M as defined for ota partition
```

Setting offset to 0x10000 for factory makes no change to this behavior.
This is for esp32c3, but for esp32 I have same situation.
What can be done to force build and upload to factory partition, but have also ota deifinitions?

valeros, 2023-07-27:
Thanks for reporting, should be fixed in the development branch.

closed 2023-07-27 by valeros

kostuch, 2023-07-27:
@valeros 
Thank you for the immediate patch.
How about flashing in such case? Will it also set the factory partition as a target for upload?

valeros, 2023-07-27:
> How about flashing in such case? Will it also set the factory partition as a target for upload?

Flashing is completely different process, the default address for application is set to `0x10000` and is not affected by the values from the partition file.

commit f1fce6c1 (2023-07-27): Improve dynamic memory size calculation // Resolves #1159

For better results a partition with either `factory` or `ota_0` subtype should be used instead of selecting the biggest partition size with the type `app`.

If both partitions are set, then the partition with the `factory` subtype is used by default.