platformio/platform-espressif32 #1160: ESP_RST_INT_WDT on deep sleep start

tnicolas42, 2023-07-27:
I have an issue with deepsleep start.

Since one month, on some random esp32 I can’t start deepsleep mode. I have 20 boards, the bug appears on 12 of them.

When I run this code, the esp32 auto restart imediatly when calling `esp_deep_sleep_start()` with reason 5 (`ESP_RST_INT_WDT: Reset (software or hardware) due to interrupt watchdog` )

The bug first appear when I did some updates but I don’t remember witch one.

Specifications

- ESP32: **ESP32-S3-WROOM-2-N32R8V** (On a custom board used everyday since 1 year)
- platformio version: ****PlatformIO Core, version 6.1.9**** (tested on other versions like ****6.1.4****)
- platform-espressif32 version: **6.3.2** (tested on ther release like **5.4.0**, **6.2.0**, etc.)
- code uploaded with [this programmer](https://www.digikey.fr/fr/products/detail/espressif-systems/ESP-PROG/10259352)
- code uploaded from macbook pro M1 and windows

My code :

```c++
#include <Arduino.h>

void setup() {
	Serial.begin(115200);
	esp_reset_reason_t reason = esp_reset_reason();
	log_printf("Wakeup reason: %d\n", reason);

	esp_sleep_enable_ext0_wakeup(GPIO_NUM_3, 1);
	delay(100);
	esp_deep_sleep_start();
}

void loop() {}
```

platformio.ini

```bash
[env]
platform = espressif32 @ 6.3.2
framework = arduino
monitor_speed = 115200
monitor_filters = direct
build_flags = -DCORE_DEBUG_LEVEL=5

[env:maxi]
board = esp32-phyling
```

custom board used

```bash
{
    "build": {
      "arduino":{
        "ldscript": "esp32s3_out.ld",
        "partitions": "default_16MB.csv",
        "memory_type": "opi_opi"
      },
      "core": "esp32",
      "extra_flags": [
        "-DARDUINO_RUNNING_CORE=1",
        "-DARDUINO_EVENT_RUNNING_CORE=0",
        "-DARDUINO_USB_MODE=1",
        "-DARDUINO_ESP32S3_DEV",
        "-DBOARD_HAS_PSRAM",
        "-DESP32S3_DEV",
        "-mfix-esp32-psram-cache-issue",
        "-fexceptions"
      ],
      "f_cpu": "240000000L",
      "f_flash": "80000000L",
      "flash_mode": "dout",
      "boot": "opi",
      "boot_freq": "80m",
      "hwids": [
        [
          "0X303A",
          "0x1001"
        ]
      ],
      "mcu": "esp32s3",
      "variant": "esp32s3",
      "psram_type": "opi"
    },
    "connectivity": [
      "wifi",
      "bluetooth"
    ],
    "debug": {
      "default_tool": "esp-builtin",
      "onboard_tools": [
        "esp-builtin"
      ],
      "openocd_target": "esp32s3.cfg"
    },
    "frameworks": [
      "arduino",
      "espidf"
    ],
    "name": "Espressif ESP32-S3 Dev Module",
    "upload": {
      "flash_size": "16MB",
      "maximum_ram_size": 327680,
      "maximum_size": 8388608,
      "require_upload_port": true,
      "speed": 921600
    },
    "url": "https://www.adafruit.com/product/5290",
    "vendor": "Espressif"
  }
```

logs

```bash
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379458
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28fc
entry 0x403c98bc
[   121][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Wakeup reason: 5
ESP-ROM:esp32s3-20210327    #  <-- auto restart (repeat forever)
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379458
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28fc
entry 0x403c98bc
[   121][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Wakeup reason: 5
```

valeros, 2023-07-27:
Hi @tnicolas42, have you tried the same code in Arduino IDE?

tnicolas42, 2023-07-28:
Yes I tried and I don't have any error

Jason2866, 2023-07-28:
Do you have an other S3 board without OPI flash/PSRAM?
Or if you have not, can you try without PSRAM?

Jason2866, 2023-07-28:
Mhh, this `"maximum_size": 8388608,` is wrong in your boards json.
Can end in endless reboot. The correct entry for 16MB flash is `"maximum_size": 1677216,`
and the entry ` "-mfix-esp32-psram-cache-issue",` is only for some esp32 boards with PSRAM. It is not needed for ANY S3 MCU.

tnicolas42, 2023-08-07:
I made a mistake actually, the error doesn't occur when I upload code from the Arduino IDE...
And I tested without PSRAM, still the same bug.
And thank you, @Jason2866, I made the changes!

tnicolas42, 2023-08-16:
SOLVED
In platformio.ini, I update the package framework-arduinoespressif32
```
platform_packages =
	framework-arduinoespressif32 @ 3.20011.230801
	tool-esptoolpy @ 1.40501.0
	tool-mkfatfs @ 2.0.1
```

closed 2023-08-16 by tnicolas42