platformio/platform-espressif32 #1162: Support "Arduino Nano ESP32"

maxgerhardt, 2023-07-27:
https://store.arduino.cc/products/nano-esp32

An ESP32-S3 board, 16 MByte QSPI flash, 8 MB octal-SPI PSRAM.

Supported in regular Arduino-ESP32 since version v2.0.10, current version is v2.0.11. This platform currently uses v2.0.9.

msrba, 2023-07-31:
Is there already a plan or roadmap known if and when the support of the Arduino Nano ESP32 will be implemented.

closed 2023-08-01 by valeros

pillo79, 2023-08-04:
Hello @maxgerhardt @valeros, Arduino dev here. Some Arduino forum users posted about issues with the Nano ESP32 and PlatformIO, so I decided to dig a little. I might be mistaken, but there may be a big issue in how the Nano ESP32 is supported in PlatformIO.

We have several Nano boards that share the same pin names, regardless of the CPU being used, so we worked with Espressif to allow Arduino users to access the Arduino API via their usual pin numbers. 
That has required introducing in 2.0.10+ a custom build step (detailed [here](https://github.com/espressif/arduino-esp32/blob/release/v2.x/platform.txt#L189)) where the core code is compiled with `-DARDUINO_CORE_BUILD`, while the user sketch is not. If the board then defines `-DBOARD_HAS_PIN_REMAP`, transparent pin remapping is performed _exactly where it is required_ - when the sketch calls the core functions. 

I see no mention of the `-DARDUINO_CORE_BUILD` step in 95125e2. If that is never set during compilation, every call inside the core would be pin-remapped as well, creating basically an Enigma machine for pin numbers. :slightly_smiling_face: 
We did not anticipate this situation, but we can detect it and `#error` to avoid generating buggy code - this will be included in the next core release.

The ideal solution would be to compile, as the Arduino IDE does, the core files with a different `#define`. 
If that is not workable, then please remove the `-DBOARD_HAS_PIN_REMAP` from your board definition. That way, the core will cleanly revert to using GPIO numbers everywhere. This will still cause untold amounts of confusion between Arduino IDE and PlatformIO users sharing sketches, but at least the functionality will be predictable.
We can also add a warning in this case, so users at least will be informed when the build system is forcing the API to use GPIO numbers on those specific boards that are supposed to have custom pin numbers. 

What do you think?

maxgerhardt, 2023-08-04:
Sounds like you've found a bug, or rather desync from the Arduino IDE, in the PlatformIO builder scripts located at https://github.com/espressif/arduino-esp32/blob/master/tools. The solution should just be to add a new entry to the `CPPDEFINES` array, in a way that's only in the construction environment (SCons stuff) for the core files. Strictly speaking that should be a new issue. What do you think, @valeros?

valeros, 2023-08-04:
Hi guys, the `ARDUINO_CORE_BUILD` macro is already handled in our latest package, but the changes hasn't yet been pushed to the upstream ESP32 core. I'll try to synch changes today a bit later.

maxgerhardt, 2023-08-04:
I see https://github.com/espressif/arduino-esp32/pull/8488 has landed 3 minutes ago, nice work :)

pillo79, 2023-08-04:
Thanks, awesome reaction time! :100:

johnrbell, 2023-08-11:
Hate to be that guy, but how do we take advantage of these changes?

maxgerhardt, 2023-08-11:
It's indeed not immediately obvious:
1. Create an arbitrary project, e.g. Board "ESP32 Dev Module", Framework "Arduino"
2. Open the platformio.ini file of the created project and overwrite it with
```ini
[env:arduino_nano_esp32]
platform = https://github.com/platformio/platform-espressif32.git
board = arduino_nano_esp32
framework = arduino
```
3. Save the file, wait for PlatformIO to reinitialize the project
4. Change to the arduino_nano_esp32 project in the project environment switcher in the bottom taskbar of VSCode

nicolasff, 2023-08-12:
The comment above did not resolve this issue for me. PlatformIO correctly read its `.ini` file but still could not find the board for some reason, failing with:
```
Resolving arduino_nano_esp32 dependencies...
UnknownBoard: Unknown board ID 'arduino_nano_esp32'
```
This was with `platformio.ini` containing exactly the same 4 lines.

This may have happened because I already had a local checkout of this repo in PlatformIO's cache, which did not include this change. In order to force a refresh, I updated the URL to point to the `develop` branch, changing the line to:
```ini
platform = https://github.com/platformio/platform-espressif32.git#develop
```
Doing this showed PlatformIO pulling commit f1fdbc5, which is currently the latest commit on this branch:
```
Platform Manager: Installing git+https://github.com/platformio/platform-espressif32.git#develop
git version 2.38.1
Cloning into …

Platform Manager: espressif32@6.3.2+sha.f1fdbc5 has been installed!
```

A note of caution (and maybe maintainers provide guidance here): doing this means you're using a version of `platform-espressif32` which is currently under development, and not part of a stable release. I would recommend keeping an eye on the release page of this repo, and changing the `platform` URL to point to a git tag instead of the `develop` branch once one is available. As of right now, only `develop` seems to contain `arduino_nano_esp32.json`, as indicated by this line on the GitHub page for the commit that introduced this file (https://github.com/platformio/platform-espressif32/commit/95125e2627e13e5625e5cb68fe69830a01f1c723):

[image]

   
I hope this helps, it worked for me. And thanks Valerii for submitting this file!

commit 95125e26 (2023-08-01): Update Arduino core to v2.0.11

- Added several new boards (including Nano ESP32) // Resolves #1162
- Added DFU upload method via dfu-util for the new Arduino Nano ESP32
- Updated IDF version to v4.4.5 for mixed Arduino/IDF projects