platformio/platform-espressif32 #1166: Program size check is not checked against all available partitions

maxbalan, 2023-08-06:
Hi,
I am trying to build a program and flash it to a different partition then factory one, hence my `app0` is of `128kb` size and `app2` is 3mb since I intend to flash it to the `app2`

when building the `checkprogsize` fails because it only check the partition size that starts at `0x10000` but does not look for other partitions size.

I think the firmware size check against the partitions should be done in fashion where it check that the program will fit in at least one partition before failing the build/upload.

platformio.ini
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
monitor_speed = 115200
board_build.partitions = partitions.csv
```

partitions.csv
```
# Name,	 Type,	 SubType,    Offset,	  Size,	 Flags
nvs,	 data,	 nvs,	     0x9000,	  0x5000,	
otadata, data,	 ota,	     0xe000,	  0x2000,	
app0,	 app,	 ota_0,	     0x10000,	  0x20000,	
app1,	 app,	 ota_1,	     0x30000,     0x300000,
```

terminal output
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.3.2+sha.95e0a73) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50100.0 (5.1.0) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Generating partitions .pio/build/esp32doit-devkit-v1/partitions.bin
Linking .pio/build/esp32doit-devkit-v1/firmware.elf
Retrieving maximum program size .pio/build/esp32doit-devkit-v1/firmware.elf
Checking size .pio/build/esp32doit-devkit-v1/firmware.elf
Error: The program size (180725 bytes) is greater than maximum allowed (131072 bytes)
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
*** [checkprogsize] Explicit exit, status 1
RAM:   [          ]   3.2% (used 10644 bytes from 327680 bytes)
Flash: [==========]  137.9% (used 180725 bytes from 131072 bytes)
================================= [FAILED] Took 3.54 seconds =================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.  
 ```

valeros, 2023-08-07:
Hi @maxbalan, PlatformIO is not aware of what partition will be used for the application. Selecting the biggest partition with the `app` type will lead to issues similar to #1159. I've pushed a workaround that will allow developers to select an application partition used for memory calculations, for example:
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
monitor_speed = 115200
board_build.partitions = partitions.csv
board_build.app_partition_name = app1
```

To test it out, just pull the latest changes from this repo.

maxbalan, 2023-08-22:
Hi @valeros thank you for your reply and solution, I will test it as soon as I can and report back with the results

maxbalan, 2023-09-17:
Hi @valeros I did test this feature as part of the release 6.4.0, and can confirm it is working as expected, the build is passed when setting parameter `board_build.app_partition_name = app1`

closed 2023-09-18 by valeros