platformio/platform-espressif32 #1170: Failure to Upload Filesystem Image to ESP32 Nano

alireza7575, 2023-08-16:
When using the developer version of PlatformIO or Arduino 2.1.1 to upload a Filesystem image to the ESP32 Nano, an error prevents the process from completing. Furthermore, SPIFFS cannot be initialized, even when using the default SPIFFS_Test example.

`
Opening DFU capable USB device...
Device ID 2341:0070
Device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(3) = Device is unable to write memory
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0101
Device returned transfer size 4096
Copying data from PC to DFU device

Download        [                         ]   0%            0 bytes failed!
DFU state(10) = dfuERROR, status(3) = Device is unable to write memory
*** [uploadfs] Error 74
`
dose anyone facing the same issue?

valeros, 2023-08-17:
Hi @alireza7575 ! Are you able to upload FS using Arduino IDE?

alireza7575, 2023-08-17:
No, I could not. I get an error.

`SPIFFS Error: partition size could not be found!`

valeros, 2023-08-17:
The entire issue is caused by the unusual for ESP32 upload method. I'd recommend asking the Arduino team how to upload FS image in the first place.

As a wild guess, you can try to specify the precise location where you want to upload FS image, just add the following line to your platformio.ini:
```ini
upload_flags =
    -s
    0x610000:leave
```

where `0x610000` is the location of the `ffat` partition from the partitions table file for this board.

closed 2023-08-18 by valeros