platformio/platform-espressif32 #1180: Complex ESP-IDF dependencies seem to break pio build script - TypeError: ‘NoneType’ object is not iterable

nicklasb, 2023-09-03:
Hi!
Thanks for a great product!
I have an issue where I have ESP-IDF components with their own dependencies seemingly causing problems for Platformio. The reason I think that may be the case, is that it doesn’t seem to break in running idf.py build in the same folder. 
ESP-IDF doesn’t seem to mind the current config.

It doesn’t compile without errors there, but that is not the problem. Abbreviated, the error is:
```
TypeError: 'NoneType' object is not iterable:
for cg in target_compile_groups:
```
It happens in espidf.py, line 614. 

If I monkey-patch the script to return [] if it is None, the script continues. But I am not sure it works properly then as I get a lot of other problems, though, so there seems to be some case that isn’t covered or something.

The repository is at: [GitHub - nicklasb/NMEA_Gateway](https://github.com/nicklasb/NMEA_Gateway)

Note that the project itself has ESP-IDF-dependencies, and that this problem only occurs when I add the nmea-gateway component. If I remove it, the pio build scripts works.

I am probably doing something crazy somewhere, but I don’t get it, I have spent a lot of time trying to fix this problem now…

(Original post att PlatformIO for reference: https://community.platformio.org/t/complex-esp-idf-dependencies-seem-to-break-pio-build-script-typeerror-nonetype-object-is-not-iterable/34881/1)

valeros, 2023-09-04:
Hi @nicklasb, could you please provide a minimal project to reproduce the issue? Some of `idf_component.yml` in https://github.com/nicklasb/NMEA_Gateway contain links to private repositories (e.g. https://github.com/nicklasb/RobArch.git).

nicklasb, 2023-09-04:
> Hi @nicklasb, could you please provide a minimal project to reproduce the issue? Some of `idf_component.yml` in https://github.com/nicklasb/NMEA_Gateway contain links to private repositories (e.g. https://github.com/nicklasb/RobArch.git).

Oh, I am so sorry, I forgot I had changed to that repo, I have pushed a fix for that now and checked that the error persists!

With regards to minimal project, I know it is annoying, but that is it. 
I have tried a *lot* to make it happen with less, but for some reason I can't make it happen in another way. 

If you have any tricks or something you want me to try, please say it.

valeros, 2023-09-04:
Even though you didn't mentioned I suppose I should clone the project from the `Server_dev` branch, right? Anyway, it seems I managed to reproduce the issue, but I had to remove the `nmea-service` component because CMake complained that there is an include path missing:
```
Include directory
  '../NMEA_Gateway/components/nmea-service/include'
  is not a directory.
```
Anyway, the issue is probably caused by a typo in the [CMakeFile.txt](https://github.com/nicklasb/NMEA_Gateway/blob/Server_dev/components/nmea-gateway/CMakeLists.txt#L3) for the `nmea-gateway` component. You specified `src/*.c` in the GLOB, but there are no `.c` files, only `.cpp`.

nicklasb, 2023-09-04:
OMG that was it! Thanks!!
When I fix the nmea-service component CMakeLists.txt it builds an runs!!!
I promise I have looked at that d*mn file a thousand times!!

I couldn't understand that it was that, since it worked in ESP-IDF. 

(Actually, the nmea-service has no files at all yet, it will mostly be .c there, probably no .cpp. )

That said, Platform IO behaves differently from ESP-IDF in this case, as ESP-IDF works with the "wrong" CMakeLists.txt.
The error in the build script is a crash, and a crash that instead should have some kind of error message. 
Maybe you'll ge att PR from me on that. 

Either way, thanks a _bunch_!!

closed 2023-09-04 by nicklasb