platformio/platform-espressif32 #1184: Increase support for Arduino v2.0.12 as soon as possible

kumv-net, 2023-09-05:
Increase support for Arduino v2.0.12 as soon as possible, as it has optimized a problem that I believe is more serious

https://github.com/espressif/arduino-esp32/pull/8541
https://github.com/espressif/arduino-esp32/issues/6129

proddy, 2023-09-06:
+1 we're also waiting for 2.0.12 that fixes heap issues on an ESP32

PBrunot, 2023-09-15:
+1 but please go for 2.0.13 so that I can complete TT Display board addition

Jason2866, 2023-10-03:
v12.0.13 not recommended to use. It has a big regression in wifi client flush().
Probably a new release is coming -> 2.0.14

bdraco, 2023-10-04:
It looks like 2.0.14 is out https://github.com/espressif/arduino-esp32/releases/tag/2.0.14

I'm mostly interested in [esp-idf 4.4.6](https://github.com/espressif/esp-idf/releases/tag/v4.4.6) being listed as an optional version though

commit d8393399 (2023-12-27): Update Arduino core to v2.0.14

Resolves #1184