platformio/platform-espressif32 #1186: Request new PlatformIO Board T-Display-S3 AMOLED

awbaxter, 2023-09-06:
Request new PlatformIO Board T-Display-S3 AMOLED

valeros, 2023-09-06:
Hi @awbaxter, there is no official variant for this board, so you need to add it to this repo https://github.com/espressif/arduino-esp32 in the first place.

cosmicDustOfLightLength, 2023-09-09:
I have board file works for me very good
[lilygo-t-amoled.txt](https://github.com/platformio/platform-espressif32/files/12566663/lilygo-t-amoled.txt)

Please remember change .txt to .json put in C:\Users\"USER NAME"\.platformio\boards

And my config from platformio:

```
[env:esp32dev]
platform = espressif32
board = lilygo-t-amoled
board_build.mcu = esp32s3
framework = arduino
monitor_port = COM9
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
upload_protocol = esptool
upload_speed = 921600
build_flags = 
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
	-D LV_CONF_INCLUDE_SIMPLE
	-I src
lib_deps = 
	bodmer/TFT_eSPI@^2.5.31
	arduino-libraries/NTPClient@^3.2.1
```

awbaxter, 2023-09-10:
Thanks @cosmicDustOfLightLength . I installed your board and it is working now 👍