platformio/platform-espressif32 #1187: When I was using platformio, I found an error when I used esp32 -2.0.3. How can I solve it?

LuBaker34, 2023-09-06:
Can you help me solve it? I just want to execute my arduino program with version 2.0.3. If it is settled, I am willing to pay a certain fee. Thank you very much.
[image]
[image]

LuBaker34, 2023-09-06:
[image]
This is the content of the esp32-s3-devkitc-1-myboard file.

valeros, 2023-09-06:
Hi @LuBaker34, support for the Arduino core v2.0.3 was added in [v4.0.3](https://github.com/platformio/platform-espressif32/releases/tag/v4.3.0) of this platform, so your `platformio.ini` should look something like this (without additional `platform_packages`):
```ini
[env:esp32-s3-devkitc-1-myboard]
platform = espressif32 @ ~4.3.0
framework = arduino
board = esp32-s3-devkitc-1-myboard
...
```

closed 2023-09-06 by valeros

Jason2866, 2023-09-06:
Why you want to use core 2.0.3? There are several bugs solved in later/latest core.

LuBaker34, 2023-09-07:
> 

If I write like you said, I will use version 2.0.11, but I will use version 2.0.3, because my screen only works normally under version 2.0.3, and other versions will display anomalies. I am also surprised at this phenomenon.

LuBaker34, 2023-09-07:
> Why you want to use core 2.0.3? There are several bugs solved in later/latest core.

Because my screen only works normally in version 2.0.3, other versions will display anomalies, which is also very strange to me.