platformio/platform-espressif32 #1190: configMINIMAL_STACK_SIZE appears inadequate for ESP32-S3 in examples

peteDDD, 2023-09-08:
When running examples espidf-arduino-wifiscan and espidf-arduino-blink on ESP32-S3-DEVKITC-1, with CONFIG_AUTOSTART_ARDUINO set such that the app-main() and xTaskCreate is used, and when running even the most simple of code with one instance of xTaskCreate (see minimal code example below), the task launched by xTaskCreate will throw a stack overflow error.

I suspect the calculation of configMINIMAL_STACK_SIZE is insufficient for the ESP32-S3 processor.  I found that adding 380 words to that value was the sweet spot for minimal stack size for xTaskCreate with this processor.

minimal code example:
`
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "sdkconfig.h"
#include <Arduino.h>

void arduinoTask(void *pvParameter)
{
    Serial.begin(115200);
    delay(100);
    Serial.println("Hello World");

    while(1) {
        Serial.println("looping");  
        //delay(1000);
        vTaskDelay(1000);  // neither delay nor vTaskDelay change the minimal amount of stack needed
    }
}

extern "C" void app_main()
{
    // initialize arduino library before we start the tasks
    initArduino();

   // configMINIMAL_STACK_SIZE throws stack overflow error with ESP32-S3
   // xTaskCreate(&arduinoTask, "arduino_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);  
   
   // 380 extra words avoids stack overflow
   xTaskCreate(&arduinoTask, "arduino_task", configMINIMAL_STACK_SIZE + 380, NULL, 5, NULL); 
} 
`

valeros, 2023-09-11:
Hi @peteDDD, the stack size is a configurable option, you can run the `menuconfig` target in your project and update the value according to your needs:
[image]

closed 2023-09-11 by valeros

peteDDD, 2023-09-11:
Understood but it seems that the MINIMAL_STACK_SIZE setting, as
constructed/calculated in the definition does not take into account that
the ESP32-S3 needs more to do ANYTHING in this case and the calculation
should be reconsidered for the ESP32-S3 to actually work for the simplist
of freertos tasks.  Examples that claim to be supported by particular
processors should run, do you disagree?

Jason2866, 2023-09-11:
@peteDDD Not the correct github to discuss. It is an espressif setup decision.
Platformio is "just" the IDE to use the espressif stuff.

valeros, 2023-09-12:
> Examples that claim to be supported by particular processors should run, do you disagree?

The `espidf-arduino-wifiscan` and `espidf-arduino-blink` examples in this repository don't have any mention of ESP32-S3 targets.