platformio/platform-espressif32 #1194: Request for Board Support Package

meakashrao, 2023-09-13:
Hello All, 
I am using ESP32-WROOM-32E-N8 and till now was unknowingly using this as my .ini file for development purposes
[image]
and when I compile my code
[image]
I see the maximum Flash Size being used is 2MB approx
but N8 is supposed to give me 8Mb, I require this help because we as an org are thinking to move to N4(which will 4MB of Flash Size) and would request for help on how can I get 4Mb or 8Mb in the Program Size

Also, my program has WiFi and BLE both and 83.5% is the best after all efforts to not use any exttra data bytes, so would really like to understand how can I increase the flash size

valeros, 2023-09-13:
Hi @meakashrao , you can adjust the flash size directly in your `platformio.ini`:
```ini
[env:esp32doit-devkit-v1]
...
board_upload.flash_size = 4MB

```

meakashrao, 2023-09-15:
Tried what was said (and something else as well) still didn't work (individually or together), is there anything else that someone can suggest.

[image]

valeros, 2023-09-15:
> Tried what was said (and something else as well) still didn't work (individually or together), is there anything else that someone can suggest.

You mean the memory size usage reports only 1966080 available bytes? This size is calculated dynamically according the size of the `app0` partition in the `min_spiffs.csv` file. If you don't need the OTA functionality you can remove the `app1` partition and increase the size of the `app0` partition according to your needs.

closed 2023-09-15 by meakashrao