platformio/platform-espressif32 #1196: Add ESP32 SVD files

maxgerhardt, 2023-09-16:
Espressif has released them at https://github.com/espressif/svd/releases.

pwm802, 2023-10-12:
Has there been any activity on this?

TheHexaCube, 2024-03-20:
I'm having the same issue - included the .svd file for an ESP32-C3, yet no data is shown under the peripherals tab. 
Instead I get the error message "Cannot read properties of null (reading 'recordType')"

inF1704, 2024-10-04:
Same issue here. I tried to include the svd file but without success

maxgerhardt, 2024-10-04:
The SVD parser of the PlatformIO extension is known to be borked with certain SVDs. Probably other extensions like https://marketplace.visualstudio.com/items?itemName=mcu-debug.peripheral-viewer can handle it though.

https://github.com/platformio/platformio-vscode-ide/issues/1675

inF1704, 2024-10-04:
Any chance it gets fixed?

maxgerhardt, 2024-10-04:
Ask the devs, not me- Push the linked issue.

Jason2866, 2024-10-04:
Looks like Platformio stopped every development work regarding espressif stuff

pmendiuk, 2024-10-07:
> Looks like Platformio stopped every development work regarding espressif stuff

They stopped working on Espressif in general or just the SVD parser?

maxgerhardt, 2024-10-08:
Having a fixed SVD parser is also in the interest of other platforms, not just ESP32, so maybe it will get fixed when looking at related SVD parsing bugs.

pmendiuk, 2024-10-08:
I don't get the impression that register level debugging is a priority, at least for ESP32 users.

inF1704, 2024-10-08:
I agree with you when using overblown libraries and when the understanding of coding is copy and paste only. But if you need to build a high quality code for performance reqsons, then you build from scratch. For example if you want to interact with an i2C device without using a library you need to debug on register level. Otherwise it is only try and error and because of timings you can't do "debugging" with serial.print

pmendiuk, 2024-10-08:
@maxgerhardt Do you have any idea if this will ever get fixed/implemented?

inF1704, 2024-10-08:
Don't know what happened but I managed to establish a debug connection to an esp32s3 with internal JTAG usb debugger and parser running.

[image]
[image]

SherifMEldeeb, 2025-01-20:
Is this issue still exist as far as my knowledge it's working now by loading the mentioned svd files to the IDE?!!