platformio/platform-espressif32 #1201: Everything is trying to eat my internal ram and not my external!

zekageri, 2023-09-28:
ESP32-Wrover-E ( 8mb psram, 16mb flash )
latest master 
Platformio
Windows10

### Flash frequency
80
### PSRAM enabled
yes
### Upload speed

115200

### Description

I want to create an https connection every hour or so.
I noticed that my internal ram is not sufficent for it.
I tought that malloc will first try to allocate from psram and if it fails, it allocates from internal.
But it does not the case.

Here is how my esp is on RAM right now

```cpp
// External
[SYSTEM] - [EXT][1]
Free:   4075727
Min:    4075727
Max:    4075727
Contig: 3997684
Diff:   808

// Internal
[SYSTEM] - [HEAP][1]
Free:   78680
Min:    78680
Max:    78680
Contig: 38900
Diff:   78680
```

I have only **38900kb** of contignous ram left in my internal!! How is this possible?
My psram is working fine and i try to allocate everything on it in my sketch.
When i try to do a `HTTPClient https;` it just fails because there is not enough internal ram.

I even call ```heap_caps_malloc_extmem_enable(0);``` so everything should be allocated from external ram.

### At least the doc says this:

```cpp
void heap_caps_malloc_extmem_enable(size_t limit)
Enable malloc() in external memory and set limit below which malloc() 
attempts are placed in internal memory. When external memory is in use,
the allocation strategy is to initially try to satisfy smaller allocation requests
with internal memory and larger requests with external memory.
This sets the limit between the two, as well as generally enabling allocation in external memory.

Parameters:
limit – Limit, in bytes.
```

### Sketch

```cpp
void setup(){
    Serial.begin(115200);
    // What can i do more so things use my psram and not my internal???????
    heap_caps_malloc_extmem_enable(0);
esp_err_t error = heap_caps_register_failed_alloc_callback([](size_t requested_size, uint32_t caps, const char *function_name) {

    Serial.printf("[SYSTEM] - %s was called but failed to allocate %d bytes with 0x%X capabilities. \n",
                     function_name, requested_size, caps);

    });
}

void loop(){
}

// Try to do an https request.
```

### Debug Message

```plain
[619840][V][HTTPClient.cpp:252] beginInternal(): url: https://api.github.com/repos/{name}/{repoName}/releases/latest
[619841][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: api.github.com port: 443 url: /repos/{name}/{repoName}/releases/latest      
[619854][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 0

[619921][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 78680
[619921][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[619956][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[619957][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[619960][V][ssl_client.cpp:178] start_ssl_client(): Loading CA cert
[619971][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[619974][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...
[SYSTEM] - heap_caps_calloc was called but failed to allocate 1051 bytes with 0x804 capabilities.
[620166][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-10368) X509 - Allocation of memory failed
[620166][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -10368   
[620172][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection. 
[620179][D][HTTPClient.cpp:1163] connect(): failed connect to api.github.com:443
[620186][W][HTTPClient.cpp:1483] returnError(): error(-1): connection refused
[Updater] - Failed to check new version. HTTP CODE: -1
[620452][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[620452][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
```

### Other Steps to Reproduce

How can i force everything to use my PSRAM??
There is 4MB of psram just sitting there without any use!!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

What i have done so far:

At setup: `heap_caps_malloc_extmem_enable(64);`

In platformio.ini

```cpp
build_flags =   
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -mfix-esp32-psram-cache-strategy=memw
    -DCORE_DEBUG_LEVEL=0
    -std=gnu++17
    -DCONFIG_ARDUINO_ISR_IRAM=0
    -DCONFIG_SPIRAM_USE_MALLOC=1
```

valeros, 2023-10-02:
Hi @zekageri, this repository is not the right place for such project-specific questions, please forward to https://github.com/espressif/arduino-esp32

zekageri, 2023-10-02:
Yeah, they sent me there because of the build flags.