platformio/platform-espressif32 #1202: LittleFS does not seem to work on VSC PlatformIO for LOLIN S3 (but does on Arduino IDE)

mars000, 2023-09-29:
**reference thread:** https://community.platformio.org/t/official-littlefs-support-for-the-esp32/23817/14
Board: LOLIN S3 board
VSC - 1.82.2
PlatformIO: 6.1.11
PLATFORM: Espressif 32 (6.4.0) > WEMOS LOLIN S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash

here is the code:

```
#include <Arduino.h>
#include "FS.h"
#include <LittleFS.h>
#include <time.h>

/* You only need to format LittleFS the first time you run a
   test or else use the LITTLEFS plugin to create a partition
   https://github.com/lorol/arduino-esp32littlefs-plugin */
   
#define FORMAT_LITTLEFS_IF_FAILED true

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");

            Serial.print(file.name());
            time_t t= file.getLastWrite();
            struct tm * tmstruct = localtime(&t);
            Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n",(tmstruct->tm_year)+1900,( tmstruct->tm_mon)+1, tmstruct->tm_mday,tmstruct->tm_hour , tmstruct->tm_min, tmstruct->tm_sec);

            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");

            Serial.print(file.size());
            time_t t= file.getLastWrite();
            struct tm * tmstruct = localtime(&t);
            Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n",(tmstruct->tm_year)+1900,( tmstruct->tm_mon)+1, tmstruct->tm_mday,tmstruct->tm_hour , tmstruct->tm_min, tmstruct->tm_sec);
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

[... 305 lines cut ...]
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:cb:38:c0
Uploading stub...
```

here is the output:
```
---- Opened the serial port COM8 ----
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   168][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/littlefs/mydir/newdir2/newdir3/hello3.txt) failed
[   170][E][vfs_api.cpp:105] open(): /littlefs/mydir/newdir2/newdir3/hello3.txt does not exist, no permits for creation
[   292][E][vfs_api.cpp:29] open(): mydir does not start with /
[   328][E][vfs_api.cpp:182] remove(): /mydir/newdir2/newdir3/hello3.txt does not exists or is directory
[   367][E][vfs_api.cpp:250] rmdir(): /mydir does not exists or is a file

```

Jason2866, 2023-09-30:
Tried. Works without any problems using this platformio.ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:lolin_s3]
platform = espressif32 @ 6.4.0
board = lolin_s3
framework = arduino
monitor_speed = 115200
upload_port = COM8
board_build.filesystem = littlefs
```

mars000, 2023-10-02:
Can you pls confirm script used and output.
If it’s the same I submitted then I so stumped 🤔

Jason2866, 2023-10-02:
Used exactly the code you provided. The only difference is the platformio.ini where I pinned to platform build `6.4.0`
Looks like your setup is in general screwed up.

mars000, 2023-10-02:
@Jason2866  I concur - I made the change and it worked :-)
I really apprecaite you taking the time to check this out - I don't know why it was causing error without adding in @ 6.4.0 line explicitly, as its quite evident from my original terminal output from compilation its using 6.4.0.
Anyway lesson learn, thank you again.

closed 2023-10-02 by mars000