platformio/platform-espressif32 #1206: Error while trying to compile

capedra, 2023-10-07:
Whenever I try to compile a project with this new release of ```toolchain-xtensa-esp32 @ 12.2.0+20230208```, I get the following error:
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.1f1a9ae 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
KeyError: 'framework-arduinoespressif32-libs':
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\xyz\.platformio\platforms\espressif32\builder\main.py", line 346:
    target_elf = env.BuildProgram()
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\xyz\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 41:
    SConscript(
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xyz\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\xyz\.platformio\packages\framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6\tools\platformio-build.py", line 40:
    FRAMEWORK_LIBS_DIR = platform.get_package_dir("framework-arduinoespressif32-libs")
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\xyz\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),
```
I already deleted all the folders from packages and platforms folders, upgraded pio with --dev option, but the same error persists. The only workaround for it to work now is to modify platformio.ini file by deleting this line:

```platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git```
This way, the toolchain used is the ```espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5``` which I guess it's not the most recent one. So, could someone please help me find out a way to solve this? Until then I'll be using Arduino IDE. Thanks!

Jason2866, 2023-10-07:
There is no official Platformio support for Arduino 3.0.0 alpha1 build. You cant use "just" with providing the build with `platform_packages`
I did a setup with modified sdkconfig and stripping libs from Arduino. You can try with
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.10/platform-espressif32.zip
```

valeros, 2023-10-09:
The upstream version of the Arduino framework v3.0 is currently unsupported.

closed 2023-10-09 by valeros