platformio/platform-espressif32 #1207: KeyError: 'framework-arduinoespressif32-libs':

flyerxp, 2023-10-09:
:\Users\Administrator\.platformio\penv\Scripts\pio run -e esp32dev
Warning! Ignore unknown configuration option `framework-arduinoespressif32-libs` in section [env:esp32dev]
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0+sha.f6ec392) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.1f1a9ae 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
KeyError: 'framework-arduinoespressif32-libs':
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\espressif32@src-5f117260f75b328038ec9d3fd0e14a68\builder\main.py", line 346:
    target_elf = env.BuildProgram()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\platforms\espressif32@src-5f117260f75b328038ec9d3fd0e14a68\builder\frameworks\arduino.py", line 41:
    SConscript(
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Administrator\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Administrator\.platformio\packages\framework-arduinoespressif32\tools\platformio-build.py", line 40:
    FRAMEWORK_LIBS_DIR = platform.get_package_dir("framework-arduinoespressif32-libs")
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),
========================== [FAILED] Took 0.99 seconds ==========================

flyerxp, 2023-10-09:
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
monitor_speed = 9600
monitor_port = COM4
upload_port = COM4
board_build.partitions = huge_app.csv

Jason2866, 2023-10-09:
Arduino 3.0.0 is not (yet) supported. It is still in alpha state. There are changes in Platformio needed.

valeros, 2023-10-09:
The upstream version of the Arduino framework v3.0 is currently unsupported.

closed 2023-10-09 by valeros