platformio/platform-espressif32 #1208: esptool elf2image is missing 120 MHz flash frenquency

toine512, 2023-10-11:
esptool is supposed to be a ESP-IDF component, to be generated by cmake according to menuconfig. Here it is not thus it refuses `--flash_freq=120m`.

> usage: esptool elf2image [-h] [--output OUTPUT] [--version {1,2,3}]
                         [--min-rev-full {0, ... 65535}]
                         [--max-rev-full {0, ... 65535}] [--secure-pad]
                         [--secure-pad-v2]
                         [--elf-sha256-offset ELF_SHA256_OFFSET]
                         [--dont-append-digest] [--use_segments]
                         [--flash-mmu-page-size {64KB,32KB,16KB,8KB}]
                         [--pad-to-size PAD_TO_SIZE]
                         [--flash_freq {80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {qio,qout,dio,dout}]
                         [--flash_size {256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                         [--spi-connection SPI_CONNECTION]
                         input
esptool elf2image: error: argument --flash_freq/-ff: invalid choice: '120m' (choose from '80m', '60m', '48m', '40m', '30m', '26m', '24m', '20m', '16m', '15m', '12m')
*** [.pio\build\esp32-s3-devkitc-1-n16r8-debug\bootloader.bin] Error 2

Jason2866, 2023-10-12:
Platformio is NOT the maintainer of esptool.py. Wrong github

toine512, 2023-10-12:
It's a misconception on my side. Max (and default) frequency for flash interaction indeed is 80 MHz.
That said, cmake is supposed to force CONFIG_ESPTOOLPY_FLASHMODE (--flash_mode) to "dout" when one tries to activate 120 MHz (CONFIG_SPI_FLASH_HPM_ENABLE). This will not happen here.

closed 2023-10-12 by toine512