platformio/platform-espressif32 #1213: [BUG]PlatformIO run menuconfig dosen't work

LUCKandII, 2023-10-20:
## Environment
* PlatformIO Core, version 6.1.11
* Windows11 22H2
* VsCode PlatformIO IDE extension setting:
```js
    "platformio-ide.useBuiltinPIOCore": false,
    "platformio-ide.customPATH": "C:/Users/lucka/.platformio/penv/Scripts",
    "platformio-ide.useBuiltinPython": false,
```
* System Python version:3.12.0,installed for all user.
* Project under tested:[platform-espressif32/examples/espidf-arduino-blink/](https://github.com/platformio/platform-espressif32/tree/516520f851f843ad79131ee5eb3affb7555cf587/examples/espidf-arduino-blink)

## Description
* VSCode PlatformIO IDE extension works fine.  I can creat project,build project,import libraries.
* __The only thing that dosen't work is `run menuconfig`__ through VsCode PlatformIO IDE.  In the terminal,it seems had problems downloading `windows-curses package`.[Windows-curses](https://github.com/zephyrproject-rtos/windows-curses) dosen't support python 3.12.0 indeed yet.
* I meet this `Installing windows-curses package fail` error while building the project,but the project was finally built successfully.  So I don't know whether `windows-curses package` have anything to do with `run menuconfig`.

```bash
Installing windows-curses package
ERROR: Could not find a version that satisfies the requirement windows-curses (from versions: none)
ERROR: No matching distribution found for windows-curses
```

* I also installed ESP-IDF tool.  So I run `idf.py menuconfig` in the folder by `ESP-IDF 5.1 CMD`,it works well.  So I though it might be a `BUG` of platformio,and I paste the error log below.

__Anyway, thank all your jobs in platformio.It's wonderful embedded environment!__

[image]

```bash

Exception:
ModuleNotFoundError: No module named '_curses'
FAILED: CMakeFiles/menuconfig
cmd.exe /C "cd /D C:\Files\Working\ESP32_display_demo\PIO\platform-espressif32\examples\espidf-arduino-blink\.pio\build\esp32dev && C:\Users\lucka\.platformio\penv\.espidf-4.4.5\Scripts\python.exe C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/tools/kconfig_new/prepare_kconfig_files.py --env-file C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/config.env && C:\Users\lucka\.platformio\penv\.espidf-4.4.5\Scripts\python.exe C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/tools/kconfig_new/confgen.py --kconfig C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/Kconfig --sdkconfig-rename C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/sdkconfig.rename --config C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.esp32dev --defaults C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.defaults --env-file C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.esp32dev && C:\Users\lucka\.platformio\penv\.espidf-4.4.5\Scripts\python.exe C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/tools/check_term.py && C:\Users\lucka\.platformio\packages\tool-cmake\bin\cmake.exe -E env COMPONENT_KCONFIGS_SOURCE_FILE=C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/kconfigs_projbuild.in IDF_CMAKE=y KCONFIG_CONFIG=C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.esp32dev IDF_TARGET=esp32 IDF_ENV_FPGA= C:/Users/lucka/.platformio/penv/.espidf-4.4.5/Scripts/python.exe -m menuconfig C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/Kconfig && C:\Users\lucka\.platformio\penv\.espidf-4.4.5\Scripts\python.exe C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/tools/kconfig_new/confgen.py --kconfig C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/Kconfig --sdkconfig-rename C:/Users/lucka/.platformio/packages/framework-espidf@3.40405.230623/sdkconfig.rename --config C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.esp32dev --defaults C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.defaults --env-file C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/.pio/build/esp32dev/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config C:/Files/Working/ESP32_display_demo/PIO/platform-espressif32/examples/espidf-arduino-blink/sdkconfig.esp32dev"
ninja: build stopped: subcommand failed.
Error: Couldn't execute 'menuconfig' target.
scons: *** [menuconfig] Explicit exit, status 1
```

valeros, 2023-10-23:
Hi @LUCKandII, apparently, the only available workaround is downgrading your Python interpreter to 3.11 until the `windows-curses` package for 3.12 is published. Besides, you're comparing the native ESP-IDF environment for v5.1 and PlatformIO integration for 4.4.5 which is not entirely correct.

LUCKandII, 2023-10-25:
@valeros Thanks! It's quiet apparently indeed. I downgrading my python to 3.11 and it works well. But I am still confuse at the below question:
* I've tried  `"platformio-ide.useBuiltinPython": true,` but still met the same problem(at that time my system python version was still 3.12). I thought platformio will automatically download the appropriate python version as a `builtin/portable python`,but it seems still ues the system python interpreter?

Let's say I just recently started using platformio, I would naturally download the latest python and then start to install pio.After I get pio installed and start to build a project, I will encounter this error. The only way to solve is to downgrading python at this time because I didn't know it when I installed this software before. This is a bit inelegant...This is just a small complaint...

valeros, 2023-10-26:
>  I thought platformio will automatically download the appropriate python version as a builtin/portable python,but it seems still uses the system python interpreter?

It should, try disabling the `platformio-ide.useBuiltinPython` and `platformio-ide.useBuiltinPIOCore` options and restart VS Code. 

>  Let's say I just recently started using platformio, I would naturally download the latest python and then start to install pio.

Unfortunately, we have no control over the `windows-curses` package to keep it up-to-date with the latest Python updates. Besides, you don't need to install Python or PlatformIO Core, they will be installed automatically with the PlatformIO IDE plugin and with all default PlatformIO settings (built-in Python, built-in PlatformIO Core) everything will work out of the box. Once you choose to modify the default PlatformIO configuration according to your needs, you are on your own.

LUCKandII, 2023-10-27:
> It should, try disabling the `platformio-ide.useBuiltinPython` and `platformio-ide.useBuiltinPIOCore` options and restart VS Code.

Thanks! I tried this option and platformio IDE started to download something. Due to the poor network environment, accessing the platformio official site by PIO IDE is very slow and usually failed to download. That's why I installed PIO core mamually ahead and modify the default PlatformIO configuration.In this case,I'm not to blame for the platformio IDE's problems.

Anyway, thank you very much for your reply!

closed 2023-10-27 by LUCKandII