platformio/platform-espressif32 #1220: ESP32-S3

HamzaHajeir, 2023-10-28:
Hi there

I'm facing an error with debugging in PlatformIO.

I have an ESP32-S3 that runs Arduino 3.0.0 (custom build) with an espressif32 platform presented by Jason2866 for supporting ESP-IDF V5.1 builds.

I've went through multiple threads talking about esp32-s3 builtin debugging, but couldn't find a similar issue except [this one](https://community.platformio.org/t/how-to-configure-debugging-esp32s3/35556).

I believe the issue occurs at some point before connecting to the on-board builtin debugger (which is reachable by the USB terminal of the development board); because it also can't upload the code. (pio run -t upload) exits with an error similar to the debug console output.

`platformio.ini`:
```
[env:esp32s3-devkitc]
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/HamzaHajeir/arduino-esp32#lwip-tls-3.0.0-rc2
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
build_flags = -DBOARD_HAS_PSRAM

debug_tool = esp-builtin
build_type = debug
debug_speed = 12000
debug_port = COM3
upload_port = COM3
; debug_load_mode = manual

upload_protocol = esp-builtin
debug_init_break = tbreak setup
```
Debug console output;
```
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: esp_usb_jtag: could not find or open device!

Reading symbols from C:\Users\Hamza\Documents\PlatformIO\Projects\ESP32S3Test\.pio\build\esp32doit-devkit-v1\firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
```

The package list of the test project:
```
pio pkg list
Resolving esp32s3-devkitc dependencies...
Platform espressif32 @ 2023.10.11+sha.ce0c9ae (required: git+https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5)
├── framework-arduinoespressif32 @ 3.0.0+sha.d94e4cc (required: git+https://github.com/HamzaHajeir/arduino-esp32#lwip-tls-3.0.0-rc2)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-dfuutil-arduino @ 1.11.0 (required: platformio/tool-dfuutil-arduino @ ~1.11.0)   
├── tool-esptoolpy @ 1.40700.0+dev2 (required: https://github.com/tasmota/esptool/releases/download/v4.7.0-dev2/esptool-4.7.0.zip)
├── tool-idf @ 1.0.1 (required: platformio/tool-idf @ ~1.0.1)
├── tool-mconf @ 1.4060000.20190628 (required: platformio/tool-mconf @ ~1.4060000.0)      
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.9.0 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 (required: espressif/tool-riscv32-esp-elf-gdb @ ~11.2.0)
├── tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 (required: espressif/tool-xtensa-esp-elf-gdb @ ~11.2.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 12.2.0+20230208 (required: espressif/toolchain-riscv32-esp @ 12.2.0+20230208)
└── toolchain-xtensa-esp32s3 @ 12.2.0+20230208 (required: espressif/toolchain-xtensa-esp32s3 @ 12.2.0+20230208)
```

Platform System Info:
```
pio system info
--------------------------  -----------------------------------------------------------------------------
PlatformIO Core             6.1.11
Python                      3.11.2-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Hamza\.platformio
PlatformIO Core Executable  C:\Users\Hamza\AppData\Local\Programs\Python\Python311\Scripts\platformio.exe
Python Executable           C:\Users\Hamza\AppData\Local\Programs\Python\Python311\python.exe
Global Libraries            0
Development Platforms       5
Tools & Toolchains          31
--------------------------  -----------------------------------------------------------------------------
```

Note that I've Windows 11.

Is there something I've missed along the setup?

With thanks,
Hamza Hajeir

Jason2866, 2023-10-28:
Windows is a pain for debugging... Looks like you have to install a driver
```
error:%20libusb_open()%20failed%20with%20LIBUSB_ERROR_NOT_SUPPORTED%0AError%3A%20libusb_open()%20failed%20with%20LIBUSB_ERROR_NOT_FOUND
```

HamzaHajeir, 2023-10-28:
> Windows is a pain for debugging... Looks like you have to install a driver called libusb
> 
> ```
> error:%20libusb_open()%20failed%20with%20LIBUSB_ERROR_NOT_SUPPORTED%0AError%3A%20libusb_open()%20failed%20with%20LIBUSB_ERROR_NOT_FOUND
> ```

I've downloaded zadiq and installed it, but nothing changed.. (Should I select the USB device and "Replace" it?)

> and you are not using my latest setup since you have installed an old gdb version. My fork does install
> 
> ```
> tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
> ```
> 
> Since you use not the version from here do not expect support. There is always the chance the fork mess up the functions.

It's actually what the `platform.py` script is doing:
https://github.com/Jason2866/platform-espressif32/blob/cc863f0e51f4ee3a2082d49f73ad129c30d1a1ff/platform.py#L60-L69

And here's 2 problems:
- It doesn't install for `arduino` framework (only for espidf).
- It specifies the version (11.2) for Windows.

Jason2866, 2023-10-28:
Ah, you are right v11 is installed for Windows that's correct. And not installing GDB for Arduino (done like here). So it needs adding in pio setup like you did.
No idea what's missing. Debugging does work on my Mac.

HamzaHajeir, 2023-10-29:
> Ah, you are right v11 is installed for Windows that's correct. And not installing GDB for Arduino (done like here). So it needs adding in pio setup like you did. No idea what's missing. Debugging does work on my Mac.

Hi

I've specified the the toolchain version so `pio pkg list` does output:
```
Resolving esp32s3-devkitc dependencies...
Platform espressif32 @ 2023.10.12+sha.cc863f0 (required: git+https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5)
├── framework-arduinoespressif32 @ 3.0.0+sha.d94e4cc (required: git+https://github.com/HamzaHajeir/arduino-esp32#lwip-tls-3.0.0-rc2)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-dfuutil-arduino @ 1.11.0 (required: platformio/tool-dfuutil-arduino @ ~1.11.0)
├── tool-esptoolpy @ 1.40700.0+dev2 (required: https://github.com/tasmota/esptool/releases/download/v4.7.0-dev2/esptool-4.7.0.zip)
├── tool-idf @ 1.0.1 (required: platformio/tool-idf @ ~1.0.1)
├── tool-mconf @ 1.4060000.20190628 (required: platformio/tool-mconf @ ~1.4060000.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 3.2.0 (required: tasmota/tool-mklittlefs @ ^3.2.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.9.0 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 (required: espressif/tool-xtensa-esp-elf-gdb @ ^12.1.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 12.2.0+20230208 (required: espressif/toolchain-riscv32-esp @ 12.2.0+20230208)
└── toolchain-xtensa-esp32s3 @ 12.2.0+20230208 (required: espressif/toolchain-xtensa-esp32s3 @ ^12.2.0)
```

And now after replacing USB driver it does upload, but still doesn't debug, as it stuck at some common point.

Iterating with USB drivers installations over Interface 2, all yielded this output:

```
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (68:B6:B3:52:B1:60)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Reading symbols from C:\Users\Hamza\Documents\PlatformIO\Projects\ESP32S3Test\.pio\build\esp32s3-devkitc\firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.

```

Any hint?

Jason2866, 2023-10-29:
Upload and debug does often not work. Use esptool.py for upload.
Use `debug without uplpad`

HamzaHajeir, 2023-10-29:
I'm already using that ...

Upload works, but debug (without uploading) doesn't.

valeros, 2023-10-30:
Hi @HamzaHajeir, the log doesn't reveal anything suspcious, do you mind sharing here your latest `platformio.ini`. Also, what's your hardware? Is it an ordinary `esp32-s3-devkitc-1` board?

HamzaHajeir, 2023-10-30:
> Hi @HamzaHajeir, the log doesn't reveal anything suspcious, do you mind sharing here your latest `platformio.ini`. Also, what's your hardware? Is it an ordinary `esp32-s3-devkitc-1` board?

Hi @valeros, ofcourse!

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32s3-devkitc]
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/HamzaHajeir/arduino-esp32#lwip-tls-3.0.0-rc2
    tool-xtensa-esp-elf-gdb@^12.1.0
    espressif/toolchain-xtensa-esp32s3@^12.2.0
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
build_flags = -DBOARD_HAS_PSRAM

debug_tool = esp-builtin
build_type = debug
debug_speed = 12000
debug_port = COM8
; upload_port = COM5
upload_port = COM9
; debug_load_mode = manual

upload_protocol = esp-builtin
; debug_init_break = tbreak setup
```

The board is an official esp32-s3-devkitc-1 by espressif, I've 2 of them, both failed.

Jason2866, 2023-10-30:
@HamzaHajeir It has to be your setup. Just tried with Tasmota build with my Arduino Core 3.0.0
```
Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4200C058, debug_reason=00000001
[esp32s3.cpu1] Target halted, PC=0x4200C058, debug_reason=00000001
Info : Detected FreeRTOS version: (10.4.3)
[New Thread 1070506736]
[New Thread 1070553556]
[New Thread 1070550820]
[New Thread 1070535968]
[New Thread 1070557048]
[New Thread 1070543152]
[New Thread 1070534572]
[New Thread 1070548084]
[Switching to Thread 1070506736]

Thread 2 "loopTask" hit Temporary breakpoint 1, setup () at /Users/hans/Git/Tasmota/tasmota/tasmota.ino:453
453	void setup(void) {
```
Works when using PIO Debug (without uploading). So my framework is okay.

HamzaHajeir, 2023-10-30:
Any hint which part of the setup? Given the platform difference (Windows vs MAC).

Does my configuration work in your machine?

valeros, 2023-10-30:
@HamzaHajeir What's the purpose of `debug_port = COM8` and `upload_port = COM9` if you're using `esp-builtin` as both the upload and debug tools?

Also, I'm wondering if you could try to debug a simple example but with the default packages and platform from this repository?
```ini
[env:esp32s3-devkitc]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
...
```

HamzaHajeir, 2023-10-30:
> @HamzaHajeir What's the purpose of `debug_port = COM8` and `upload_port = COM9` if you're using `esp-builtin` as both the upload and debug tools?
>
> Also, I'm wondering if you could try to debug a [...]

Hi @valeros 

Not specifying debug_port yields this error:
```
Reading symbols from C:\Users\Hamza\Documents\PlatformIO\Projects\ESP32S3Test\.pio\build\esp32s3-devkitc\firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
.pioinit:11: Error in sourced command file:
error starting child process '| "C:/Users/Hamza/.platformio/packages/tool-openocd-esp32/bin/openocd.exe" -s "C:/Users/Hamza/.platformio/packages/tool-openocd-esp32" -c "gdb_port pipe; tcl_port disabled; telnet_port disabled" -s "C:/Users/Hamza/.platformio/packages/tool-openocd-esp32/share/openocd/scripts" -f "interface/esp_usb_jtag.cfg" -f "target/esp32s3.cfg"': CreateProcess: No such file or directory
```

By specifying it (with the simple packages and platform), it stuck where it stucks usually:
```
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (68:B6:B3:52:B1:6C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Reading symbols from C:\Users\Hamza\Documents\PlatformIO\Projects\ESP32S3Test\.pio\build\esp32s3-devkitc\firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
```

platformio.ini of both logs:
```ini
[env:esp32s3-devkitc]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
debug_tool = esp-builtin
debug_port = COM8
; debug_load_mode = manual
upload_protocol = esp-builtin
debug_init_break = tbreak setup
```

Upload does work, debug stuck in that part.

Also tried to uncomment `debug_load_mode` and do a two separate steps: upload and debug (without uploading). Same results.

valeros, 2023-10-30:
> error starting child process '| "C:/Users/Hamza/.platformio/packages/tool-openocd-esp32/bin/openocd.exe" -s "C:/Users/Hamza/.platformio/packages/tool-openocd-esp32" -c "gdb_port pipe; tcl_port [...]

This error message is actually much more helpful, so please don't use `debug_port = COM8` with `debug_tool = esp-builtin` to hide it. There is a bug in GDB v12 on Windows, try downgrading your GDB package to v11.2.0:
```
platform_packages =
    tool-xtensa-esp-elf-gdb@~11.2.0
```
Also please share here the build log with all used packages and their versions.

HamzaHajeir, 2023-10-30:
> This error message is actually much more helpful, so please don't use `debug_port = COM8` with `debug_tool = esp-builtin` to hide it. There is a bug in GDB v12 on Windows, try downgrading your GDB [...]

Great progress! just one issue left (as I can see).

Build log:
```
Processing esp32s3-devkitc (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2023.10.12+sha.cc863f0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0)
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 23 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio\build\esp32s3-devkitc\firmware.elf
Checking size .pio\build\esp32s3-devkitc\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.4% (used 11136 bytes from 327680 bytes)
Flash: [          ]   4.9% (used 164249 bytes from 3342336 bytes)
```

Debug Console log:
```
Reading symbols from C:\Users\Hamza\Documents\PlatformIO\Projects\ESP32S3Test\.pio\build\esp32s3-devkitc\firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (68:B6:B3:52:B1:6C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x40378A0E, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40378A0E, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 12 KB
[... 231 lines cut ...]
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Temporary breakpoint 1 at 0x42001087: file src/main.cpp, line 5.
Note: automatically using hardware breakpoints for read-only addresses.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak setup`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
```

Wherein it doesn't really halt/stop on any breakpoint (setup/[another on line 6])

valeros, 2023-10-30:
What happens if you remove `debug_init_break = tbreak setup`?

HamzaHajeir, 2023-10-30:
It goes to something like an infinite loop (after somepoint):
```
Processing esp32s3-devkitc (platform: espressif32; board: esp32-s3-devkitc-1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2023.10.12+sha.cc863f0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0)
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 23 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Compiling .pio\build\esp32s3-devkitc\src\main.cpp.o
Building .pio\build\esp32s3-devkitc\bootloader.bin
Generating partitions .pio\build\esp32s3-devkitc\partitions.bin
esptool.py v4.7-dev
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\chip-debug-report.cpp.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-i2c-slave.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-periman.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32s3-devkitc\FrameworkArduino\esp32-hal-tinyusb.c.o
[... 180 lines cut ...]
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
...
```

The part which repeats indifinitely:
```
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
```

valeros, 2023-10-30:
Could you also try to erase the entire flash memory via the `pio run -t erase` command in the IDE terminal (note: serial port will be used, so you need to connect it as well)?

HamzaHajeir, 2023-10-30:
Thank you very much @valeros, That worked well!

Now it's time to actual debugging xD ;)

closed 2023-10-30 by HamzaHajeir

Jason2866, 2023-10-30:
Debug with upload does NOT work. If you have uploaded successfully with jtag stop this task.
Now use Debug without upload. Doing this way it always works for me

ostropik, 2025-03-30:
Dears, I have the same. Could You please share your final Platformio configuration and short keypoint on which I should pay attention.