platformio/platform-espressif32 #1224: Debugging on ESP32-C6 not possible

maxgerhardt, 2023-11-01:
Per https://community.platformio.org/t/cannot-run-debugger-on-esp32-c6/36473.

Currently there seem to be at least the following issues for debugging ESP32-C6 based boards:
1. Outdated Espressif/OpenOCD version used
2. Board definitions missing the declaration of the board file (`"debug": { "openocd_board": "esp32c6-builtin.cfg" }`)
3. Missing GDB setup commands as posted in [here](https://www.esp32.com/viewtopic.php?f=12&t=36501&sid=c7f0bae5598bc5921283c3b280b4b5ba#p122697)

valeros, 2023-11-01:
Thanks for reporting, indeed an updated OpenOCD v0.12 is required to debug ESP32-C6, sadly there is an issue with this version which prevents us from updating this package at the moment.

Jason2866, 2023-11-01:
Espressif updated there 12.x OpenOCD version (again) which is used in IDF5.1
This one https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20230921 is used now