platformio/platform-espressif32 #1225: Support Arduino ESP32 v3.0 based on ESP-IDF v5.1

ivankravets, 2023-11-04:


ivankravets, 2023-11-10:
See the updated comment https://github.com/platformio/platform-espressif32/issues/1225#issue-1977204439

Jason2866, 2023-11-10:
Adding Arduino 3.0.0 support is the only way to make the ESP32-C6 available for Arduino projects,

Jason2866, 2023-11-10:
> If you're an Espressif user who relies on PlatformIO for your projects, kindly share your thoughts in the comments below and cast your vote on this matter. Once we gather enough interest from the Espressif community, we plan to reach out to the Espressif company and request a reconsideration of support for PlatformIO.

Does this mean that Espressif has dropped any support? Looking in current Arduino 3.0.0 code (branch master) everything regarding Platformio support is there.

VojtechBartoska, 2023-11-10:
Hello all,

I do work in Arduino Core ESP32 project as a Project Manager so I would like to clarify this from our Team point of view.

Please take a look [on this comment in original Pull request](https://github.com/espressif/arduino-esp32/pull/8606#issuecomment-1805781410).

We also update a warning note in our documentation which should be a bit confusing.

To not duplicate what is written in original PR, we will be glad if PlatformIO supports our latest version and we are open to any contributions.

@ivankravets Feel free to contact us any time directly, we will be happy to discuss this if you see any points we can do differently.

Thanks

0x0fe, 2023-11-22:
> Adding Arduino 3.0.0 support is the only way to make the ESP32-C6 available for Arduino projects.
> The needed changes are not that many, i have added Arduino 3.0.x alpha support in my Fork and bulding Arduino and IDF projects with the C6 are working.
> We have ported Arduino project Tasmota to the C6 and it runs already surprisingly stable.

@Jason2866 
Do you know how i can setup the .ini file in a new project for using your 3.0.0 fork and 5.1 IDF at the same time (to keep the SDK menuConfig)?

0x0fe, 2023-11-22:
@ivankravets 
there is something i dont understand, why is there a separate platformespressif32 core? It is based on arduinoespressif32, but  always lags behind, and arduinoesp32 which already lags behind IDF. What is the point of it? Why is PIO not using directly arduinoesp32 core as is directly?

I think a reason why many users use PIO is also to keep some of the arduinoesp32 APIs while having the possibility to modify the IDF SDK configuration (menuconfig) which is necessary in many cases and not possible in arduino, unless using the libbuilder which has its own set of problems and is not really convenien to use, especially if we need to often modify SDK options.

Maybe it would be wise for PIO to directly use arduinoesp32 core as is, and just focusing on allowing users to user arduinoesp32 core along with IDF to keep access to menuconfig.

mhaberler, 2023-11-23:
I am concerned about the viability of this repo long-term as folks like @Jason2866 have started to roll their own and this repo not catching up

I am currently using his https://github.com/tasmota/platform-espressif32/tree/Arduino/IDF5 branch and it works mostly great for me - even if I feel uneasy about relying on a private effort with unclear future - how do you see the longer-term perspective of your fork? 

This fork has diverged from this repo massively - not sure this can brought back here easily 

the other issue I have is the abysmal debugging experience esp32/pio/arduino compared to using straight IDF

0x0fe, 2023-11-23:
@mhaberler i did the same yesterday and used https://github.com/tasmota/platform-espressif32/tree/Arduino/IDF5

```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha2
framework = 
	arduino
	espidf
board = esp32dev
```

But then when building it complained of missing SPIFFS, i added it manually from previous version (in components directory) but then i get a dependancy error with esp_partition.h, despite it is well present in "components". 

```
   17 | #include "esp_partition.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp32dev\tcp_transport\transport.o
Compiling .pio\build\esp32dev\tcp_transport\transport_ssl.o
Compiling .pio\build\esp32dev\tcp_transport\transport_internal.o
*** [.pio\build\esp32dev\spiffs\esp_spiffs.o] Error 1
*** [.pio\build\esp32dev\spiffs\spiffs_api.o] Error 1
```

Did you met this issue too?

Jason2866, 2023-11-23:
@0x0fe @mhaberler  My fork has not the goal to be compatible to anything official.
The fork is made for project Tasmota. Everything is optimized for that use. The sdkconfig settings differs a lot from the official espressif ones. There are components added and others are removed (for example SPIFFS).
Breaking changes can and will happen.

0x0fe, 2023-11-23:
@Jason2866 oh, i see, that explains the error with SPIFFS

Jason2866, 2023-11-23:
@mhaberler Does debugging work in espressif IDF env setup? If yes imho Platformio has the job for the devices which are currently supported with IDF 5.1. Not saying it will be easy to solve or it will be done soon. The debug issue for the C6 is known and there is an open issue.

mhaberler, 2023-11-23:
> Did you met this issue too?

@0x0fe  yes, same here `Failed to resolve component 'spiffs'

mhaberler, 2023-11-23:
> @mhaberler Does debugging work in espressif IDF env setup? If yes imho Platformio has the job for the devices which are currently supported with IDF 5.1. Not saying it will be easy to solve or it will be done soon. The debug issue for the C6 is known and there is an open issue.

you mean like command line/shell? yes, repeatably

ivankravets, 2023-11-23:
> I'm a bit confused. Why is there a separate platform-espressif32 core? It seems to be based on arduino-espressif32, but it always seems to lag behind...

https://github.com/platformio/platform-espressif32 isn't a core itself. It's a software piece that makes the PlatformIO Build API work seamlessly with Espressif 32 products within the PlatformIO ecosystem. If you check the source code of this repository, you'll find other integrations, like ESP-IDF. If we support Zephyr/ESP32, it'll also be found here.

> Would it be smarter for PIO to directly use the arduinoesp32 core as it is?

We do use it directly (https://registry.platformio.org/tools/platformio/framework-arduinoespressif32). There's no common ground between Arduino IDE and PlatformIO. The "arduino" in PlatformIO is just a framework. PlatformIO supports over 10+ different frameworks. The reason many developers see PlatformIO as a "replacement for Arduino IDE" is due to source code compatibility and advanced project management with lots of configuration and collaboration features based on PlatformIO's unique declarative principles.

If you've tried another framework besides "arduino", you'd notice there's no difference in the configuration aspect. You and your team can focus on project development without wrestling with bootstrapping and collaboration issues.

Lastly, for a VSCode-based solution with native ESP32/Arduino core support, I recommend the [Arduino IDE for VSCode](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) extension backed by Microsoft. It's similar to Arduino IDE 2.0 but uses VSCode as the editor. Both leverage "arduino-cli", so the build system and workflows are the same.

0x0fe, 2023-11-23:
@ivankravets  I see, glue logic layer. the totally different versioning between these platforms and the cores they implement is rather annoying, though.

mhaberler, 2023-11-23:
@ivankravets make me understand what your recommendation means - are you saying:

When I prefer to use VSCode, I should take the Microsoft Arduino extension for VSCode and drop Platformio?

ivankravets, 2023-11-23:
@mhaberler, we value openness in our community and aim to give you the freedom to choose. If there's a newer version of the software not yet supported by PlatformIO, we won't hide it.

The current supported version is Arduino Core v2.x for ESP32. Our collaboration with Espressif, including discussions about renewal, is ongoing. It's worth noting that we have @VojtechBartoska, a project manager from Espressif, in this thread. We're all working together to ensure you receive the best features and support. We'll keep everyone posted on any updates to ensure a smooth continuation of our services.

hoeken, 2023-11-25:
I can see there's a bit of political project stuff happening, but as an end-user, all I want to do is be able to use the latest Arduino framework.  Consider this a vote of support for making that happen, however it needs to be.

ivankravets, 2023-11-25:
PlatformIO is a commercial open-source project. In the past, it used to be a [paid service before 2020](https://community.platformio.org/t/platformio-plus-goes-open-source-improving-embedded-development-community-worldwide/8240), following a business-to-consumer (B2C) model. Unexpectedly, PlatformIO gained widespread popularity among millions of developers globally. Consequently, we shifted our strategy to make powerful tools for professional embedded development freely accessible to everyone.

The active development and maintenance of PlatformIO, along with its infrastructure, are now supported by technology partners dedicated to delivering an excellent developer experience. [Espressif was one such partner](https://piolabs.com/blog/press-releases/espressif-leverages-platformio-labs-technology.html), and we appreciate their long-standing collaboration.

Currently, Espressif has ceased support for new products in PlatformIO, but rest assured, we are committed to providing support for existing Espressif products integrated before this change, as per our technology licensing policy. Your projects won't face disruptions, and services will continue as usual.

If there's significant interest from Espressif customers in PlatformIO, we'll reach out to Espressif and explore the possibility of continuing our collaboration.

If you have any questions or need clarification, feel free to ask. We're here to provide all the answers you need.

mhaberler, 2023-11-28:
ok, so Espressif jumped ship 

@ivankravets  what does that mean for the subject topic - Arduino3? 

your wording would suggest to me "no new Espressif releases - be it Arduino or ESP-IDF, existing releases go on life support" 

are you saying this repo stops at ESP-IDF 5.1.1/Arduino2 until Espressif changes its mind ?

Jason2866, 2023-11-28:
@mhaberler Looks like: as long there is not enough feedback from Platformio users wanting actual espressif framework support, there is no good position from Platformio side to argument why espressif should invest money having Platformio support.
espressif is thinking supporting ArduinoIDE is enough :-(

HaenZ33, 2023-11-29:
How can we provide enough feedback?? Do we need to start a poll?
I agree with the term: Arduino-IDE toy...

maciekish, 2023-12-16:
Any news on this please? Still waiting for C6 support :<

Jason2866, 2023-12-16:
@maciekish No, there are two reasons. 
- Core 3.0.0 is not released. Platformio did not support not released cores in the past. So why it should happen now?
- espressif canceled partnership with platformio

maciekish, 2023-12-16:
> @maciekish No, there are two reasons.
> 
> * Core 3.0.0 is not released. Platformio did not support not released cores in the past. So why it should happen now?
> * espressif canceled partnership with platformio, so no support for something new

Well, it **looked** [released](https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4). I didn't dig that deep.

Is a partnership necessary? Can't the community figure this out?

Jason2866, 2023-12-16:
I roll my own fork, but this will probably not merged here, since why should Platformio do that? PlatformIO is a company and the people working there needs to get paid. Either the users do or companys who wants to see there products supported from Platformio.

richievos, 2023-12-16:
I just went to my hardware closet and pulled out a dev esp32-h2 board I'd been saving. Weeks back I'd see comments about a new release of arduino/express frameworks with support for this, and I'd been waiting for a time slot to finally move forward with my Matter+Thread project. In researching it I now came across this thread.

I'm _gutted_ to see the contents of this thread. I don't honestly know what to think in regards to either platformio or esp boards. The arduino IDE is not compatible with software development, and VSCode + PlatformIO offers a real experience and real tools. Arduino itself I could punt on, but there's so many libraries out there which depend on it, that completely going native ESP32 basically means roll-from-scratch everything. ESP boards are almost the standard IMO at this point, trying to find a different board set with a good form factor and low price is a huge momentum killer.

Together the 3 of these things fit together making IoT projects almost trivial. Losing parts of each of them basically means you either do minor toy projects or you have to be a sophisticated embedded software company with its own library stack.

_Mega, huge, downer_.

philm001, 2023-12-28:
Hey guys,

I am currently working with the W5500 chips and using it in my application with the ESP32. My IDE is platform.io

It looks like that you have a ETH library that can be used to communicate with the chip. However, it is only available with the 3.0 version.

When I try to clone the commit that has the ETH library for the W5500 chip, I get the same errors as above. I have tried the latest development version of the platform to no avail.

What are my options to get this issue solved? My project for my client is hinging on using this chip as the PCB is in production. I don't want to use Arduino IDE for obvious reasons.

Should I fork a copy of v2.14 and manually add in the new ETH code? or has someone figured out a way to get it to work with the latest platform version?

philm001, 2023-12-28:
Also, I read through the espressif comments here:
https://github.com/espressif/arduino-esp32/pull/8606

It looks like that they are not dropping support per say as they never officially supported platform.io but rather updating the language of the docs to say that they don't officially support it.

Jason2866, 2023-12-28:
@philm001 You should read the comment from Ivan Kravets in this thread (again). Clearly telling espressif canceled Partnership support. So no support for NEW stuff.
About PHY W5500 support, only possible without conflicting with SPI with upcoming Arduino 3.0.0. If you can't wait use IDF 5.1.2. This version is supported from Platformio and does support the W5500.

Or do your own fork of Platformio for supporting Arduino 3.0.0.

philm001, 2023-12-28:
@Jason2866 Hello, thank you for information. Can I get a link to IDF 5.1.2?

Jason2866, 2023-12-28:
Use
```
platform = espressif32 @ 6.5.0
framework = espidf
```
```

pitlochry, 2023-12-29:
Please request a reconsideration of support for PlatformIO. Otherwise I hope that there will emerge better opportunities in the future.

philm001, 2024-01-02:
Hey guys, pardon my ignorance but what would it take for us to add in the compatibility ourselves?

Looking through the previouse releases, it appears that you need to add the SDK source files just like what you did in version 2.0.14. From what I can tell, everything is there. What else is missing?

philm001, 2024-01-02:
For my application, I basically, updated the SDK Ethernet library to the most recent one (along with updating other needed files) and I was able to get everything the compile with the updated Ethernet library. 

Now, I am getting some other SPI errors and the MCU crashes but that is a separate issue that I am working on.

Also, how are you guys testing the development of the alpha versions? I am assuming that there is some basic Arduino tests that you are running in Platform.IO?

Jason2866, 2024-01-02:
@philm001 Forked the needed repos -> Arduino Lib Builder, Platformio Platform espressif32
and added changes to get it working for the needs of project Tasmota
This means settings in Arduino Lib Builder are different and stripped stuff we don't need / use. The main breaking change is the remove of SPIFFS. Using LittleFS as better alternative.
The actual Arduino master branch can be used with Platformio with this
```
platform = https://github.com/Jason2866/platform-espressif32.git
```
CI test are done too, and can be seen here https://github.com/Jason2866/platform-espressif32/actions
Running the examples from folder `examples`

philm001, 2024-01-02:
@Jason2866 Thanks Jason, I will explore this for my own application. However, my application utilizes the SPIFFS. What would it take to get SPIFFS to work with this Tasmota branch?

Jason2866, 2024-01-02:
> @Jason2866 Thanks Jason, I will explore this for my own application. However, my application utilizes the SPIFFS. What would it take to get SPIFFS to work with this Tasmota branch?

@philm001 You can take my Platformio Platform as boilerplate. Needed changes are in  platform.json -> Reference your framework(s). You have to compile the Arduino framework with the original IDF 5.1 from espressif and fork espressif Arduino -> Needs change(s). Use this file [tools](https://github.com/tasmota/arduino-esp32/blob/main/tools/platformio-build.py) else the framework will not be compatible with my Pio Platform.
Fork my Arduino Lib Builder https://github.com/Jason2866/esp32-arduino-lib-builder
remove in the *.sh files which does remove Libs and components you want to use!
Adjust the CMakeLists.txt file accordingly.
Add a valid package.json and version.txt (see my forks) to your IDF and Arduino fork.
If you run in errors take a look in this commit https://github.com/tasmota/arduino-esp32/commit/413362d8e250eb73d8089274f34f3d5aa081aa91
Most changes are not needed! Our (working) IPv6 implementation differs from the WiP in espressif/ Arduino so most changes are related to this. Most others are for removing stuff.
It needs some work to get it going. You have to dive into. There is no guide for and no hand holding possible from my side.

Johboh, 2024-01-02:
One alternative is to use the ESP-IDF framework directly (with their visual studio code extension), instead of platform IO. It's possible to add the arduino core as an ESP-IDF component if one has many arduino library dependencies. But for projects without that many arduino dependencies, it's usually not needed as libraries can be found in the [component registry](https://components.espressif.com/). I migrated five platform IO + arduino framework projects to use ESP-IDF directly as I needed functionally from ESP-IDF v5.2 (beta). I still use platform IO + arduino framework for simpler projects.
Porting projects between different environments is not always an option of course, but in case someone is blocked there are always solutions.

ivankravets, 2024-01-08:
@danalvarez , we're currently awaiting a response from Espressif. Due to the holiday season, there might be a slight delay in communication.

sgryphon, 2024-01-14:
> as indicated in their official statement: espressif/arduino-esp32#8606

As an outsider, this seems to have gotten way out of hand.

From what I can see:

* ESP32 Arduino has a new version coming out, Platform IO doesn't support it yet
* ESP32 Arduino put up a pull request to remove the link the Platform IO from documentation, because users were confused when it didn't work with the latest. That is the official statement -- an in progress pull request.
* The pull request was changed to keep the link, but add a disclaimer for users to contact PlatformIO.
* I don't see anything in the linked statement about cancelling partnership, but some posts by PlatformIO seem to indicate this
* PlatformIO is waiting for feedback from users they want new ESP32 Arduino (i.e. the feedback in this thread), and for contact from ESP32.

There may be some contention over who does the work to update platformio/platform-espressif32 with the necessary changes, if neither ESP32 Arduino or PlatformIO want to do the change, maybe the community can step in and submit a PR? Is that the issue, that neither side want to put time into the change?

@Jason2866 I know it will take away from your existing Tasmota work, but if you already know the changes, can you make the minimum changes to get it working (no Tasmota staff) and submit a PR?

From the outside, it looks like a bunch of over reactions, and ESP32 Arduino changed the PR. Maybe the wording "not officially supported" is irking PlatformIO, but maybe that wording can be changed? Is that the issue.

Or, as hinted by some of the PlatformIO comments, have Espressif removed financial support from PlatformIO (not sure if this is true; it isn't directly stated, but not sure what 'partnership' means... there is nothing in the link official statement that indicates this)?  If that is the case, then I guess it is outside the scope of a simple pull request.

sgryphon, 2024-01-14:
I'll note that the latest version of Arduino ESP32 has integration builds that test PlatformIO.

So, the necessary changes to the integration are already in the Arduino ESP32 build scrips: https://github.com/espressif/arduino-esp32/blob/e4d6a8abf9fab7422e21f72d3aeb2e9c9484ab36/.github/scripts/install-platformio-esp32.sh#L35

At least for the error I am getting, "KeyError: 'framework-arduinoespressif32-libs':", it looks like the Arduino ESP32 integration test scripts have a fix up to extract the URL from "package_esp32_index.template.json"

There is a similar section in platform-espressif32 where if you are using an upstream version it dynamically extracts toolchain versions from "package_esp32_index.template.json": https://github.com/platformio/platform-espressif32/blob/ee1dd5a6c65a250dc5e3031c62e8c688f2665ce6/platform.py#L60

The function is self.configure_upstream_arduino_packages() and maybe just needs to add extracting the URL and downloading 'framework-arduinoespressif32-libs' (if it is there, i.e. an older custom version of 2.0 won't have the libs change).

I'm playing around with this in a fork, and might chat to the PR author of the integration test fixes: https://github.com/espressif/arduino-esp32/pull/7733

Jason2866, 2024-01-14:
> Or, as hinted by some of the PlatformIO comments, have Espressif removed financial support from PlatformIO (not sure if this is true; it isn't directly stated, but not sure what 'partnership' [...]

@sgryphon A PR which simply adds C2 functionality to existing IDF 5.1. support is not merged
https://github.com/platformio/platform-espressif32/pull/1203
So for me it does look like any PR which enhances product support over the currently existing one is not accepted.
And for me it is clearly written espressif canceled partnership.

sgryphon, 2024-01-14:
> And for me it is clearly written espressif canceled partnership.

Maybe I am dumb or something, but I don't see any reference to partnership in the link posted, or cancelling. Maybe there is a different link?

The link referred to above as an official statement is a pull request: https://github.com/espressif/arduino-esp32/pull/8606

The PR only contains two small things, and only one of them is relevant. It doesn't mention anything about partnership or cancelling.

".. warning:: PlatformIO is not officially supported by Espressif. Please report any issues with PlatformIO directly to its developers on the official `PlatformIO repositories <https://github.com/platformio>`_."

Now, maybe you could interpret "not officially supported" as meaning you shouldn't use it... but from reading the discussion and context it means if a user wants support with PlatformIO they should contact PlatformIO, not Espressif.

This is much clearer after subsequent changes to the working, on the website installing section: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-platformio

It now reads:
"Warning Integration of the Arduino Core ESP32 project in PlatformIO is maintained by PlatformIO developers. Arduino Core ESP32 Project Team cannot support PlatformIO-specific issues. Please report these issues in official PlatformIO repositories."

So, even the wording of "not supported" has been changed to "Arduino Core ESP32 Project Tema cannot support PlatformIO-specific issues", and clarifies that the integration is maintained by PlatformIO (and I can see this repository is under the ownership of PlatformIO).

Maybe I have missed some statement that talks about cancelling partnership (a different link?), or maybe there are some deeper issues.

Jason2866, 2024-01-14:
@sgryphon Post from Ivan Kravets https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-1826363522

richievos, 2024-01-14:
+1 @sgryphon's statements. Seems like there's a play going on to try and force Espressif to pay PlatformIO some amount of money in a partnership. I can understand the justification for that.

It would clarify matters if a statement could be made on if the community can provide a solution, and have that merged/included, or if it'll only be done if something like a partnership payment is made.

0x0fe, 2024-01-14:
> Seems like there's a play going on to try and force Espressif to pay PlatformIO some amount of money in a partnership
Yes, that is what it looks like, pio wants to get a funding from espressif, rather than espressif wants to fund pio, strange business model tho. 
I can understand why espressif doesnt want to fund one more ecosystem, they already allocate ressources to support the IDF and the Arduino core. And it seems like there is enough options between IDF with vscode, Arduino core in arduino IDE, arduino core as a IDF component in VSCode, not much difference difference between using the latter and PIO. I stopped believing in "unified embedded development ecosystems" a long time ago. Using the vendor SDK and IDE saves time and hassle.

Jason2866, 2024-01-14:
@0x0fe espressif was a technology parter of Platformio and canceled this partnership. Leaving users of Platformio now standing in the rain. 
Sorry support of IDF is needed since no chip can be sold without a SDK. It is the proprietary espressif stuff. To address the DIY the decided to be Arduino compatible. But ArduinoIDE is not an IDE. No complex project can be done with. It is a toy. Using Arduino as component of IDF is everything else than productive when just wanting to use Arduino. There is NO comparable Platformio alternative which is easy to use Arduino when doing complex projects.

sgryphon, 2024-01-14:
@Jason2866 thanks for your help in getting a fix for the integration.

Although, I still don't think there is a clear statement. The post you refer to says:

"Espressif was one such partner, and we appreciate their long-standing collaboration.
Currently, Espressif has ceased support for new products in PlatformIO,"

While the partner statement is in the past tense, it doesn't say it has been cancelled; and the "ceased support for new products in PlatformIO" seems to be the same reference to providing support to users, the wording of which has been further clarified on the current Espressif website.

The comments about funding model seem to imply that Espressif used to provide funding as a partner, and maybe now don't ... although if that is the case, why not just come out and say it. (My opinion ... it probably would be a good idea for Espressif to help provide for PlatformIO, as a vibrant ecosystem is probably good for them).

There is a lot of other evidence that they are still working as partners ... they feature heavily on each other's web sites, and (I think) 5.1 is supported by the IDF platform, just not via Arduino.

sgryphon, 2024-01-14:
Anyway @ivankravets I have voted, by putting up a PR (backwards compatible) that fixes the issue and allows overrides of platform packages to ardino-esp32 alpha v3.0.0 to build, by adding the missing library (but only when it is referenced in the package, i.e. only for alpha).

https://github.com/platformio/platform-espressif32/pull/1281

Thanks @Jason2866 for pointing me in the right direction to the libraries and changes needed.

0x0fe, 2024-01-14:
@Jason2866 
> @0x0fe espressif was a technology parter of Platformio and canceled this partnership. espressif is leaving users of Platformio now standing in the rain.

I assume that by "technology partner" you mean "funder". 

>Sorry support of IDF is needed since no chip can be sold without a SDK

I worked with countless SoCs and obviously every vendor is providing an SDK and some kind of ecosystem, most vendors use "free" tools such as eclipse or vscode as an IDE, that is what espressif does, they provide official support for their sdk, with vscode, they also provide official support for the arduino ESP32 core, that is more than most vendors. 

>No complex project can be done with. It is a toy

I did not mention arduino IDE which is targetted to DIY users and not an actual tool. The official IDE for the IDF is vscode, i dont see where it is more "proprietary" than PIO which also uses vscode.
IMO the only case where pio cannot be replaced yet is for the (unsupported) case of running arduino esp32 core on top of IDF, where it is possible to modify the sdkconfig on the fly, but i suspect this can be done by adding arduino core as a component of IDF. 

>Using Arduino as component of IDF is everything else than productive when just wanting to use Arduino.

I did not have to use it, so i am not sure, but i guess there is an arduino extension for vscode, which presumably runs the precompiled IDF libraries just like Arduino IDE, but in vscode.  ( vscode-arduino )

empewoow, 2024-02-19:
Not sure if this is the right place to ask, but I thought the new ESP32 C6 would work with the Arduino platform in PlatformIO if I used the pre-release of the framework-arduinoespressif32 repo.

But when I try to add this to `platformio.ini`:

```
platform_packages =
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
```

It still says: `Error: This board doesn't support arduino framework!`

Is there a fix for this, or is it just not supported yet in this way?

Jason2866, 2024-02-19:
@empewoow You have to use your own boards.json where you add "arduino".
Something like this:
```
{
  "build": {
    "core": "esp32",
    "f_cpu": "160000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32c6",
    "variant": "esp32c6"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32c6.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif ESP32-C6-DevKitC-1",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html",
  "vendor": "Espressif"
}
```

empewoow, 2024-02-19:
@Jason2866 thanks for your message, I tried that but then I get this:

```
KeyError: 'framework-arduinoespressif32-libs':

...

  File "/home/alex/.platformio/packages/framework-arduinoespressif32/tools/platformio-build.py", line 40:
    FRAMEWORK_LIBS_DIR = platform.get_package_dir("framework-arduinoespressif32-libs")

...
```

as also described in this issue: https://github.com/espressif/arduino-esp32/discussions/8572

Maybe I just need to wait until there's official support...

Or is there some easy fix for this at the moment?

Jason2866, 2024-02-19:
@empewoow take a look at https://github.com/platformio/platform-espressif32/pull/1281

empewoow, 2024-02-19:
Thanks, it works now! FYI, I added this to get it to work:

```
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```

ajay10000, 2024-03-05:
As an implementer rather than a coder, I just like to get things done.  PlatformIO with arduinoespressif32 framework enables me to do just that.  I use PlatformIO for other types of projects/coding and it helps me a great deal to have the IDE consistency and also the great editing/support features of the platform.

Because I am not a C coder, I find everything about how these things work together extremely confusing.  I have spent days trying to work out whether ESP32-C6 could be supported or not.  I decided to try the last few comments above (from @Jason2866 and @empewoow) and was thrilled this morning to get my standard MQTT client code to compile and run on an ESP32-C6, along with voltage measurement.  

To get this to work, I have used a combination of adding a section to platform.ini:

```
[env:esp32c6]
board = esp32-c6-devkitc-1
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
monitor_speed = 115200
build_flags=
	-fexceptions
	-DESP32_STAGE
	-DESP_IDF_VERSION_MAJOR=5
	-DLIBRARIES_NO_LOG=1
	-DDISABLE_SC16IS752_SPI
	-DCONFIG_PM_ENABLE
	-DCONFIG_LWIP_L2_TO_L3_COPY
	-DCONFIG_FREERTOS_USE_TICKLESS_IDLE=1
	-DCONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP=3
	-DNEOPIXEL_ESP32_RMT_DEFAULT
	-DCRON_USE_LOCAL_TIME
	-I$PROJECT_DIR/src/include
	-include "sdkconfig.h"
```

some of these lines come from https://github.com/letscontrolit/ESPEasy/blob/mega/platformio_core_defs.ini as advised elsewhere.  I'm not sure if they are all necessary.

I also added a file: <project root>/boards/esp32-c6-devkitc-1.json as per @Jason2866's https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-1952513412 (above, a few comments up).

**A huge thank you to all**, both paid and unpaid, for putting so much effort into these projects.  I wish it was simpler to understand the versions and how they interact, but the fact that they are there in the first place is an amazing feat of human endeavour.  

I hope to try some ZigBee code next and get an endpoint up and running.  Now that would be something.

sgryphon, 2024-03-16:
Hi @ivankravets .  I'm wondering if you have had a chance to review my pull request to add Arduino ESP32 v3.0 support to PlatformIO. https://github.com/platformio/platform-espressif32/pull/1281

Thanks

(Note: This is not about Espressif supporting PlatformIO, but updating PlatformIO to support Espressif ESP32 v3.0, which was a relatively small change that I am happy to contribute to PlatformIO).

hitecSmartHome, 2024-04-26:
Well. I, and my team are very confused hearing this. What is espressif thinking? As others said, the Arduino IDE is a toy. We need a much more robust IDE. Our project grow so much in the past year using PIO with arduino,idf, we can't go back to only arduino or only idf. Maybe its time to find an other chip... :( Really sucks to stuck on idf 4.4.0 and arduino 2.0.0...

Jason2866, 2024-04-26:
We will know soon what happens. Platformio never supported not released versions.
In a few weeks espressif Arduino Core 3.0.0 will be released.
The few needed (breaking) changes in Platformio to support core 3.0.0 where I did PRs for are merged.

[276 more replies]

closed 2026-10-07 by valeros