platformio/platform-espressif32 #1226: LittleFS fails on LOLIN_S3 board if you set board = lolin_s3 but works if you set board = lolin_s3_mini in platformio.ini file

mars000, 2023-11-06:
What kind of issue is this?

I have a LOLIN S3 board.
In `platformio.ini file`, when I compile below code using `board = lolin_s3 `and it fails to run/mount LittleFS.
If I simply change `board = lolin_s3_mini` all works perferctly.

- [ ] **PlatformIO IDE**.
VSC on Windows 11

- [ ] **Development Platform or Board**.
LOLIN S3

- [ ] **PlatformIO Core**.
PlatformIO Core: 6.1.11

------------------------------------------------------------------

### Configuration

Windows 11

**PlatformIO Version** (`platformio --version`):

### Description of problem
Fails when `board = lolin_s3`
but works perfectly when `board = lolin_s3_mini`
in `platformio.ini `file
My actual board is **LOLIN_S3** for all tests.

platformio.ini file:
```
[env:lolin_s3]
platform = espressif32@6.4.0
board = lolin_s3
framework = arduino
board_build.filesystem = littlefs
upload_port = COM8
monitor_speed = 115200
```

Serial Output when **board = lolin_s3** in `platformio.ini` file
```
---- Opened the serial port COM8 ----
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   138][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/littlefs/mydir/newdir2/newdir3/hello3.txt) failed
[   140][E][vfs_api.cpp:105] open(): /littlefs/mydir/newdir2/newdir3/hello3.txt does not exist, no permits for creation
[   227][E][vfs_api.cpp:29] open(): mydir does not start with /
[   254][E][vfs_api.cpp:182] remove(): /mydir/newdir2/newdir3/hello3.txt does not exists or is directory
[   283][E][vfs_api.cpp:250] rmdir(): /mydir does not exists or is a file
```

Serial Output when **board = lolin_s3_mini** in `platformio.ini` file
```
---- Opened the serial port COM8 ----
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
[... 320 lines cut ...]
	writeFile(LittleFS, "/mydir/hello2.txt", "Hello2");
    writeFile(LittleFS, "/mydir/newdir2/newdir3/hello3.txt", "Hello3");
    writeFile2(LittleFS, "/mydir/newdir2/newdir3/hello3.txt", "Hello3");
	listDir(LittleFS, "/", 3);
	deleteFile(LittleFS, "/mydir/hello2.txt");
    deleteFile(LittleFS, "/mydir/newdir2/newdir3/hello3.txt");
    deleteFile2(LittleFS, "/mydir/newdir2/newdir3/hello3.txt");
	removeDir(LittleFS, "/mydir");
	listDir(LittleFS, "/", 3);
    writeFile(LittleFS, "/hello.txt", "Hello ");
    appendFile(LittleFS, "/hello.txt", "World!\r\n");
    readFile(LittleFS, "/hello.txt");
    renameFile(LittleFS, "/hello.txt", "/foo.txt");
    readFile(LittleFS, "/foo.txt");
    deleteFile(LittleFS, "/foo.txt");
    testFileIO(LittleFS, "/test.txt");
    deleteFile(LittleFS, "/test.txt");
	
    Serial.println( "Test complete" );
}

void loop(){

}
```

valeros, 2023-11-06:
Hi @mars000, the only obvious difference is the partition table, `lolin_s3_mini` uses the default partition while `lolin_s3` uses a special `default_16MB`. What happens if you manually set your partition to the default one directly in the `platformio.ini` file, for example:
```ini
[env:lolin_s3]
platform = espressif32@6.4.0
board = lolin_s3
framework = arduino
board_build.filesystem = littlefs
upload_port = COM8
monitor_speed = 115200

board_build.partitions = default.csv
```

mars000, 2023-11-08:
> ```ini
> board_build.partitions = default.csv
> ```

hi @valeros I updated and added above suggestion i.e. `board_build.partitions = default.csv` but no change i.e. same serial output as I placed in original post above is displayed.

valeros, 2023-11-09:
Have you tried erasing the entire flash memory? Just run `pio run -t erase` directly in the IDE terminal.

mars000, 2023-12-05:
thanks @robertlipe  for insights.
Would that imply a configurstion issue in platformio profiles ? I'm just using standard config as per docs.

mars000, 2023-12-06:
thank you @robertlipe 
@ivankravets your thoughts on this would be greatly appreciated

Jason2866, 2023-12-11:
The main difference of the two boards json are the type of the PSRAM flash.
S3 mini has QSPI and the Lolin S3 has OPI. This looks correct according to the description from the Lolin we page. Maybe in real the Lolin S3 has no OPI PSRAM? Can be verified by the mounted module on the board. Upload an highres picture and we can check.

Wrong PSRAM selection does generate Issues with Filesystem. The s3mini boards json does work since no valid PSRAM is found and it works as S3 without any PSRAM

mars000, 2023-12-12:
@Jason2866  here is hi-res picture, hope it helps answer question
[image]

Jason2866, 2023-12-12:
Mhh, the last line looks like "N16R8" regarding to the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf) it is 16MB QSPI Flash and 8MB OPI PSRAM

mars000, 2023-12-16:
hello @valeros and @Jason2866
I updated my lolin_s3.json file manually (not sure how else to best updates platfromio in vsc with your changes) with your updates as noted in https://github.com/platformio/platform-espressif32/issues/1226 but still get same errors.
Something i'm missing...this is the lolin_s3.json file

`{
"build": {
"arduino": {
"ldscript": "esp32s3_out.ld",
"partitions": "default_16MB.csv",
"memory_type": "qio_opi"
},
"core": "esp32",
"extra_flags": [
"-DARDUINO_LOLIN_S3",
"-DBOARD_HAS_PSRAM",
"-DARDUINO_USB_MODE=1",
"-DARDUINO_USB_CDC_ON_BOOT=1",
"-DARDUINO_RUNNING_CORE=1",
"-DARDUINO_EVENT_RUNNING_CORE=1"
],
"f_cpu": "240000000L",
"f_flash": "80000000L",
"flash_mode": "qio",
"hwids": [
[
"0x303A",
"0x1001"
]
],
"mcu": "esp32s3",
"variant": "lolin_s3"
},
"connectivity": [
"wifi"
],
"debug": {
"openocd_target": "esp32s3.cfg"
},
"frameworks": [
"arduino",
"espidf"
],
"name": "WEMOS LOLIN S3",
"upload": {
"flash_size": "16MB",
"maximum_ram_size": 327680,
"maximum_size": 16777216,
"require_upload_port": true,
"speed": 460800
},
"url": "https://www.wemos.cc/en/latest/s3/index.html",
"vendor": "WEMOS"
}

Jason2866, 2023-12-16:
@robertlipe There is some code to take care that correct and needed flash modes are set in Platformio here https://github.com/platformio/platform-espressif32/blob/develop/builder/main.py#L45-L86
My guess, the board PSRAM is defect.

mars000, 2023-12-17:
Hi @valeros  @Jason2866 @robertlipe 

don't think its a hardware fault on my lolin s3 board, as the exact same example code works perfectly when using Arduino IDE.
Something going on in platformio config ??
Anyway, I managed to get it going by editing/hacking the` lolin_s3.json` file.
It seems it work with the ` lolin_s3.json` file pasted below.
any insights would be much appreciated as i know these lolin s3 cards are very popular so i'm sure others will fall into these issues as some point.

you will notice all i did is **remove the following lines** in `"extra_flags"`
```
 "-DARDUINO_USB_CDC_ON_BOOT=1",
 "-DARDUINO_RUNNING_CORE=1",
 "-DARDUINO_EVENT_RUNNING_CORE=1"

```

Working `lolin_s3.json` file:

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_LOLIN_S3",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "lolin_s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "WEMOS LOLIN S3",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.wemos.cc/en/latest/s3/index.html",
  "vendor": "WEMOS"
}
```

closed 2025-02-25 by stale[bot]