platformio/platform-espressif32 #1230: Component with pre-built library and one level above the project root

jacque99, 2023-11-10:
Dear Support,

I have ESP32 S3 project and I am aiming to hold my generic components one level above the project root.
Project structure is as below.
	  - components
		- zmod4xxx  (component directory)
          - common
               - sensor 
               - wifi 
	  - project_root
		- src

Now, issue is that when I compile project it says "undefined reference to `init_iaq_2nd_gen'". This function is from the 
zmod4xxx component which contains a pre-built binary file "lib_iaq_2nd_gen.a". 
Please note that if I copy components folder to same level as project root where platformio.ini exists, then it compiles without any error. 

I suppose zmod4xxxx component is ESP-IDF components, therefore I declared it in the CMakeLists.txt as below, according to the  
[document](https://github.com/platformio/platform-espressif32/issues/479).
    cmake_minimum_required(VERSION 3.16.0)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
    list(APPEND EXTRA_COMPONENT_DIRS ../components)
    project(esp32s3_wo_pm)

Also I declared components folder in the platformio.ini lib_extra_dirs. 
lib_extra_dirs =
    ../../common/esp-idf-lib/components
    ../../common/components
    ../components
    ../common

Please help me to find an issue. Below is my project folder.
https://github.com/jacque99/iaq_zmod4xxx.git

Thank you and best regards,
Lhagva

closed 2023-11-13 by valeros

valeros, 2023-11-13:
Thanks for reporting, should be fixed in the development branch.

jacque99, 2023-11-14:
Hello, thank you very much for your reply.
I have created develop branch, please fix there.

valeros, 2023-11-14:
It's already fixed in this repository, you can pull the latest version directly in your `platformio.ini`:
```ini
[env:esp32-s3-devkitc-1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = espidf
```

jacque99, 2023-11-14:
Thank you for fixing it, it works perfectly.
Do you recommend updating the platform-espressif32? If yes please tell me which version I shall update.
I would be happy if you provide me with a link guiding for an update.

valeros, 2023-11-14:
Currently, the upstream version of the platform from the repository is the only way to obtain the fix. When a new release is published you will be able to update it automatically.

commit ff6ec884 (2023-11-13): Properly handle precompiled static archives located outside IDF project root

Resolves #1230