platformio/platform-espressif32 #1231: Add support for Adafruit Qualia ESP32-S3 RGB666 board

aremmell, 2023-11-11:
Would be nice to have; I must be missing something because I can get it in Arduino IDE by setting the extra board URL to: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

Product URL: https://www.adafruit.com/product/5800

HaenZ33, 2023-11-13:
ty for moving! looking forward to a solution!

aremmell, 2023-11-28:
I don't know if this will help, since I'm relatively new to PlatformIO, but [here](https://github.com/platformio/platform-espressif32/files/13483671/qualia.txt) is what I thought relevant from Arduino IDE's configuration for this board.

If I knew enough about how this all worked I would be glad to do it myself, but I can copy/paste into the insufferable Arduino IDE until someone more experienced than me can take a look.

commit 13a4062e (2023-12-27): Add new boards

Resolves #1231, resolves #209
closed 2023-12-05 by valeros