platformio/platform-espressif32 #1232: ESP32-S3-WROOM-1 fails with espressif32 @ 5.4.0, framework = espidf

rgov, 2023-11-13:
I can successfully program my board, an ESP32-S3-WROOM-1U-N16, with espressif32 @ 6.4.0 but not espressif32 @ 5.4.0.

As a test case, I made a basic project like this:

```
$ cat platformio.ini
[env:test]
platform = espressif32 @ 6.4.0
framework = espidf

# Settings for ESP32-S3-WROOM-1U-N16
board = esp32-s3-devkitc-1
board_build.partitions = partitions.csv
board_upload.flash_size = 16MB
board_upload.before_reset = no_reset
board_upload.after_reset = no_reset
```

<details>
<summary>Contents of other source files.</summary>

```
$ cat src/main.cpp
extern "C" void app_main() {
    while (1) { }
}

$ cat partitions.csv
# This is a based on default_16MB.csv from
# https://registry.platformio.org/tools/platformio/framework-arduinoespressif32

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
data,     data, spiffs,  0xc90000,0x360000,
coredump, data, coredump,0xFF0000,0x10000,
```
</details>

Everything seems to go well when flashing and I can tell from serial output that my `app_main()` is being called.

<details>
<summary>esptool output for flashing with espressif32 @ 6.4.0</summary>

```
"/opt/homebrew/Cellar/platformio/6.1.11/libexec/bin/python" "/Users/rzg/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32s3 --port "/dev/cu.usbmodem1102" --baud 460800 --before no_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 /private/tmp/blinky/.pio/build/test/bootloader.bin 0x8000 /private/tmp/blinky/.pio/build/test/partitions.bin 0xe000 /private/tmp/blinky/.pio/build/test/ota_data_initial.bin 0x10000 .pio/build/test/firmware.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem1102
WARNING: Pre-connection option "no_reset" was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 70:04:1d:be:bd:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00041fff...
Compressed 20976 bytes to 13326...
Writing at 0x00000000... (100 %)
Wrote 20976 bytes (13326 compressed) at 0x00000000 in 0.6 seconds (effective 283.0 kbit/s)...
Hash of data verified.
[... 136 lines cut ...]
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 618.7 kbit/s)...
Hash of data verified.
Compressed 227792 bytes to 126741...
Writing at 0x00010000... (12 %)
Writing at 0x0001cef2... (25 %)
Writing at 0x00022979... (37 %)
Writing at 0x00027c80... (50 %)
Writing at 0x0002d24e... (62 %)
Writing at 0x00033ab7... (75 %)
Writing at 0x0003dc3b... (87 %)
Writing at 0x000433a9... (100 %)
Wrote 227792 bytes (126741 compressed) at 0x00010000 in 3.2 seconds (effective 572.5 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```
</details>

A telling sign is that this message only appears with `espressif32 @ 5.4.0` and `platform = espidf`:

```
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash mode setting. Use the --flash_mode=keep option instead of --flash_mode=dio in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
```

rgov, 2023-11-13:
The warning is about the image file at 0x0 is the `bootloader.bin`, so I wonder if it's not being built correctly.

espressif32 @ 5.4.0:

```
"/opt/homebrew/Cellar/platformio/6.1.11/libexec/bin/python" "/Users/rzg/.platformio/packages/tool-esptoolpy@1.40400.0/esptool.py" --chip esp32s3 elf2image --flash_mode qio --flash_freq 80m --flash_size 16MB -o .pio/build/test/bootloader.bin .pio/build/test/bootloader.elf
```

espressif32 @ 6.4.0

```
"/opt/homebrew/Cellar/platformio/6.1.11/libexec/bin/python" "/Users/rzg/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 16MB -o .pio/build/test/bootloader.bin .pio/build/test/bootloader.elf
```

Note that the older version uses `--flash_mode qio` instead of `--flash_mode dio` which is also related to the warning ("not changing the flash mode setting").

rgov, 2023-11-13:
Looks like this is the commit that fixed it: https://github.com/platformio/platform-espressif32/commit/d0835e994c2af374198cbdf976d6019dc7dafe5f

I confirmed that applying this patch to 5.4.0 resolved the issue. Is it possible to backport to 5.4.x?

I am not sure how SCons works if I'm able to override `BOARD_FLASH_MODE` instead of applying the patch?

valeros, 2023-11-13:
Hi @rgov, have you tried overriding the flash mode directly in your `platformio.ini` as described [here](https://docs.platformio.org/en/latest/platforms/espressif32.html#flash-mode)?

rgov, 2023-11-13:
I think that does work around the issue, thanks. Still might be worth backporting if otherwise the default generates unbootable firmware.

rgov, 2023-11-13:
Could you explain more? I'm building a project which embeds the `arduino` component, which only supports IDF < 4.5 (looks like support for >= 5 is in alpha now). I chose espressif32 @ 5.4.0 because it comes with esp-idf 4.4.5. The [framework-espidf](https://registry.platformio.org/tools/platformio/framework-espidf/installation) versions like 3.50101.230828 don't mean anything to me, they don't obviously correspond with tags of the [espressif/esp-idf](https://github.com/espressif/esp-idf) repo.

Jason2866, 2023-11-13:
Latest version 6.4.0 will fall back to IDF 4.4.5 when using Arduino as Component.
You have to do nothing else.

rgov, 2023-11-13:
I think it would be `platform = espressif32; framework = espidf, arduino` (per [this post](https://community.platformio.org/t/esp32-dev-difference-between-framework-espidf-framework-arduino-and-framework-arduino-espidf/10560/2)) but I get some "multiple definitions errors" I will have to resolve.

Jason2866, 2023-11-13:
Better pin the plaform with `platform = espressif32 @ ^6.4.0`

closed 2025-02-25 by stale[bot]