platformio/platform-espressif32 #1234: v6.4.0: Deployed ESP-IDF version vs. advertised version?

0xPIT, 2023-11-13:
As per 
platform-espressif32 v6.4.0 requires esp-idf version **~3.50101.0** (source: https://github.com/platformio/platform-espressif32/blob/v6.4.0/platform.json#L94), yet the release notes claim "**Updated ESP-IDF to v5.1.1 (release notes)**"

Can anyone help me understand how the versioning works and/or why there's an outdated esp-idf version available?

Thanks!

valeros, 2023-11-13:
Hi @0xPIT, `3.50101.0` is a special version of the IDF package used by PlatformIO internally. The minor part of the version `50101` basically translates to the original `5.1.1`.

closed 2023-11-13 by valeros

0xPIT, 2023-11-13:
Thanks @valeros 

That is curious, as esp-idf 5.1.1 should contain newer libraries/drivers (eg. MCPWM) that are not included when platformio installs esp-idf...

Jason2866, 2023-11-13:
The release 5.1.1 version is from August. Don't mix branch release/v5.1 with released version.

valeros, 2023-11-13:
> That is curious, as esp-idf 5.1.1 should contain newer libraries/drivers (eg. MCPWM) that are not included when platformio installs esp-idf...

The `MCPWM` driver is present in the latest IDF package for 5.1.1 shipped by PlatformIO, should be somewhere here  `/YOUR_HOME_DIR/.platformio/framework-espidf/components/driver/mcpwm`

0xPIT, 2023-11-13:
Thanks for clarifying.

That’s what I thought as well, but it does not look to be the case. That’s
why I posted my question.

Even when I delete my ~/.platformio/framework-espidf/ and clean->rebuild my
project, the esp-idf version expected (5.1.1, I’m on the latest pio release
6.4.0) is not being installed, or at least the header files present in
v5.1.1 are not available.

For example, "driver/mcpwm_prelude.h" is not available, even though it is
when I switch tag to v5.1.1 on espressif’s esp-idf repo.

valeros, 2023-11-13:
> What lands in my  `~/.platformio/packages/framework-espidf/package.json` is actually  version **3.40405.230623**.

Are you working on an IDF project with Arduino as a component? If so, then `3.40405.230623` is the correct package because Arduino core v2 depends on IDF v4.4.

0xPIT, 2023-11-13:
Ah, yes, thanks.
looks like Arduino is the issue there. (I use `framework = espidf, arduino`)

It seems to silently downgrade, even when explicitly specifying espressiv32@6.4.0.

Any known workarounds? 

```
Platform Manager: espressif32@6.4.0 has been installed!
Tool Manager: toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5 has been installed!
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch5 has been installed!
Tool Manager: toolchain-esp32ulp@1.23500.220830 has been installed!
Tool Manager: framework-arduinoespressif32@3.20011.230801 has been installed!
--> Tool Manager: framework-espidf@3.40405.230623 has been installed!
```

Jason2866, 2023-11-13:
There is currently no Arduino compatible version for IDF 5.1. So the only possible way is how Platformio is doing.

0xPIT, 2023-11-13:
Thanks everyone, very helpful!