platformio/platform-espressif32 #1235: Failed to resolve component 'spiffs' in PlatformIO but not using idf.py (ESP-IDF)

nicklasb, 2023-11-13:
Hi, 

When I try to use the spiffs component, it is not being resolved in PlatformIO, but it works in IDF.
I have a minimal project here that has both a main/ and a src/ : 
https://github.com/nicklasb/spifftest

I have:
* Reinstalled esp-idf
* Tried different versions of the platform
* Tried different board configs
* Rebooted, restarted and so on

nicklasb, 2023-11-14:
Thanks!
Ultra odd that they have gotten reversed at some stage before I pushed.. 
Ok, will try and get back. Obviously something on my side.
Again, thank you!

nicklasb, 2023-11-14:
(funny that the CMakeLists.txts worked even being wrong)

So the errors are like this. Including CMake aswell. I am stumped, I must say.

```
 *  Executing task: platformio run --environment ttgo-t-beam 

Processing ttgo-t-beam (platform: espressif32; board: az-delivery-devkit-v4; framework: espidf)
----------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (2023.10.12+sha.cc863f0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0 
 - framework-espidf @ 3.50101.0 (5.1.1) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0) 
 - tool-mklittlefs @ 3.2.0 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/*home_directory*/Documents/PlatformIO/Projects/spifftest/.pio/build/ttgo-t-beam/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /*home_directory*/.platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'spiffs'.
Call Stack (most recent call first):
  /*home_directory*/.platformio/packages/framework-espidf/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /*home_directory*/.platformio/packages/framework-espidf/tools/cmake/build.cmake:595 (__build_expand_requirements)
  /*home_directory*/.platformio/packages/framework-espidf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:3 (project)

================================= [FAILED] Took 3.77 seconds =================================

 *  The terminal process "platformio 'run', '--environment', 'ttgo-t-beam'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

Cmake output: 

```
The target system is: Generic -  - 
The host system is: Darwin - 21.6.0 - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/*home_directory*/Documents/PlatformIO/Projects/spifftest/.pio/build/ttgo-t-beam/CMakeFiles/3.21.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address;
Id flags:  

[... 515 lines cut ...]
    arg [-L/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib] ==> dir [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib]
    arg [CMakeFiles/cmTC_e6283.dir/CMakeCXXCompilerABI.cpp.obj] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lnosys] ==> lib [nosys]
    arg [-lc] ==> lib [c]
    arg [-lgcc] ==> lib [gcc]
    arg [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o] ==> obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o]
    arg [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtn.o] ==> obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtn.o]
  collapse obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/crt0.o] ==> [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/lib/crt0.o]
  collapse obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crti.o] ==> [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0/crti.o]
  collapse obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtbegin.o] ==> [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0/crtbegin.o]
  collapse obj [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o] ==> [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o]
[... cut ...] in.o;/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0/crtend.o;/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0/crtn.o]
  implicit dirs: [/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc/xtensa-esp32-elf/12.2.0;/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/lib/gcc;/*home_directory*/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/lib]
  implicit fwks: []

```

valeros, 2023-11-15:
The project you're compiling and project in the repository are quite different:
```
PLATFORM: Espressif 32 (2023.10.12+sha.cc863f0) > AZ-Delivery ESP-32 Dev Kit C V4
```

Where did you get this platform? There were no official releases with this version. The latest is [6.4.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.4.0)

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0 
 - framework-espidf @ 3.50101.0 (5.1.1) 
```

How come the Arduino package (with unsupported and unstable version v3.0.0) ended up in the IDF project?

```
Failed to resolve component 'spiffs'.
```

Taking into account that you're working with a custom setup of the platform, please make sure your IDF distribution is not corrupted.

nicklasb, 2023-11-15:
I have no idea why the project is all there is, can board configs and similar make a difference? 
I have tried with different boards earlier.
The project is in a new folder. 
I have just deleted and reinstalled the IDF distribution using the ESP-IDF plugin in VSCode. 
And compiling using idf.py works.

nicklasb, 2023-11-15:
Note that I updated my response.

Jason2866, 2023-11-15:
Looks like using my fork for Tasmota. Never claimed to be compatible to an official build. I have adopted and removed parts not needed for Tasmota. This would explain the behaviour.
Completely removed SPIFFS (deprecated and not maintained anymore). Deleted in IDF and in Arduino. Using LittleFS instead.
@nicklasb Please do NOT open an issue in official Platformio Github when issues occur using a fork version.

nicklasb, 2023-11-15:
> Looks like using my fork for Tasmota. Never claimed to be compatible to an official build. I have adopted and removed parts not needed for Tasmota. This would explain the behaviour. Completely [...]

Based on my platform.ini, how can I know that I am having problems with a fork?

Jason2866, 2023-11-15:
To force your Platformio setup to use official framework delete hidden folder `.cache` and `.platformio`  Platformio will rebuild and download everything needed as configured in your Platformio.ini.
Platformio does NOT use the espressif IDF installed version.

closed 2023-11-15 by nicklasb