platformio/platform-espressif32 #1236: How can I install an old work environment or export a work environment from a PC to other?

JAHTKELD, 2023-11-16:
Hello:

I have an environment on my laptop that can build my project

**VSC 1.69.2
C/C++ 1.2.0
Azure IoT Hub 2.16.4
Platformio IDE v1.10.0 Core 5.1.0 Home 3.3.3**

I have tried to mount the work environment on another computer, for this

I have downloaded and installed VSC version 1.69.2
Inside VSC I have

> Install C/C++ and downgrade to 1.2.0
> Install Azure IoT Hub and downgrade to 2.16.4
> Platformio IDE and downgrade to v1.10.0 (the core keep in 6.1.11 Home 3.4.4)

In the platformio.ini file I have

```
platform = espressif32@1.12.4
platform_packages =
toolchain-xtensa32@~2.80400.0
```

And the result for build is

> c:/users/jaht.keld/.platformio/packages/toolchain-xtensa32/bin/…/lib/gcc/xtensa-esp32-elf/8.4.0/…/…/…/…/xtensa-esp32-elf/bin/ld.exe: cannot find -lUsersjaht.KELD.platformiopackagesframework-espidf@3.40001.200521componentsxtensaesp32libhal
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\esp32dev\bootloader.elf] Error 1

I have maked (At the laptop when I had this error it fixed downgrade the platformio versión)

```
pip uninstall platformio
pip install platformio==5.1.0
```

When I start VSC, the Platformilo Home can´t be show and the loader say

> Error: Error: Traceback (most recent call last):
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio_main_.py”, line 109, in main
> cli() # pylint: disable=no-value-for-parameter
> ^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\click\core.py”, line 829, in call
> return self.main(*args, **kwargs)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\click\core.py”, line 782, in main
> rv = self.invoke(ctx)
> ^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\commands_init_.py”, line 44, in invoke
> return super(PlatformioCLI, self).invoke(ctx)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\click\core.py”, line 1259, in invoke
> return _process_result(sub_ctx.command.invoke(sub_ctx))
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\click\core.py”, line 1066, in invoke
> return ctx.invoke(self.callback, **ctx.params)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\click\core.py”, line 610, in invoke
> return callback(*args, **kwargs)
> ^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\commands\home\command.py”, line 93, in cli
> run_server(
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\commands\home\run.py”, line 63, in run_server
> contrib_dir = get_core_package_dir(“contrib-piohome”)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\manager\core.py”, line 48, in get_core_package_dir
> pkg = pm.get_package(spec)
> ^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\manager\base.py”, line 236, in get_package
> for pkg in self.get_installed():
> ^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\manager\base.py”, line 209, in get_installed
> pkg = PackageItem(pkg_dir)
> ^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\meta.py”, line 405, in init
> self.metadata = self.load_meta()
> ^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\meta.py”, line 440, in load_meta
> return PackageMetaData.load(manifest_path)
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File “C:\Users\jaht.KELD.platformio\penv\Lib\site-packages\platformio\package\meta.py”, line 393, in load
> data[“spec”] = PackageSpec(**data[“spec”])
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^
> TypeError: PackageSpec.init() got an unexpected keyword argument ‘uri’
> 
> ============================================================
> 
> An unexpected error occurred. Further steps:
> 
> Verify that you have the latest version of PlatformIO using
> pip install -U platformio command
> 
> Try to find answer in FAQ Troubleshooting section
> [Redirecting...](https://docs.platformio.org/page/faq.html)
> 
> Report this problem to the developers
> [Issues · platformio/platformio-core · GitHub](https://github.com/platformio/platformio-core/issues)
> 
> ============================================================
> 
> at c:\Users\jaht.KELD.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
> at ChildProcess.d (c:\Users\jaht.KELD.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
> at ChildProcess.emit (node:events:390:28)
> at ChildProcess.cp.emit (c:\Users\jaht.KELD.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
> at maybeClose (node:internal/child_process:1064:16)
> at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)

Out of desperation, I tried to replace the .platformio folder on the PC where it doesn’t work, with the folder on the laptop that does work, but this wasn’t valid either.

```
Installing PlatformIO Core…
Please do not close this window and do not open other folders until this process is completed.
PlatformIO IDE installed successfully.

Please restart VSCode.
```

After Restar
PlatformIO IDE installation has been suspended, because PlatformIO IDE Installer is already started in another window.

I restart again, teh Pio Home doesný load and the build not works

**Is there a way to export a working configuration from one computer to another?**

valeros, 2023-11-16:
Hi @JAHTKELD, v1.12.4 is three years old, is there any reason to stick to it?

> I have maked (At the laptop when I had this error it fixed downgrade the platformio versión)

it's not recommended to downgrade PlatformIO core. So, first of all, close VSCode, remove `.platformio` and `.cache` folders here `C:\Users\YOUR_USER` and reinstall the PlatformIO extension from scratch. 

Now try to compile your project.

JAHTKELD, 2023-11-16:
Thank you for you help

I use v1.12.4 because I have modified some files related to tls such as tlsio_esp_tls.c due to the use of encryption chips and the latest versions do not work correctly for me.

I have removed the .plaftformio and install again (the last), and the platformio HOme doesn´t show. 

The error report is

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.2
  PIO IDE: v3.3.1
  System: Windows_NT, 10.0.19045, x64

  %23 Exception
  ```
  Error: AttributeError: Traceback (most recent call last):
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\platformio\home\cli.py", line 97, in cli
    home_url=home_url,
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\platformio\home\run.py", line 102, in run_server
    log_level="warning",
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\uvicorn\main.py", line 461, in run
    server.run()
  File "C:\Users\jaht.KELD\.platformio\penv\lib\site-packages\uvicorn\server.py", line 67, in run
    return asyncio.run(self.serve(sockets=sockets))
AttributeError: module 'asyncio' has no attribute 'run'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\jaht.KELD\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:10851
	at ChildProcess.c (c:\Users\jaht.KELD\.vscode\extensions\platformio.platformio-ide-3.3.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4598)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

valeros, 2023-11-16:
That's strange, are you using a global Python interpreter? Please open a new terminal in the IDE and run the `pio system info` command.

JAHTKELD, 2023-11-16:
Hello. This is the result

PS D:\TrabajoJose\DOMUSA\CCDP_WIFI\CCDPWIFI_VSC_8.00K2> pio system info
--------------------------  ----------------------------------------------------------
PlatformIO Core             6.1.11
Python                      3.6.2-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\jaht.KELD\.platformio
PlatformIO Core Executable  C:\Users\jaht.KELD\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\jaht.keld\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10
--------------------------  ----------------------------------------------------------
PS D:\TrabajoJose\DOMUSA\CCDP_WIFI\CCDPWIFI_VSC_8.00K2>

valeros, 2023-11-16:
> Python 3.6.2-final.0

Python 3.6.2 is six years old, it has reached end-of-life and no longer supported. If you want to use PlatformIO Home then you need to update it. Alternatively, you can disable it globally, so that PlatformIO will install its own Python package.

JAHTKELD, 2023-11-17:
Hello:

Now I have installed

```
VSC 1.69.2
C/C++ 1.2.0
Azure IoT Hub 2.16.4
Platformio IDE v3.3.1 Core 6.1.11 Home 3.4.4
```
Platformio ini is
```
platform = espressif32@1.12.4
platform_packages = 
   toolchain-xtensa32@~2.50200.0
   framework-espidf@3.40001.200521
```

and the  pio system info show

```
--------------------------  ----------------------------------------------------------
PlatformIO Core             6.1.11
Python                      3.11.4-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\jaht.KELD\.platformio
PlatformIO Core Executable  C:\Users\jaht.KELD\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\jaht.KELD\.platformio\penv\Scripts\python.exe    
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10
```

And then I have some compiler errors that I dont have when I build in my labtop, for that I would like to install the same configuration as I have in labtop to my PC, beacause I thought it was going easier that fix the compiler errors, but both options are complicated to solve

> Processing esp32dev (platform: espressif32@1.12.4; board: esp32dev16; framework: espidf)
> ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev16.html
> PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
> HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
> DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-espidf @ 3.40001.200521 (4.0.1)
>  - tool-cmake @ 3.16.4
>  - tool-esptoolpy @ 1.20600.0 (2.6.0)
>  - tool-idf @ 1.0.1
>  - tool-mconf @ 1.4060000.20190628 (406.0.0)
>  - tool-ninja @ 1.9.0
>  - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
>  - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
> Reading CMake configuration...
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 0 compatible libraries
> Scanning dependencies...
> No dependencies
> Building in release mode

```
Compiling .pio\build\esp32dev\esp-idf\src\custom_hsm_x509.c.o
In file included from C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/arch.h:228:0,
                 from C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/debug.h:40,
                 from C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/opt.h:52,
                 from C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/lwip/lwip/src/include/lwip/sockets.h:42,
                 from C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/lwip/port/esp32/include/sys/socket.h:33,
                 from components/esp-tls/esp_tls.h:5,
                 from src/custom_hsm_x509.c:13:
src/custom_hsm_x509.c: In function 'CheckATECC608A':
C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/newlib/include/ctype.h:57:54: error: array subscript has type 'char' [-Werror=char-subscripts]
 #define __ctype_lookup(__c) ((__ctype_ptr__+sizeof(""[__c]))[(int)(__c)])
                                                      ^
C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/newlib/include/ctype.h:61:24: note: in expansion of macro '__ctype_lookup'
 #define islower(__c) ((__ctype_lookup(__c)&(_U|_L))==_L)
                        ^
C:/Users/jaht.KELD/.platformio/packages/framework-espidf/components/newlib/include/ctype.h:86:7: note: in expansion of macro 'islower'
       islower (__x) ? (int) __x - 'a' + 'A' : (int) __x;})
       ^
src/custom_hsm_x509.c:113:26: note: in expansion of macro 'toupper'
             serialstr[i]=toupper(serialstr[i]);
                          ^
src/custom_hsm_x509.c: In function 'LeeCertificado':
src/custom_hsm_x509.c:165:11: warning: unused variable 'restart' [-Wunused-variable]
   int32_t restart = 0; //0 normal, 1 bluetooth y OTA
           ^
Compiling .pio\build\esp32dev\esp-idf\app_trace\app_trace.c.o
Compiling .pio\build\esp32dev\esp-idf\app_trace\app_trace_util.c.o
src/custom_hsm_x509.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1.exe: some warnings being treated as errors
Compiling .pio\build\esp32dev\esp-idf\app_trace\host_file_io.c.o
Compiling .pio\build\esp32dev\esp-idf\app_trace\gcov\gcov_rtio.c.o
Compiling .pio\build\esp32dev\esp-idf\app_update\esp_ota_ops.c.o
Compiling .pio\build\esp32dev\esp-idf\app_update\esp_app_desc.c.o
Compiling .pio\build\esp32dev\esp-idf\asio\asio\asio\src\asio.cpp.o
*** [.pio\build\esp32dev\esp-idf\src\custom_hsm_x509.c.o] Error 1
```

valeros, 2023-11-17:
```
platform_packages = 
   toolchain-xtensa32@~2.50200.0
   framework-espidf@3.40001.200521
```
You cannot that easily switch between toolchain package, often each IDF version requires a specific toolchain package. I'd recommend removing the entire `platform_packages` section, mainly because v1.12.4 will still install IDF `v3.40001.200521` and a proper toolchain compatible with this version.

JAHTKELD, 2023-11-17:
Hello:

Thank you for you help.
I have remove platforme and packages folder inside .platformio and when I start VSC I get

```
Resolving esp32dev dependencies...
Platform Manager: Installing espressif32 @ 1.12.4
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@1.12.4 has been installed!
Tool Manager: Installing toolchain-xtensa32 @ ~2.50200.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-xtensa32@2.50200.97 has been installed!
Tool Manager: Installing toolchain-esp32ulp @ ~1.22851.190618
Tool Manager: Warning! More than one package has been found by toolchain-esp32ulp @ ~1.22851.190618 requirements:
Tool Manager:  - platformio/toolchain-esp32ulp@1.23500.220830
Tool Manager:  - espressif/toolchain-esp32ulp@2.35_20220830
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-esp32ulp@1.22851.191205 has been installed!
Tool Manager: Installing framework-espidf @ 3.40001.200521
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-espidf@3.40001.200521 has been installed!
Tool Manager: Installing tool-esptoolpy @ ~1.20600.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@1.20600.0 has been installed!
Tool Manager: Installing tool-cmake @ ~3.16.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-cmake@3.16.4 has been installed!
Tool Manager: Installing tool-ninja @ ^1.7.0
Unpacking 0% 10% 20%
Tool Manager: tool-ninja@1.9.0 has been installed!
Tool Manager: Installing tool-mconf @ ~1.4060000.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80%
Tool Manager: tool-mconf@1.4060000.20190628 has been installed!
Tool Manager: Installing tool-idf @ ~1.0.1
Unpacking 0% 10% 20%
Tool Manager: tool-idf@1.0.1 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40502.0 has been installed!
Updating metadata for the vscode IDE...
UserSideException: Processing esp32dev (platform: espressif32@1.12.4; board: esp32dev; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40001.200521 (4.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.20600.0 (2.6.0) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.36.1.windows.1") 
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/jaht.KELD/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/jaht.KELD/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/jaht.KELD/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/jaht.KELD/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/jaht.KELD/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.

-- Configuring incomplete, errors occurred!
See also "D:/TrabajoJose/DOMUSA/CCDP_WIFI/CCDPWIFI_VSC_8.00K2/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".
See also "D:/TrabajoJose/DOMUSA/CCDP_WIFI/CCDPWIFI_VSC_8.00K2/.pio/build/esp32dev/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/jaht.KELD/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/jaht.KELD/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)
  C:/Users/jaht.KELD/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:11 (project)
```

[CMakeError.log](https://github.com/platformio/platform-espressif32/files/13391622/CMakeError.log)
[CMakeOutput.log](https://github.com/platformio/platform-espressif32/files/13391623/CMakeOutput.log)

valeros, 2023-11-17:
```
PACKAGES: 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
```

Strange, the toolchain v5.2.0 is still installed, have you removed the `platform_packages` from your `platformio.ini`?

> pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.

Have you also installed PIP when installed Python?

JAHTKELD, 2023-11-19:
I made a new installation of VSC and platfromio

**VSC  1.84 (last)
C++ 1.2.0 (as my labtop where the build is ok)
Azure iot (2.16.4) (as my labtop where the build is ok)
Cmaketool (1.5.3) (as my labtop where the build is ok)
Python v2021.1.501.502429796 (as my labtop where the build is ok) 
Python instal pip_packages 
Platfomio 1.1.0.0 with core  6.1.11 Home 3.4.4**

The firts time I tried to build I need to install toolchain-xtensa32 @ 2.80400
and the result is

```
Executing task: C:\Users\vboxuser\.platformio\penv\Scripts\platformio.exe run 

Processing esp32dev (platform: espressif32@1.12.4; board: esp32dev16; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev16.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40001.200521 (4.0.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.20600.0 (2.6.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 @ 2.80400.210211 (8.4.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio\build\esp32dev\bootloader.elf
c:/users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lUsersvboxuser.platformiopackagesframework-espidfcomponentsxtensaesp32libhal
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\bootloader.elf] Error 1
========================================================================================= [FAILED] Took 74.65 seconds =========================================================================================
```
I decided install last versión of C++, Azure Iot, Python, Platformio and Cmaketools

pio system info is
```
PS C:\CCDPWIFI_V8.00TNGOk> pio system info
--------------------------  ---------------------------------------------------------
PlatformIO Core             6.1.11
Python                      3.11.4-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\vboxuser\.platformio
PlatformIO Core Executable  C:\Users\vboxuser\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\vboxuser\.platformio\penv\Scripts\python.exe    
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10
```

And the build result is 
[CMakeOutput.log](https://github.com/platformio/platform-espressif32/files/13402932/CMakeOutput.log)

```
> Executing task: C:\Users\vboxuser\.platformio\penv\Scripts\platformio.exe run 
> 
[... 58 lines cut ...]
> Call Stack (most recent call first):
>   C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)
>   C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)
>   CMakeLists.txt:11 (project)
```

I dont know why in my labtop I can build right with all old version and I can not install this configuration in any other computer

The labtop that works fine have VSC 1.69.2 and
```
C:\Jose\DOMUSA\CCDPWIFIVSC\CCDPWIFI_V8.00TNGOk2>pio system info
--------------------------  -----------------------------------------------------
PlatformIO Core             5.1.0
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\JOSE\.platformio
PlatformIO Core Executable  C:\Users\JOSE\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\JOSE\.platformio\penv\Scripts\python.exe
Global Libraries            1
Development Platforms       3
Tools & Toolchains          18
```

valeros, 2023-11-20:
@JAHTKELD I won't be able to help you if you reinstall everything from scratch again and again.

First thing first, the `Azure IoT, Python, and CMakeTools` extensions have nothing to do with PlatformIO and shouldn't be taken into account.

> pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.

This error message is pretty much self-descriptive, IDF's Python dependencies are not installed. There might be a few reasons, for example because you have several Python interpreters available in your system and IDF picks up the wrong one. This behavior fixed in the later versions of this development platform, but since you want to use an old version, you need to make sure that IDF uses the right Python. Have you installed any Python interpreters from Microsoft Store?

JAHTKELD, 2023-11-20:
Hello:

I did the last installation in a virtual machine with the idea of being able to take it to any PC
I have installed Python as extension of VSC, is that not right? I thought that was enough, I have python in two folders,  .platformio/python3  and .platformio/penv/scripts   Do I must to install other python?

valeros, 2023-11-20:
> I have python in two folders, .platformio/python3 and .platformio/penv/scripts Do I must to install other python?

Do you need Python for other projects? Anyway, have you tried a clean installation in a but this time try to install PlatformIO first, without any other extensions.

valeros, 2023-11-20:
Now please open a new terminal window in the IDE and run `where python`.

JAHTKELD, 2023-11-20:
**where python** doesnt show anything

valeros, 2023-11-20:
That's strange, what happens if you directly specify the path to Python, for example:
```ini
[env:yourenv]
platform = espressif32@1.12.4
...
board_build.cmake_extra_args = -DPYTHON=C:/Users/vboxuser/.platformio/penv/Scripts/platformio.exe
```

JAHTKELD, 2023-11-20:
Hello:

Without make build, only open the VSC I have

```
Resolving esp32dev dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing esp32dev (platform: espressif32@1.12.4; board: esp32dev16; framework: espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev16.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40001.200521 (4.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.20600.0 (2.6.0) 
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 @ 2.80400.210211 (8.4.0)
Reading CMake configuration...
-- Could NOT find Git (missing: GIT_EXECUTABLE) 
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
-- Configuring incomplete, errors occurred!
See also "C:/CCDPWIFI_V8.00TNGOk/.pio/build/esp32dev/CMakeFiles/CMakeOutput.log".

CMake Warning at C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/git_submodules.cmake:4 (message):
  Git executable was not found.  Git submodule checks will not be executed.
  If you have any build issues at all, start by adding git executable to the
  PATH and rerun cmake to not see this warning again.
Call Stack (most recent call first):
  C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/idf.cmake:34 (include)
  C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:9 (include)

Usage: platformio [OPTIONS] COMMAND [ARGS]...

Try 'platformio -h' for help.

Error: No such command "C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/check_python_dependencies.py"

CMake Error at C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/build.cmake:270 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/build.cmake:396 (__build_check_python)
  C:/Users/vboxuser/.platformio/packages/framework-espidf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:11 (project)

========================== [FAILED] Took 3.44 seconds ==========================
```
[CMakeOutput.log](https://github.com/platformio/platform-espressif32/files/13415490/CMakeOutput.log)

My platformio ini is

```
[env:esp32dev]
platform = espressif32@1.12.4
board = esp32dev16
platform_packages =
    toolchain-xtensa32@~2.80400.0
framework = espidf
board_build.partitions = partitions.csv
board_build.embed_files =
	src/websitefiles/index.html
	src/websitefiles/jquery-3.4.1.min.js
	src/websitefiles/favicon.ico
build_flags = 
    -I "components/cryptoauthlib/cryptoauthlib/lib"
    -I "components/cryptoauthlib"
    -I "src"
    -DMBEDTLS_ECDH_GEN_PUBLIC_ALT 
    -DMBEDTLS_ECDH_COMPUTE_SHARED_ALT 
    -DMBEDTLS_ECDSA_SIGN_ALT 
    -DMBEDTLS_SSL_CIPHERSUITES=MBEDTLS_TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256,MBEDTLS_TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256,MBEDTLS_TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256
	-DSET_TRUSTED_CERT_IN_SAMPLES
board_build.cmake_extra_args = -DPYTHON=C:/Users/vboxuser/.platformio/penv/Scripts/platformio.exe
upload_port = COM6
```

valeros, 2023-11-20:
My bad, it should be 
```ini
board_build.cmake_extra_args = -DPYTHON=C:/Users/vboxuser/.platformio/penv/Scripts/python.exe
```

JAHTKELD, 2023-11-20:
Hello:
I can build now but at the end, the linker show a error

```
Linking .pio\build\esp32dev\bootloader.elf
c:/users/vboxuser/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lUsersvboxuser.platformiopackagesframework-espidfcomponentsxtensaesp32libhal
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\bootloader.elf] Error 1
====================================================

I have seen this error when on the laptop where I can build, it is updated to a different version of platformio 5.1.0, and when I downgrade platformio to 5.1.0 it builds again.
```

https://community.platformio.org/t/lib-not-found-with-last-version-core-change-home-change-framework/25274/2/

https://community.platformio.org/t/xtensa-esp32-elf-bin-ld-exe-cannot-find-library/23963/3

In the last link I can read something about the memory size (4MB) but I don't understand it very well because I don't use the Arduino environment. I use a board definition made by me on the esp32dev expanding to 16MB since I use the ESP32 which has 16MB. Could this be the error?

valeros, 2023-11-20:
> I use a board definition made by me on the esp32dev expanding to 16MB since I use the ESP32 which has 16MB. Could this be the error?

No, it's fine.

> I have seen this error when on the laptop where I can build, it is updated to a different version of platformio 5.1.0, and when I downgrade platformio to 5.1.0 it builds again.

OK, even though it's not the right solution, let's try downgrading PlatformIO with your current setup, run the following command in the IDE terminal:
```
C:/Users/vboxuser/.platformio/penv/Scripts/python.exe -m pip install platformio==5.1.0
```
Then clean and compile your project again.

JAHTKELD, 2023-11-20:
Hello:
I have the folowing error,

**************************************************************************************************************************************```
Obsolete PIO Core v5.1.0 is used (previous was 6.1.11)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
**************************************************************************************************************************************

Processing esp32dev (platform: espressif32@1.12.4; board: esp32dev16; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Error: Traceback (most recent call last):
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\commands\run\command.py", line 133, in cli
    process_env(
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
                                                            ^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\commands\run\processor.py", line 70, in process
    p = PlatformFactory.new(self.options["platform"])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\platform\factory.py", line 52, in new
    pkg = PlatformPackageManager().get_package(pkg_or_spec)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\package\manager\base.py", line 236, in get_package
    for pkg in self.get_installed():
               ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\package\manager\base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
          ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\package\meta.py", line 405, in __init__
    self.metadata = self.load_meta()
                    ^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\package\meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vboxuser\.platformio\penv\Lib\site-packages\platformio\package\meta.py", line 393, in load
    data["spec"] = PackageSpec(**data["spec"])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: PackageSpec.__init__() got an unexpected keyword argument 'uri'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

 *  The terminal process "C:\Users\vboxuser\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'clean'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  
```
```

I think I must to unistall before the platformio 6.1.11 but at the link 
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html#multiple-pio-cores-issue
it say "uninstall at System Terminal Not Platfomio IDE.

Where I have to do the unistall and install?

valeros, 2023-11-20:
> Where I have to do the unistall and install?

Please don't uninstall anything. What happens if you delete the `C:/Users/vboxuser/.platformio/packages` folder and compile your project again.

JAHTKELD, 2023-11-21:
Hello:

I have compared the laptop installation where I can build and the new installation that does not work

I can see differences in the packages

Laptop
```
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)
```
New Instalation
```
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 @ 2.80400.210211 (8.4.0)
```

**Laptop**

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev16.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40001.200521 (4.0.1)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.190618 (2.28.51)
 - toolchain-xtensa32 2.80200.200226 (8.2.0)

Microsoft Windows [Versión 10.0.19045.3693]
(c) Microsoft Corporation. Todos los derechos reservados.

C:\Jose\DOMUSA\CCDPWIFIVSC\CCDPWIFI_V8.00TNGOk2>pio system info
--------------------------  -----------------------------------------------------
PlatformIO Core             5.1.0
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\JOSE\.platformio
PlatformIO Core Executable  C:\Users\JOSE\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\JOSE\.platformio\penv\Scripts\python.exe
Global Libraries            1
Development Platforms       3
Tools & Toolchains          18
```

**New instalation**

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev16.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40001.200521 (4.0.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.20600.0 (2.6.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 @ 2.80400.210211 (8.4.0)
Reading CMake configuration...

PS C:\CCDPWIFI_V8.00TNGOk> pio system info
--------------------------  ---------------------------------------------------------
PlatformIO Core             6.1.11
Python                      3.11.4-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\vboxuser\.platformio
PlatformIO Core Executable  C:\Users\vboxuser\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\vboxuser\.platformio\penv\Scripts\python.exe    
Global Libraries            0
Development Platforms       1
Tools & Toolchains          11
```

valeros, 2023-11-21:
How come it shows `PlatformIO Core 6.1.11` in your VM if we downgraded it several steps ago?

JAHTKELD, 2023-11-21:
I dont knowI dont have installed anything . After downgraded it said   "Please remove multiple PIO Cores from a system" ; I didn`t remove anything, but when I restart VSC the platformio start it self to install the last version

JAHTKELD, 2023-11-21:
With  my labtop I allways work VSC without internet connection to avoid platformio update.
If I ever forget to remove the internet connection and the platform is updated, the downgrade always works for me

valeros, 2023-11-21:
OK, do you mind giving me a remote access to your VM, for example via TeamViewer or AnyDesk? Just send me credentials to `vkoval[at]platformio.org`

JAHTKELD, 2023-11-21:
Solved by Valeros in a particular way.

The main problem is that the version is quite old and there have been many bug fixes that are not present in this version.

The best solution would be to upgrade to a newer version to get the latest bug fixes.

Thank you very much Valeros

closed 2023-11-21 by JAHTKELD