platformio/platform-espressif32 #1237: WiFi on esp32-C6 / IDF 5.1.1 is unstable

vladm, 2023-11-22:
Hi - 

I'm using esphome (which uses platformio for esp32) with esp32c6 board and my wifi is very unstable. I see some WiFi fixes for C6 board in latest IDF release 5.1.2. Is it possible to upgrade IDF version for platformio?

TY

Jason2866, 2023-11-23:
Platformio is not the maintainer of IDF code. The "just" provide an IDE to use.
So wrong github to ask.

stintel, 2023-11-23:
Actually I was just referred here by @ivankravets to request publishing ESP-IDF 5.1.2 on https://registry.platformio.org/tools/platformio/framework-espidf.

vladm, 2023-11-23:
> Platformio is not the maintainer of IDF code. The "just" provide an IDE to use. So wrong github to ask.

I thought platformio maintainers are the one who pick which version of IDF is used in this project... Am I wrong?
https://github.com/platformio/platform-espressif32/commit/167c149b2aad83b69668569f90e3ffd14a45af0e

Jason2866, 2023-11-23:
@vladm Ahh, yes. Sorry. Maybe better change the headline to please update to release [IDF v5.1.2](https://github.com/espressif/esp-idf/releases/tag/v5.1.2)

Jason2866, 2023-11-25:
Until Platformio or espressif releases IDF 5.1.2 you can use
```
platform_packages =
        framework-espidf @ https://github.com/tasmota/esp-idf/releases/download/v5.1.2-org/esp-idf-v5.1.2-org.zip
```

vladm, 2023-11-27:
thanks @Jason2866 ... I'm using esphome's docker image, which doesn't allow to define an alternative source of platform package via configuration, so I'd need to dig in way deeper into that in order to try it... I'll probably just wait a bit to see if IDF gets upgraded naturally.

commit bc5b51f0 (2023-12-27): Update ESP-IDF to v5.1.2

Resolve #1237
closed 2023-12-06 by valeros