platformio/platform-espressif32 #1239: ESP32 S2 CPU clock stuck at 20MHz

swifty99, 2023-11-25:
Hello,

I use platformio via ESPHome. I do not know how to exactly proof it, however there are strong indications:

- When I use external pulses they are too slow by a factor of 80. Like the generated clock line of I2C.
- xt_clock_freq reports always 160 MHz, even if the platformio.ini requests otherwise.
- some SW modules take way too long, compared to other boards.

Timing issue can be seen here: https://github.com/esphome/issues/issues/5151
Maybe the problem is not related to ESPHome

How could I track this down?

the .ini:
```
[common]
lib_deps =
build_flags =
upload_flags =

; ========== AUTO GENERATED CODE BEGIN ===========
[platformio]
description = ESPHome 2023.11.2
[env:autogl-powa]
board = lolin_s2_mini
board_build.f_cpu = 120000000L
board_build.partitions = partitions.csv
board_upload.flash_size = 4MB
build_flags =
    -DESPHOME_LOG_LEVEL=ESPHOME_LOG_LEVEL_DEBUG
    -DUSE_ESP32
    -DUSE_ESP32_FRAMEWORK_ESP_IDF
    -DUSE_ESP32_VARIANT_ESP32S2
    -DUSE_ESP_IDF
    -Wno-nonnull-compare
    -Wno-sign-compare
    -Wno-unused-but-set-variable
    -Wno-unused-variable
    -fno-exceptions
extra_scripts =
    post:post_build2.py
framework = espidf
lib_deps =
    ${common.lib_deps}
lib_ldf_mode = off
platform = platformio/espressif32@5.4.0
platform_packages =
    platformio/framework-espidf@~3.40405.0
    espressif/toolchain-esp32ulp@2.35.0-20220830
```

Jason2866, 2023-11-25:
Known issue in espressif Arduino core. Not related to Platformio.
https://github.com/espressif/arduino-esp32/issues/8480

swifty99, 2023-11-25:
ok, thanks
I will remeasure with clock derived PWM. Or are there better ways?
As, the platformio clock setting still does not match.

Jason2866, 2023-11-25:
The bug is in IDF. It hits Arduino since it uses this function from IDF

valeros, 2023-12-06:
Closing as it's not a PlatformIO issue.

closed 2023-12-06 by valeros