platformio/platform-espressif32 #1240: ESPIDF provided example zigbee can't find zboss_* components

asward, 2023-11-28:
I created an espidf project through the espidf tooling. Demonstrated to myself that this example project would build through a container. Since the container is a bit finicky about USB I wanted to explore using platformio w/ espidf framework directly. 

I created an espidf project for my board through the platformio ui in vscode, then copied in the simple project files from the espidf generated example. Ran menuconfig and ensure zigbee component was enabled. Built the project and I get the following error:

```
Linking .pio\build\esp32-c6-devkitm-1\firmware.elf
XXX/riscv32-esp-elf/bin/ld.exe: cannot find -lzboss_stack.zczr: No such file or directory
XXX/riscv32-esp-elf/bin/ld.exe: cannot find -lzboss_port: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-c6-devkitm-1\firmware.elf] Error 1
```

A little research shows this is an espressif provided component. The platformio documentation says to drop the component code in the a 'components' folder in the root of the project. After doing this, I still have the same error.

I've tried w/ and w/o the CMakeLists.txt modification to specify the components directory.

pio.ini
```
[platformio]
src_dir = main

[env:esp32-c6-devkitm-1]
platform = espressif32
framework = espidf
board = esp32-c6-devkitm-1
```
Directory layout
[image]

Worth a note the espressif component manager still runs and brings in the zboss 3.0 lib from espressifs github. This is placed in a 'managed_components' directory at the root level. This if I remove the idf_components.yaml this doesn't occur.

Any help would be appreciated. Thanks!

Jason2866, 2023-11-28:
You have to add a YML file. Example https://github.com/Jason2866/Arduino_IDF_LittleFS/blob/main/src/idf_component.yml
This project is an example for using components via the component manager.

valeros, 2023-12-08:
Hi @asward, please retest your project with the latest platform version from the development branch that supports IDF 5.1.2:
```ini
[platformio]
src_dir = main

[env:esp32-c6-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf
board = esp32-c6-devkitm-1
```