platformio/platform-espressif32 #1242: ESP32-C6 Support?

mwls-sean, 2023-11-30:
Not entirely sure why it keeps saying `Error: This board doesn't support arduino framework!` even when using the master branch of the `framework-arduinoespressif32` package. 

This is my platformio.ini for the specific task/env:

```
[env:esp32-c6-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
framework = arduino
board = esp32-c6-devkitm-1
```

Full output:
```
Processing esp32-c6-devkitm-1 (platform: https://github.com/platformio/platform-espressif32.git; framework: arduino; board: esp32-c6-devkitm-1)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitm-1.html
PLATFORM: Espressif 32 (6.4.0+sha.ff6ec88) > Espressif ESP32-C6-DevKitM-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Error: This board doesn't support arduino framework!
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.0+sha.5fcdb84
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
================================================ [FAILED] Took 0.47 seconds ================================================
```

Jason2866, 2023-11-30:
There is no Arduino version released which support the C6. So the error message is completely correct.

seankovacs, 2023-11-30:
Gotcha. Yeah wasn't sure since I'm able to use the Arduino IDE with this board.

valeros, 2023-12-06:
A stable Arduino core for ESP32-C6 is not available at the moment.

closed 2023-12-06 by valeros