platformio/platform-espressif32 #1244: TLS failure when downloading firmware OTA on espressif32 6.4.0

robeastwood, 2023-12-06:
When using espressif32 platform v6.4.0, OTA fails with an `esp_tls_conn_read` error.

When using espressif32 platform v6.2.0 it works fine.

Output:
```
Starting OTA
Updating... Do not power off!
Http Event On Connected
Http Event Header Sent
Http Event On Header, key=Content-Type, value=application/octet-stream
Http Event On Header, key=Content-Length, value=1234592
Http Event On Header, key=Connection, value=keep-alive
Http Event On Header, key=Date, value=Tue, 05 Dec 2023 22:07:39 GMT
Http Event On Header, key=Last-Modified, value=Tue, 05 Dec 2023 22:05:26 GMT
Http Event On Header, key=ETag, value="745608613805a9c1b77d025a8c924b31"
Http Event On Header, key=Server, value=AmazonS3
Http Event On Header, key=X-Cache, value=Hit from cloudfront
Http Event On Header, key=Via, value=1.1 905eac6c91c9858bd0f20b56e9c842d4.cloudfront.net (CloudFront)
Http Event On Header, key=X-Amz-Cf-Pop, value=LHR50-P1
Http Event On Header, key=Alt-Svc, value=h3=":443"; ma=86400
Http Event On Header, key=X-Amz-Cf-Id, value=CQGeYEtC3e0yMgr_am3tl8lq30MwIPx11osiyEcME-s_KNvfdQUEUQ==
Http Event On Header, key=Age, value=126
E (66637) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
E (70489) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 48
Http Event Error
E (70491) esp_https_ota: data read -1, errno 0
Http Event Disconnected
Http Event Disconnected
Firmware Upgrade Fail
```

valeros, 2023-12-06:
Hi @robeastwood, please share here your `platformio.ini`. Could you also check if it works with 6.3.0?

robeastwood, 2023-12-06:
Ok sure. Will give it a go on 6.3. here's platformio.ini in the meantime:

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
monitor_filters = esp32_exception_decoder
framework = arduino
monitor_speed = 115200
board_build.partitions = partitions_custom.csv
board_build.flash_size = 8MB
board_upload.flash_size = 8MB
lib_deps = 
	gin66/FastAccelStepper@^0.27.5
	bblanchon/ArduinoJson@^6.19.4
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit SSD1306@^2.5.7
	adafruit/Adafruit GFX Library@^1.11.7
build_type = debug
```

robeastwood, 2023-12-06:
Hi @valeros I just tested with platform v6.3.0 and it works fine. So something has gone wrong in v6.4.0 somewhere!
Just shout if i can do any more troubleshooting or you need more info.

valeros, 2023-12-06:
The Arduino core was updated from 2.0.9 to 2.0.11 in v6.4.0. Could you also check it out with the latest platform version from the dev branch? The dev branch has support for the latest Arduino core v2.0.14 (Note: a Git client is required.)

```ini
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32doit-devkit-v1
...
```

robeastwood, 2023-12-06:
Yes, it's working fine on Espressif 32 (6.4.0+sha.bc5b51f)

And just to be double check, i restored back to the live version of 6.4.0 and it's again broken.

So it looks like it'd be good to do a patch asap - this particular bug doesn't show itself (in this use case) until the device in question is promted to do an OTA update, which by that point may have been put out in the field and thus is unrecoverable remotely.

valeros, 2024-01-08:
Should be fixed in the latest stable v6.5.0.

closed 2024-01-08 by valeros