platformio/platform-espressif32 #1248: [ESP8266/MacOS] Debugging with GDBStub hang

DenPlusPlus, 2023-12-08:
Hello. I'm trying to setup interactive debugging with Clion for Wemos D1 mini using GDBStub library. Following the GDBStub instruction, I configured my project and debugging works smoothly if I use xtensa-lx106-elf-gdb directly, but If I try to debug with _pio debug_ it hangs on **target remote** GDB command. This is what running debugging looks like:

```
% /opt/homebrew/bin/pio debug --interface=gdb -x .pioinit -v -e d1_mini 
GNU gdb (GDB) 8.2.50.20190202-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin14 --target=xtensa-lx106-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
**started gdbcmds!
finished gdbcmds!**
(gdb) Reading symbols from /Users/user/CLionProjects/blinker/.pio/build/d1_mini/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = custom
PlatformIO: Initializing remote target...
**started debug_init_cmds!    #Here it hangs for some time**
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
```

Here is the C code I'm using:
```c
#include <Arduino.h>
#include <GDBStub.h>

void setup() {
    gdbstub_init();
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
    digitalWrite(LED_BUILTIN, HIGH); 
    delay(1000); 
    digitalWrite(LED_BUILTIN, LOW); 
    delay(1000); 
}
```
And the _platformio.ini_:
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
debug_port = /dev/cu.usbserial-110

build_flags = -ggdb3 -Og -g3 -O0

debug_tool = custom
debug_server =
    /Users/user/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gdb
    -b
    115200
    -x
    /Users/user/.platformio/packages/framework-arduinoespressif8266/libraries/GDBStub/gdbcmds

debug_init_cmds =
    echo started debug_init_cmds!\n
    target remote $DEBUG_PORT
    echo finished debug_init_cmds!\n

debug_init_break = break loop
```

_gdbcmds_ is based on the gdbcmds provided with GDBStuы contains:
```
echo started gdbcmds!\n
set remote hardware-breakpoint-limit 1
set remote hardware-watchpoint-limit 1

# Some GDBstub settings
set remote interrupt-on-connect on
set remote kill-packet off
set remote symbol-lookup-packet off
set remote verbose-resume-packet off
# The memory map, so GDB knows where it can install SW breakpoints
mem 0x20000000 0x3fefffff ro cache
mem 0x3ff00000 0x3fffffff rw
mem 0x40000000 0x400fffff ro cache
mem 0x40100000 0x4013ffff rw cache
mem 0x40140000 0x5fffffff ro cache
mem 0x60000000 0x60001fff rw

# Change the following to your serial port and baud
set serial baud 115200

file /Users/user/CLionProjects/blinker/.pio/build/d1_mini/firmware.elf

echo finished gdbcmds!\n
```

Please help me to debug and fix this scenario.

**Environment:**
PlatformIO Core, version 6.1.11
MacOS Sonoma
framework-arduinoespressif8266, version 3.30102.0

Jason2866, 2023-12-10:
This is Platformio platform-espressif32 repo. No esp8266.

DenPlusPlus, 2023-12-10:
My fault. Moved to https://github.com/platformio/platform-espressif8266/issues/311.

closed 2023-12-10 by DenPlusPlus