platformio/platform-espressif32 #1255: Problem reading pictures from littlefs (espressif32 @ 6.4.0)

IZ76, 2023-12-15:
After updating the espressif32 platform from version 6.3.2 to version 6.4.0, an issue occurred when reading the picture.ico file from the littlelefs file system.
When a file is requested, the kernel freezes and the watchdog timer is triggered.

-------
Запит файлу: /img/favicon.ico
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40089f37  PS      : 0x00060930  A0      : 0x80106793  A1      : 0x3ffe8330
A2      : 0x3fff3528  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fffcc88  
A6      : 0x00000000  A7      : 0xffe1eaf6  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000016  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40089e95  LEND    : 0x40089eb7  LCOUNT  : 0x00000000  

Backtrace: 0x40089f34:0x3ffe8330 0x40106790:0x3ffe8350 0x401068b6:0x3ffe8370 0x400f4811:0x3ffe8390 0x400f3016:0x3ffe8400 0x400f2f69:0x3ffe8450 0x400e0e42:0x3ffe8470 0x400f5daa:0x3ffe84d0 0x400f3969:0x3ffe8520 0x400f3a79:0x3ffe8570 0x400f3c61:0x3ffe85c0 0x401ab38d:0x3ffe85e0 0x401ab409:0x3ffe8610 0x401abbfa:0x3ffe8630

ELF file SHA256: 8c5ccc123d11b89f

Rebooting...
--------

I indicate a downgraded version # platform = espressif32 @ 6.3.2
And everything is OK

BlueAndi, 2023-12-28:
The backtrace can not be used to determine the problem, please provide it analyzed. Use VSCode + PIO to get it automatically during monitoring.

IZ76, 2024-01-01:
I'm using VSCode + PIO.
Tell me how to get it automatically during monitoring.

_________________________

З повагою, Іван Захаров
*tel: +38 097 017 6323 (Viber, WhatsApp, Telegram)*
*_________________________*

BlueAndi, 2024-01-01:
[image]

Change to PIO project tasks and choose "Monitor" in your environment.

IZ76, 2024-01-05:
Please. Part at the time of reboot.

10:03:05  Старт повторного підключення MQTT...
10:03:05  MQTT перепідключення...підключено
10:03:05  mqtt_sub[0]=Informer/temp
10:03:05  mqtt_sub[1]=Informer/hum
10:03:05  mqtt_sub[2]=Informer/press
10:03:05  mqtt_sub[8]=ESP/BrightsDay
10:03:05  mqtt_sub[9]=ESP/BrightsNight
auth =1234asdf5678
Запит файлу: /sens.html
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was
unhandled.

Core  1 register dump:
PC      : 0x40089f37  PS      : 0x00060d30  A0      : 0x801067f3  A1      :
0x3ffe8330
A2      : 0x3fff42bc  A3      : 0x00000000  A4      : 0x00000001  A5      :
0x3fffcfec
A6      : 0x00000000  A7      : 0x00000074  A8      : 0x00000000  A9      :
0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     :
0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000016  EXCCAUSE:
0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40089e95  LEND    : 0x40089eb7  LCOUNT  :
0x00000000

Backtrace: 0x40089f34:0x3ffe8330 0x401067f0:0x3ffe8350
0x40106916:0x3ffe8370 0x400f4871:0x3ffe8390 0x400f3076:0x3ffe8400
0x400f2fc9:0x3ffe8450 0x400e07d5:0x3ffe8470 0x400f5e0a:0x3ffe84d0
0x400f39c9:0x3ffe8520 0x400f3ad9:0x3ffe8570 0x400f3cc1:0x3ffe85c0
0x401ab3ed:0x3ffe85e0 0x401ab469:0x3ffe8610 0x401abc5a:0x3ffe8630

ELF file SHA256: b2536ad9ca39536a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
0:30:45  Зчитування данних WiFi (146):
{"APSSID":"ESP-Info","APPASS":"11112222","AUTH":"1234asdf5678","AUTHON":1,"HIDDENNETWORKS":0,"num":2,"nets":["st-509","

_________________________

З повагою, Іван Захаров
*tel: +38 097 017 6323 (Viber, WhatsApp, Telegram)*
*_________________________*

BlueAndi, 2024-01-05:
The exception decoder is not active as it looks like.
Please additional add to your platformio.ini

```
monitor_filters = esp32_exception_decoder
```
Please note, without this information or a link to your code or example code to reproduce the problem, its not possible to see the real problem. From the register dump its a null pointer exception and I guess its not the problem of LittleFS, but in the application.

IZ76, 2024-01-05:
Hello. Here's the monitor.
Запит файлу: /sens.html
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was
unhandled.

Core  1 register dump:
PC      : 0x40089f37  PS      : 0x00060730  A0      : 0x801067f3  A1      :
0x3ffe8330
A2      : 0x3fff18d0  A3      : 0x00000000  A4      : 0x00000001  A5      :
0x3fffa600
A6      : 0x00000000  A7      : 0x00000074  A8      : 0x00000000  A9      :
0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     :
0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000016  EXCCAUSE:
0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40089e95  LEND    : 0x40089eb7  LCOUNT  :
0x00000000

Backtrace: 0x40089f34:0x3ffe8330 0x401067f0:0x3ffe8350
0x40106916:0x3ffe8370 0x400f4871:0x3ffe8390 0x400f3076:0x3ffe8400
0x400f2fc9:0x3ffe8450 0x400e07d5:0x3ffe8470 0x400f5e0a:0x3ffe84d0
0x400f39c9:0x3ffe8520 0x400f3ad9:0x3ffe8570 0x400f3cc1:0x3ffe85c0
0x401ab3ed:0x3ffe85e0 0x401ab469:0x3ffe8610 0x401abc5a:0x3ffe8630

  #0  0x40089f34:0x3ffe8330 in memmove at
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/string/memmove.c:66
  #1  0x401067f0:0x3ffe8350 in String::move(String&) at
C:/Users/ivan/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.cpp:237
  #2  0x40106916:0x3ffe8370 in String::operator=(String&&) at
C:/Users/ivan/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.cpp:277
  #3  0x400f4871:0x3ffe8390 in
AsyncBasicResponse::_respond(AsyncWebServerRequest*) at
.pio/libdeps/esp32doit-devkit-v1/ESP Async
WebServer/src/WebResponses.cpp:205 (discriminator 1)
  #4  0x400f3076:0x3ffe8400 in
AsyncWebServerRequest::send(AsyncWebServerResponse*) at
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/WebRequest.cpp:726
  #5  0x400f2fc9:0x3ffe8450 in AsyncWebServerRequest::send(int, String
const&, String const&) at .pio/libdeps/esp32doit-devkit-v1/ESP Async
WebServer/src/WebRequest.cpp:773
  #6  0x400e07d5:0x3ffe8470 in std::_Function_handler<void
(AsyncWebServerRequest*),
server_init()::{lambda(AsyncWebServerRequest*)#40}>::_M_invoke(std::_Any_data
const&, AsyncWebServerRequest*&&) at src/SRV.h:664
      (inlined by) _M_invoke at
c:\users\ivan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #7  0x400f5e0a:0x3ffe84d0 in std::function<void
(AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at
c:\users\ivan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
     (inlined by)
AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at
.pio/libdeps/esp32doit-devkit-v1/ESP Async
WebServer/src/WebHandlerImpl.h:118
  #8  0x400f39c9:0x3ffe8520 in AsyncWebServerRequest::_parseLine() at
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/WebRequest.cpp:582
(discriminator 1)
  #9  0x400f3ad9:0x3ffe8570 in AsyncWebServerRequest::_onData(void*,
unsigned int) at .pio/libdeps/esp32doit-devkit-v1/ESP Async
WebServer/src/WebRequest.cpp:123
  #10 0x400f3cc1:0x3ffe85c0 in std::_Function_handler<void (void*,
AsyncClient*, void*, unsigned int),
AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*,
AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned
int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&,
std::_Any_data const&, unsigned int&&) at
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/WebRequest.cpp:76
      (inlined by) _M_invoke at
c:\users\ivan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #11 0x401ab3ed:0x3ffe85e0 in std::function<void (void*, AsyncClient*,
void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int)
const at
c:\users\ivan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:915
  #12 0x401ab469:0x3ffe8610 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*,
signed char) at
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:1191
  #13 0x401abc5a:0x3ffe8630 in _async_service_task(void*) at
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:159
      (inlined by) _async_service_task at
.pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:194

ELF file SHA256: b2536ad9ca39536a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

If in platformio.ini I specify platform = espressif32 @ 6.3.2 then
everything is OK

_________________________

З повагою, Іван Захаров
*tel: +38 097 017 6323 (Viber, WhatsApp, Telegram)*
*_________________________*

BlueAndi, 2024-01-05:
It looks like its related to this issue: https://github.com/espressif/arduino-esp32/issues/6258

How does your lambda function looks like in SRV.h line 664 ?

IZ76, 2024-01-08:
> It looks like its related to this issue: espressif/arduino-esp32#6258
>
> How does your lambda function looks like in SRV.h line 664 ?

Hi
[image: 1.jpg]
Here is the function to read from memory.
[image: 2.jpg]

If you comment on line 19, then there is no error.
If you change to use "platform = espressif32 @ 6.3.2" - then there is no
error.

_________________________

З повагою, Іван Захаров
*tel: +38 097 017 6323 (Viber, WhatsApp, Telegram)*
*_________________________*

valeros, 2024-01-08:
Hi @IZ76, could you please retest your project with the latest v6.5.0 of the `espressif` development platform?

closed 2024-01-09 by valeros