platformio/platform-espressif32 #1256: Don't know where the 10ms delay comes from? 【不知道哪里来的10ms延时？】

mysteriousTreasure, 2023-12-15:
Using the software environment,【使用软件环境，】
  vscode+platformIO  arduino
  espressif32 6.4.0
Use hardware【硬件使用】：ESP32-S3 N16R8

Question:
There is an inexplicable 10ms do not know what task is being performed.
【问题：
出现莫名其妙的10ms不知道在执行什么任务。】

for example
【举个简单例子】

`
#include <Arduino.h>
unsigned long lasttime = 0;
unsigned long nowtime = 0;
void setup()
{
    Serial.begin(115200);
}

void loop()
{
    lasttime = micros();
    Serial.printf("1->lasttime=%u,", lasttime);

    nowtime = micros() - lasttime;
    Serial.printf("%u\n", nowtime);

    vTaskDelay(1);
}`
[image]

And if vTaskDelay(1); Change to vTaskDelay(2). There is no such phenomenon
The output is normal.
【而如果vTaskDelay(1);修改为vTaskDelay(2);就没这个现象
输出很正常】
[image]

There is also a normal writing output, or vTaskDelay(1); 
However, when you do not use a string of more than two bytes to print, it is normal
【还有一种写法也输出正常，还是vTaskDelay(1);但是不使用2个字节以上的字符串打印时候，也正常】
[image]

And when I thought I had found the problem, what was this routine?
【而我满心欢喜以为找到问题时候，结果这个例程又是为什么呢？】
[image]

joba-1, 2023-12-15:
can’t explain everything, but the 10ms is roughly what it takes to output the ~180 chars on Serial that accumulate between the 10ms delays.
Probably some threshold where this is done asynchronous vs sync is at work.

Have you tried Serial.flush() after each print?

valeros, 2023-12-19:
Hi @mysteriousTreasure, the issue tracker in this repository is not the right place for such generic questions, please forward to https://github.com/espressif/arduino-esp32

closed 2023-12-19 by valeros