platformio/platform-espressif32 #1260: ESP32-S3 using encryption not setting fuses

apppie123, 2023-12-18:
**In short:**
I have a product developed using platformIO and using encryption in release mode. I needed to make changes and so I changed to development mode. The result after flashing is always "Invalid header".

**Doesn't work (using PlatformIO):**
As a test I reinstalled ESP-IDF, PlatformIO and VSCode. Then ran a very simple test:
1) I loaded a new example project called “espidf-hello-world”
2) Changed the board to board = esp32-s3-devkitc-1
3) Flashed the board…
Succes! Board boots up fine.

4) I go to menuconfig in platformio
5) I check security features → Enable flash encryption on boot (READ DOCS FIRST) → Development
 UART ROM download mode (UART ROM download mode (Enabled (not recommended)))
6) I flash the board again…

“invalid header: 0x115542bf”
There’s not much I could do wrong there, right? I checked the fuses and it seems that SPI_BOOT_CRYPT_CNT is still set to 0. Expected it 0x01.

**Does work using ESP-IDF directly:**
1) Loading example project "hello-world" in VSCode.
2) idf.py menuconfig and set the same flash encryption fuses as above.
3) flash the board

Succes!! I also checked the fuses and now SPI_BOOT_CRYPT_CNT is set to 0x01.

valeros, 2023-12-18:
Hi @apppie123, unfortunately, encryption is not supported at the moment. Tracked here https://github.com/platformio/platform-espressif32/issues/305

closed 2023-12-18 by valeros

apppie123, 2023-12-18:
> Hi @apppie123, unfortunately, encryption is not supported at the moment. Tracked here #305
That's quite an old issue and flash encryption through platformio used to work fine for the past months. So I'm quite confused that it's now suddenly broken and not supported anymore.