platformio/platform-espressif32 #1261: Platformio : Heltec ESP32 - Wireless_Paper board

Lapporatory992, 2023-12-18:
Hi,

Would it possible to add the HeltecESP32 "wireless_paper" board to the list of boards supported in PlatformIO ?
https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/boards.txt

Line 1406:
##############################################################
#############################################################
wireless_paper.name=wireless_paper
wireless_paper.vid.0=0x303a
wireless_paper.pid.0=0x1001

wireless_paper.bootloader.tool=esptool_py
wireless_paper.bootloader.tool.default=esptool_py

wireless_paper.upload.tool=esptool_py
wireless_paper.upload.tool.default=esptool_py
wireless_paper.upload.tool.network=esp_ota

wireless_paper.upload.maximum_size=4026368
wireless_paper.upload.maximum_data_size=327680
wireless_paper.upload.flags=
wireless_paper.upload.extra_flags=
wireless_paper.upload.use_1200bps_touch=false
wireless_paper.upload.wait_for_upload_port=false

etc.
*********

Thanks!

valeros, 2023-12-18:
Hi @Lapporatory992, there is no support for this board in the official Arduino core for ESP32 https://github.com/espressif/arduino-esp32, so a proper variant for this board should be added there in the first place.

Lapporatory992, 2023-12-18:
Thanks so much Valeros for your quick answer! I have just asked the same on the Espressif repo as per your suggestion
Have  a great day!

closed 2023-12-18 by Lapporatory992