platformio/platform-espressif32 #1263: Debuuging on Seeed Studio xiao-ESP32-S3 is not working

AlbertoLopSie, 2023-12-21:
Hi!

Debugging is not working in the Seeed Studio xiao-ESP32-S3 board.  According to documentation, the S3 supports direct (no external probe) debugging through its internal JTAG USB interface however, I cannot make i work.

Also, there's a confusion on the Debugging section of the [PlatformIO Board doc](https://docs.platformio.org/en/latest/boards/espressif32/seeed_xiao_esp32s3.html): it says that this board **doesn't support 1-click debugging** when it does.

``` yaml
// platformio.ini

[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
build_type = debug 
debug_tool = esp-builtin
; debug_tool = cmsis-dap              // Even when doc says this the default it doesn't work
; upload_protocol = esp-builtin       // Doesn't work

debug_init_break = tbreak setup

debug_port = /dev/cu.usbmodem21301     // on my Mac
monitor_port = /dev/cu.usbmodem21301
upload_port = /dev/cu.usbmodem21301
```
Only causes debugger to start but freeze with the following at the debug console:

```sh
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (34:85:18:91:38:88)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Reading symbols from /Users/Alberto/dev/DMI/HelloS3/.pio/build/seeed_xiao_esp32s3/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
```
and debugging session never starts.

It looks like the JTAG interface connects ok because the CPU is recognized, but after that nothing happens.

Uploading through the `esp-builtin` protocol doesn't work either.

Any hint?

AlbertoLopSie, 2023-12-21:
> Hi @AlbertoLopSie, what's the error message when uploading?

Hi Valerrii,

Log attached:

``` sh
pyenv shell 3.7.8
Processing seeed_xiao_esp32s3 (platform: espressif32; board: seeed_xiao_esp32s3; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/seeed_xiao_esp32s3.html
PLATFORM: Espressif 32 (6.4.0+sha.2575a81) > Seeed Studio XIAO ESP32S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio/build/seeed_xiao_esp32s3/firmware.elf
Checking size .pio/build/seeed_xiao_esp32s3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.7% (used 18608 bytes from 327680 bytes)
Flash: [=         ]   7.6% (used 253677 bytes from 3342336 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-builtin
Uploading .pio/build/seeed_xiao_esp32s3/firmware.bin
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Trying to use configured scan chain anyway...
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Warn : Bypassing JTAG setup events due to errors
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Unexpected OCD_ID = 00000000
Warn : target esp32s3.cpu0 examination failed
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Unexpected OCD_ID = 00000000
Warn : target esp32s3.cpu1 examination failed
Error: couldn't bind gdb to socket on port 3333: Address already in use
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Trying to use configured scan chain anyway...
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Warn : Bypassing JTAG setup events due to errors
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Unexpected OCD_ID = 00000000
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Unexpected OCD_ID = 00000000
Error: Target not examined yet
Error: Target not examined yet
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
Error: missing data from bitq interface
Error: Unexpected OCD_ID = 00000000
embedded:startup.tcl:1184: Error: ** Unable to reset target **
in procedure 'program_esp' 
in procedure 'program_error' called at file "/Users/Alberto/.platformio/packages/tool-openocd-esp32@2.1100.20220706/share/openocd/scripts/target/esp_common.cfg", line 104
at file "embedded:startup.tcl", line 1184
*** [upload] Error 1
================================================================ [FAILED] Took 1.71 seconds ================================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'seeed_xiao_esp32s3'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

AlbertoLopSie, 2023-12-21:
Well... new findings...

After trying to start a debug session, a `python3.9` process is spawned. When (after realizing it wont work) I stop debugging (pressing the Stop button or Run|Stop debugging menu) the python process goes to 99% CPU utilization.

In this state, uploading through `esp-builtin` protocol behaves as shown in my previous message. But, if a kill the `python3.9` process AND the `openocd` the uploads completes succesfully !!

``` sh
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 12000 kHz

** Programming Started **
** Programming Finished in 6079 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1296 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1028 ms **
** Verify Started **
** Verify OK **
** Programming Started **
** Programming Finished in 1151 ms **
** Verify Started **
** Verify OK **
shutdown command invoked
```

In any case Debug is still NOT WORKING

valeros, 2023-12-22:
1. Please make sure there are no orphan python3, platformio or openocd processes. 
2. Run the `pio debug --interface=gdb -- -x .pioinit` directly in the IDE terminal.
3. Attach logs here

AlbertoLopSie, 2023-12-22:
I don't know if adds something, but I followed the gdb script creation all the way to `.platformio/platforms/espressif32/platform.py` and discovered that the script gets stuck on the command:

> target extended-remote $DEBUG_PORT

that in my system resolves to:

> target extended-remote /dev/cu.usbmodem21301

the script never completes that instruction.

valeros, 2023-12-22:
How come the OpenOCD version changed from `v0.11.0-esp32-20220706 (2022-07-06-15:49)` to `v0.12.0-esp32-20230313 (2023-03-13-09:08)` ?

AlbertoLopSie, 2023-12-22:
By upgrading the pio package

valeros, 2023-12-22:
> By upgrading the pio package

Have you installed it explicitly? This version of the OpenOCD package is currently not recommended for use. 

> debug_port = /dev/cu.usbmodem21301

Please remove this line from your configuration, in case with `esp-builtin` this option is used to describe a custom debug port of a debug server rather than a name of an end device in your system.

AlbertoLopSie, 2023-12-22:
Hi Valerii,

Yes, upgraded via pio pkg upgrade. I can switch to another version (the
upgrade does not delete other versions) which one do you advise to use?

I tried removing the “debug_port” setting, but didn’t work (LIB_USB_ errors)

valeros, 2023-12-22:
> Yes, upgraded via pio pkg upgrade. I can switch to another version (theupgrade does not delete other versions) which one do you advise to use?

Just keep the default one v0.11 by deleting all other `tool-openocd-esp32` packages in `/Users/Alberto/.platformio/packages`

> I tried removing the “debug_port” setting, but didn’t work (LIB_USB_ errors)

Again, please make sure there are no orphan python3, platformio or openocd processes. Then try to debug the project with the following config and attach share the error log here. 

```ini
[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
build_type = debug 
debug_tool = esp-builtin
upload_protocol = esp-builtin 
```

gerekon, 2023-12-22:
@AlbertoLopSie 

> I don't know if adds something, but I followed the gdb script creation all the way to `.platformio/platforms/espressif32/platform.py` and discovered that the script gets stuck on the command:
>
>
> that [...]

Looks like this could be the reason. `DEBUG_PORT` should be `localhost:3333` - OpenOCD port for GDB to connect to, not USB_SERIAL one.

AlbertoLopSie, 2023-12-22:
## Wow!!!  That was a hit!!!

Now I'm having something!!

First time I tried your suggestions debugger started in almost no time and my cursor appeared just into main().

**BUT** (there's always a BUT)

> disclaimer: it's going to be a loooong post due to logs

When I tried to add `tbreak setup` to platformio.ini I got:

``` sh
Reading symbols from /Users/Alberto/dev/DMI/HelloS3/.pio/build/seeed_xiao_esp32s3/firmware.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
Running esp32s3.cfg (20220706)...
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: serial (34:85:18:91:38:88)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x4201F166, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4201F166, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 129 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 129 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 132 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 129 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 52 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
0x4201f166 in esp_pm_impl_waiti () at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32s3/include/hal/cpu_ll.h:182
182	/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32s3/include/hal/cpu_ll.h: No such file or directory.
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
[esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Core was reset.
[... 130 lines cut ...]
```
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Hardware assisted breakpoint 15 at 0x42002340: file /Users/Alberto/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp, line 56.
Info : [esp32s3.cpu0] Target halted, PC=0x42002340, debug_reason=00000001
[esp32s3.cpu0] Target halted, PC=0x42002340, debug_reason=00000001
Info : Set GDB target to 'esp32s3.cpu0'
Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x4201F166, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x4201F166, debug_reason=00000000

Thread 8 "main" hit Temporary breakpoint 15, app_main () at /Users/Alberto/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:56
56	{
```

We are getting closer indeed, but the debugger behavior look a little unstable.

BTW.. If I have more than on ESP32 connected to my Mac (I usually do) How to tell which one I want to debug ?

Another question: What's causing those orphan openocd and python processes?  I developed a script to kill them all. Can it be run automatically?

Thank you very much Valerii !!

AlbertoLopSie, 2023-12-22:
I was able to capture the case when the debug console logs repeats for ever...  It was just after posting my last response and repeating the whole process to get sure I got the procedure... It looks like I don't

This time PlatformIO caused a rebuild because i added the line `debug_speed = 5000` to see if that helps.  The log after building got a lot bigger and took more time to start the blue/red messages. After that the process entered an infinite loop

``` sh
Processing seeed_xiao_esp32s3 (platform: espressif32; board: seeed_xiao_esp32s3; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/seeed_xiao_esp32s3.html
PLATFORM: Espressif 32 (6.4.0+sha.2575a81) > Seeed Studio XIAO ESP32S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Compiling .pio/build/seeed_xiao_esp32s3/src/main.cpp.o
Building .pio/build/seeed_xiao_esp32s3/bootloader.bin
Generating partitions .pio/build/seeed_xiao_esp32s3/partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/Print.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/USB.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/USBCDC.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/USBMSC.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/WString.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/base64.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-dac.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-gpio.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-i2c-slave.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-i2c.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-ledc.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-matrix.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-misc.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-psram.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-rgb-led.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-rmt.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-sigmadelta.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-time.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/seeed_xiao_esp32s3/FrameworkArduino/esp32-hal-uart.c.o
[... 148 lines cut ...]
Info : [esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Hardware assisted breakpoint 1 at 0x42002340: file /Users/Alberto/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp, line 56.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = thb app_main`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Info : [esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Core was reset.
...
...
```

Last 4 lines repeated forever...

Hope it helps!

AlbertoLopSie, 2023-12-22:
Confirmed: any debug session, sucesfull or not, leaves the program memory corrupt (it doesn't run alone unless it's reflashed)

Did the debugger left the breakpoints in flash after quitting ?

gerekon, 2023-12-25:
> Confirmed: any debug session, sucesfull or not, leaves the program memory corrupt (it doesn't run alone unless it's reflashed)
> 
> Did the debugger leave the breakpoints in flash after quitting ?

If openocd exits normally it removes all flash breakpoints. If you kill it they remain there.

felixFleetZero, 2025-01-16:
@AlbertoLopSie did you ever figure something out to be able to debug with esp-builtin the XIAO esp32s3?

davidandreoletti, 2025-02-25:
> [@AlbertoLopSie](https://github.com/AlbertoLopSie) did you ever figure something out to be able to debug with esp-builtin the XIAO esp32s3? I'm also having trouble trying to debug it with platformio!

@AlbertoLopSie (ping)

davidandreoletti, 2025-05-02:
No improvements with:
- PlatformIO Core, version 6.1.18
   |-> Open On-Chip Debugger v0.11.0-esp32-20220706 (2022-07-06-15:49)

```
Use 'set mi-async'.

PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-builtin
PlatformIO: Initializing remote target...
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.

Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : starting gdb server for esp32s3.cpu0 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s3.cpu0] Target halted, PC=0x4037AA42, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x420081D1, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 177 KB
Info : Flash mapping 1: 0x40020 -> 0x42000020, 635 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 177 KB
Info : Flash mapping 1: 0x40020 -> 0x42000020, 635 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 636 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0a0020, 177 KB
Info : Flash mapping 1: 0x40020 -> 0x42000020, 635 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 180 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
0x4037aa42 in esp_cpu_wait_for_intr () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:64
warning: 64	/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c: No such file or directory
Info : Detected FreeRTOS version: (10.5.1)
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
[esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[... 202 lines cut ...]
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Breakpoint 1 at 0x420937d4: file src/main.cpp, line 71.
Note: automatically using hardware breakpoints for read-only addresses.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = break setup`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
```

closed 2025-07-18 by stale[bot]