platformio/platform-espressif32 #1264: Got stuck on the `develop` branch for my `platform` & could not build for hours, even after changing the value several times and removing directories entirely

aremmell, 2023-12-24:
I am not entirely sure what happened in what order, but:

1. At some point today, in an attempt to get a new board working, I changed my `platformio.ini` to look like this:

```ini
[env:adafruit_qualia_s3_rgb666]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = framework-espidf
board = adafruit_qualia_s3_rgb666
framework = arduino, espidf
```

2. After I made this change, removing `~/.espressif/packages/platform-espressif32*` entirely, plus completely removing and reinstalling ESP-IDF several times, I was apparently stuck on the `develop` branch of the git repository for my platform package.

I know this is true, because I reverted the syntax to `platform = espressif32`, and even `platform = espressif32 @ 6.4.0` _(*and* re-deleted the directories)_, but I still got the following error from `CMakeFiles.txt:19` every time I tried to build:

```cmake
if (idf_version VERSION_GREATER max_supported_idf_version)
  message(FATAL_ERROR "Arduino-esp32 can be used with ESP-IDF versions "
                      "between ${min_supported_idf_version} and ${max_supported_idf_version}, "
                      "but a newer version is detected: ${idf_version}.")
endif()
```

The numbers were something like: "between 4.4.0 and 4.4.99, but a newer version is detected: 5.1.2"

I knew I had explicitly installed ESP-IDF 4.x.x, so I could not understand how I kept getting this error.

3. In an effort to figure out WTF was going on, I inadvertently stumbled upon commit https://github.com/platformio/platform-espressif32/commit/bc5b51f0202afc794e1a49dfb34ebcb82b080140 and noticed that the version number had been bumped up for ESP-IDF. I first thought I must be going crazy, because I wasn't supposed to be pulling this library from git any longer, but sure enough, I reverted those changes and set my `platform = espressif32 @ 6.4.0` and `platform_packages = framework-espidf @ ~3.40405.0`; now everything is back to normal.

I suppose I am suggesting two things:

1. The aforementioned commit is bad and has broken the build, since the CMake scripts are still looking for 4.x.x, yet 5.1.2 is purposely installed.
2. There is a bug lurking somewhere around setting the `platform` to a git repository URL, then changing it back after the fact. It's getting stuck and continuing to pull from git _(or it has locally cached it)_ instead of the specified version in the ini file.

Please let me know if there are any logs I can send, or if you'd like me to repro with some conditions set up in a certain way. I'd be glad to do that in order to prevent someone else from losing half a day on this!

**Versions:**
- macOS 14.2.1
- VS Code latest, PlatformIO latest

valeros, 2023-12-25:
Hi @aremmell 

> After I made this change, removing ~/.espressif/packages/platform-espressif32*

I'm not sure if it's a typo, but the `~/.espressif/packages/*` folder has nothing to do with PlatformIO. Just delete the following folders `~/.platformio/packages` and `~/.platformio/platforms/platform-espressif32*` and then run your project again, PlatformIO will install all required default packages (for mixed IDF/Arduino projects it will be Arduino core v2.0.14 and IDF v4.4.6).

aremmell, 2023-12-25:
Yeah, sorry. I meant ~/.platformio.

Thanks for the info. Perhaps a better error message is needed that includes this information, since it is not obvious and I had to do a ton of digging to resolve it.

Maybe something explicit with the instructions you just provided.

valeros, 2023-12-26:
I just realized I didn't comment your suggestions:

> The aforementioned commit is bad and has broken the build, since the CMake scripts are still looking for 4.x.x, yet 5.1.2 is purposely installed.

If I understood correctly, CMake scripts are doing the right thing as Arduino Core v2.0.14 require IDF v4.4.6, so unfortunately, there isn't much we can do. 

> There is a bug lurking somewhere around setting the platform to a git repository URL, then changing it back after the fact. It's getting stuck and continuing to pull from git (or it has locally cached it) instead of the specified version in the ini file.

I guess there is a misconception of how PlatformIO handles installed development platforms, so let me clarify this behavior a bit:

In the first try after removing the `.platformio` folder, PlatformIO automatically installed the platform directly from this repository (via a repo URL). In this case the version of the platform was something like `6.4.0+sha.2571753`. Then you decided to remove the repo URL from the `platform` field and just specified the platform name `espressif32` without any version constraint. Since you didn't set a precise version and there is already available platform with the version `6.4.0+sha.2571753`, PlatformIO just used it. The case with `platform = espressif32 @ 6.4.0` is quite similar, the `6.4.0+sha.d9d2738` version satisfies the `6.4.0` constraint, so nothing new installed. 

> Maybe something explicit with the instructions you just provided.

Those instructions are mostly specific to your situation, I'm not sure if there is a proper place for them in this repo.

aremmell, 2023-12-27:
@valeros Thanks for the response. I am simply trying to relate that as a consumer of the software, it is not obvious how things are done behind the scenes whenever you make changes to `platformio.ini`.

In fact, all of the platform/framework/arduino-this and espressif-that and platformio-that is really confusing to a newcomer.

I would recommend making the error message in question more actionable by including some text that says something like "... Please check your platformio.ini configuration. You may also have cached installations at `~/.platformio/packages/` and `~/.platformio/platforms/`. Try removing those directories if you continue to see this error"

That is just from my perspective; maybe everyone else knows something I don't 😉

closed 2025-02-25 by stale[bot]