platformio/platform-espressif32 #1268: fatal error: WiFi.h: No such file or directory

peterus, 2023-12-30:
In the process of updating the platform from 6.4.0 to 6.5.0 a new compilation issue is popping up:

```
In file included from ~/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.cpp:24:
~/.platformio/packages/framework-arduinoespressif32/libraries/Ethernet/src/ETH.h:27:10: fatal error: WiFi.h: No such file or directory
```

I could trace the issue back to the esp32 arduino core as there where changes in the `ETH.cpp` file, but they do not look like changes which would introduce such an compilation issue. My code is using the ETH part of the arduino core.
Some suggestions?

my CI pipeline with compilation fail: https://github.com/lora-aprs/LoRa_APRS_iGate/pull/342

BlueAndi, 2024-01-03:
Hi Peter,
can you add WiFi to the lib_deps and try again?

peterus, 2024-01-04:
thanks for looking into it. sadly it shows the same:

https://github.com/lora-aprs/LoRa_APRS_iGate/actions/runs/7404812034/job/20146866405

BlueAndi, 2024-01-04:
I see ... in this case remove WiFi from lib_deps again and change lib_ldf_mode to deep (without +).

Note, you will have to remove Werror to get it compiled at all. Unfortunately there is warning in the framework, which is already fixed in the main stream.

73

valeros, 2024-01-08:
Hi @peterus, thanks for reporting! It seems like a bug in the Library Dependency Finder, so I've filed an issue in the PlatformIO Core repository https://github.com/platformio/platformio-core/issues/4818 so it will be tracked there. 

As a workaround you can specify `build_flags = -DCONFIG_ETH_ENABLED` in your `platformio.ini` to help LDF detect all necessary dependencies correctly.

closed 2024-01-08 by valeros

peterus, 2024-01-08:
@valeros thanks for taking a look into this! Just tried your workaround and it is working now 👍 will monitor the other ticket to see when this issue is resolved.