platformio/platform-espressif32 #1271: Native USB seems to be unstable with 6.5.0

evansgl, 2024-01-03:
When I use 6.5.0 platform all of my ESP32-S2 devices (different boards and implementations) have a very unstable USB CDC.
After some minutes (random from 5 to 30+ minutes) the communication is lost and the device is reported as a faulty device in the Windows 11 device manager. (same happens with OSX. The device is unresponsive after a while) The only way to recover it is to reboot the device.

This does not happen with the 6.4.0 platform, which seems solid as a rock.
I have checked it with 5 different devices but I am unsure why this happens!

Has anybody experienced any similar issue?

valeros, 2024-01-08:
Hi @evansgl, could please run your project using Arduino IDE just to make sure it's not a regression in the framework itself?

evansgl, 2024-01-18:
I further investigated the issue.
It seems that it happens on older framework but after longer time.
Enabling both USB MSC and CDC causes unstable connection in CDC. The CDC after some times fails and the only way to recover is the reboot the ESP.
There must be a tune-in USB stack size to allow both MSC and CDC work stable.

Jason2866, 2024-01-18:
@evansgl This is not a Platformio issue. You should open an issue [here](https://github.com/espressif/arduino-esp32/issues)

evansgl, 2024-01-19:
Thank you. I opened this for the record: https://github.com/espressif/arduino-esp32/issues/9138

closed 2024-01-19 by evansgl