platformio/platform-espressif32 #1273: BLE connect fails after light sleep

flynny75, 2024-01-03:
I have raised this issue in the NimBLE issue tracker, but seeing as it worked on 5.1.1 and nothing later I will raise it here also

I have made a simple platformio project to demonstrate. Two configurations, one acts are the BLE server, the other as the client. The demo shows that the BLE is not working correctly after waking from light sleep.

The example project is attached to the NimBLE issue here
https://github.com/h2zero/NimBLE-Arduino/issues/616

valeros, 2024-01-08:
Hi @flynny75, please check with the latest platform version 6.5.0.

flynny75, 2024-01-08:
Unfortunately the problem remains in version 6.5.0

valeros, 2024-01-09:
Could you please try the same project in Arduino IDE? Does it work there?

flynny75, 2024-01-09:
Unfortunately the problem remains in Arduino IDE 2.2.1. 
Platform version `esp32 2.0.11`
Lib versions are:
`NimBLE-Arduino 1.4.1`
`FastLED 3.6.0`
The board I am using is an adafruit S3 No PSRAM.

I have included the sketch below:
```
#include <Arduino.h>
#include "NimBLEDevice.h"
#include <FastLED.h>
#include "WiFi.h"

bool isServer = false;

NimBLEClient* client;
NimBLEAdvertisedDevice* server;

int connectCount = 0;
CRGB leds[1];

class AdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    public:
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
        if(advertisedDevice->getName() == "NimBLE_light_sleep")
        {
            server = advertisedDevice;
            NimBLEDevice::getScan()->stop();
        }
    };
};

void flashLed(CRGB colour, int timeOn, int timeOff, int times)
{    
    leds[0] = CRGB::Black;
    FastLED.show();
    vTaskDelay(timeOff / portTICK_PERIOD_MS);

    int i = 0;
    while(i < times)
    {
        leds[0] = colour;
        FastLED.show();
        vTaskDelay(timeOn / portTICK_PERIOD_MS);

        leds[0] = CRGB::Black;
        FastLED.show();
        vTaskDelay(timeOff / portTICK_PERIOD_MS);
        i++;
    }
}

void setup() {
    FastLED.addLeds<NEOPIXEL, 33>(leds, 1);

  if(isServer)
  {
        NimBLEDevice::init("NimBLE_light_sleep");
        auto bleServer = NimBLEDevice::createServer();
        auto advertising = NimBLEDevice::getAdvertising();
        auto service = bleServer->createService("1111");
        auto c = service->createCharacteristic("2222");
        c->setValue("3333");

        service->start();
        advertising->start();

        Serial.begin(115200);
  }
  else
  {
        pinMode(LED_BUILTIN, OUTPUT);
        NimBLEDevice::init("");
        auto scanner = NimBLEDevice::getScan();
        scanner->setAdvertisedDeviceCallbacks(new AdvertisedDeviceCallbacks());
        scanner->start(0, [](NimBLEScanResults r){});
        client = NimBLEDevice::createClient();
        client->setConnectTimeout(10);
  }
}

void loop() {
  if(isServer)
  {
    // N/A
  }
  else
  {
    if(server == NULL){
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        return;
    }

    // Cycle for 5 times without light sleeping
    if(connectCount > 2)
    {
        log_e("===== LIGHT SLEEP =====");

        // light sleep for 5 seconds, user disconects serial
        auto configureLightSleepTimer = esp_sleep_enable_timer_wakeup(5 * 1000000);
        if(configureLightSleepTimer != ESP_OK)
        {
            log_e("bad sleep timer wake enable: %s\n", esp_err_to_name(configureLightSleepTimer));
            flashLed(CRGB::Red, 20, 20, 50);
        }

        auto sleepResult = esp_light_sleep_start();
        if(sleepResult != ESP_OK){
            log_e("bad sleep: %s\n", esp_err_to_name(sleepResult));
            flashLed(CRGB::Red, 20, 20, 50);
        }

        digitalWrite(LED_BUILTIN, HIGH);
        //wake up, pause for 5s to reconnect serial
        vTaskDelay(5000 / portTICK_PERIOD_MS);
        digitalWrite(LED_BUILTIN, LOW);
    }
    else
    {
        log_e("===== DELAY =====");
        vTaskDelay(5000 / portTICK_PERIOD_MS);
    }

    if(client->connect(server))
    {
        flashLed(CRGB::Green, 100, 100, 5);
    }
    else
    {
        flashLed(CRGB::Red, 100, 100, 5);
    }

    vTaskDelay(2000 / portTICK_PERIOD_MS);
    client->disconnect();
    vTaskDelay(2000 / portTICK_PERIOD_MS);

    connectCount++;
  }
}
```

Green flashes are observed while `connectCount < 2` and `vTaskDelay`, but red flashes after `esp_light_sleep_start`

flynny75, 2024-01-09:
The behaviour is the same with platform `2.0.14`. I also tried with the latest dev `3.X`, but it did not compile.

valeros, 2024-01-10:
Please forward your issue to https://github.com/espressif/arduino-esp32 repository, since the problem has nothing to do with PlatformIO and can be reproduced in the Arduino IDE.

closed 2024-01-10 by valeros