platformio/platform-espressif32 #1274: Upload filesystem image OTA uploads to the wrong Flash Location according to min_spiffs.csv

meakashrao, 2024-01-04:
This is the official min_spiffs.csv partition
[image]

So, the SPIFFs location is from 0x3D0000
I am using a 4MB Flash and ESP32 Wroom

My Platformio.ini file is 
[image]

Now, 
When I use Upload File System Image 
[image]
The correct Flash location is erased according to min_spiffs.csv

But, 
When I use Upload File System Image OTA
[image]
[image]
The Flash location erased is completely different. 

I am using SPIFFS to write a configuration File as I have the same product with some changes in Pin Map between production batches, so I was planning to have one firmware but based on Configuration bit read from EEPROM, I will read configuration_file_1 or configuration_file_2 from SPIFFS.

The demo testingConfiguration File is 
[image]
And I get a very simple but helpful log in the setup itself 
[image]

My current FW Version is: 202401031745-0000 as can be verified from my backed Server Logs
[image]
but when I do, Upload File System Image OTA
The Build does go to incorrect Flash Location and also makes my firmware go back to a different version
[image]

My OTA Mechanism is 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino
using Python

Now, I am finding two simple issues
a) what is the difference between Upload FileSystem Image and Upload FileSystem Image OTA 
b) why does Upload FileSystem Image OTA not write to the correct SPIFFS Location like Upload FileSystem Image does

Please advice on the same. I checked a previous issue similar to this but the solution seems to be not working. 

I use, Espressif 32 V6.4.0
Platfrom IO Core 6.1.11 and Home 3.4.4 as can be seen on the PIO Home below the Logo

valeros, 2024-01-11:
Hi @meakashrao 

>  what is the difference between Upload FileSystem Image and Upload FileSystem Image OTA

`Upload FileSystem Image` is the default target used for uploading via any wired interface (Serial, JTAG, etc.). `Upload FileSystem Image OTA` is meant to be used when you want to upload your image remotely and set the `--upload-port` to point to a network device. Here is documentation for this feature https://docs.platformio.org/en/latest/platforms/espressif32.html#over-the-air-ota-update

> why does Upload FileSystem Image OTA not write to the correct SPIFFS Location like Upload FileSystem Image does

The ` Upload FileSystem Image OTA` target is not supposed to be used when you want to upload via a wired interface, so it doesn't evaluate several offset options required by `esptool.py` since `espota.py` is used instead.

meakashrao, 2024-01-17:
Thank you. this helps a lot.

closed 2024-01-17 by meakashrao